gpiozero/gpiozero #329: Add ADC.voltage property?

lurch, 2016-05-24:
While looking at the example code shown on https://www.kickstarter.com/projects/raspitv/raspio-analog-zero-read-8-sensors-at-once-on-raspb I wondered if it might be worth adding an optional `max_voltage` property to the `AnalogInputDevice` constructor (defaulting it to `3.3`) which would then allow us to add a `voltage` property, which would simply return `self.value * self._max_voltage`, and then the example code would then become:
```python
from gpiozero import MCP3008
adc = MCP3008(channel=0)
print("measured voltage is: ", adc.voltage)
```

i.e. it'd work very similar to the `distance` / `max_distance` properties of the current `DistanceSensor` class.  And allowing `max_voltage` to be set (rather than fixing it at 3.3V) would allow you to adjust it as necessary (to get the 'true value') if using the MCP3008 with a voltage-divider to bring a higher-voltage down into the 3.3V range input of the MCP3008. (Or if using a lower voltage than 3V3 on the Vref input pin)

Good idea / bad idea? 
Or perhaps this should be added to `MCP3xxx` rather than `AnalogInputDevice` given that (strictly speaking) a generic analog-input-device might not necessarily be reading voltage? (is that possible / likely?)

Ping @raspitv  ;-)

bennuttall, 2016-05-26:
I think this makes sense.

pcopa, 2016-05-30:
@lurch I support this idea.

I've found out (the hard way) that the RPI 3.3V is very unstable and I needed to use a separate reference voltage to get a stable reading.  That's why I used the "raw_values" and did my own calculations.

This would help me.

pcopa, 2016-06-26:
First of all I'm impressed with all the progress that's been made lately. Finding an reading the documents has become much smother.

In reading through the ADC text, I was wondering if it would make sense to mention somewhere that the ADC output "(raw_)value" is the voltage_value relative to the "analog ground pin" and the "analog reference voltage pin" in the Bit-resolution specified by the device.
And that for the purpose of this document, it is assumed, that the "analog ground pin" is connected to GND and the "analog reference voltage pin" is connected to 3.3V.

On the other hand now that I'm writing this, I think it might confuse more than clarify.

Anyway just a thought.

lurch, 2016-06-26:
It's been a while since I read through the datasheets, but IIRC for the non-differential ADCs, the Analog ground **has** to be the same as the Digital ground (i.e. GND) - it's only the "fully differential" MCP33xx chips that allow an 'independent' analog ground voltage.
And ISTR that the "analog reference voltage pin" can't be greater than the chip's supply voltage (i.e. it has to be less than or equal to 3V3 when used with the Pi).

But I'm no electronics expert, and I can't be bothered re-reading the datasheets right now, so I could be wrong ;-)

pcopa, 2016-08-24:
Yes, @lurch you'r correct. That's how I understand it too.

commit 339fb1bd (2018-02-19): Merge pull request #331 from RPi-Distro/fix-329

Add ADC.voltage property, fix #329
commit cd879589 (2018-02-19): Add ADC.voltage property, fix #329
closed 2018-02-19 by waveform80