gpiozero/gpiozero #330: Add support for more motors

bennuttall, 2016-05-26:
As raised in #300, there's a suggestion to add more varieties of motor, such as the L298N.

> I believe to 'properly' support the L298N we'd need a Motor class with (Digital_forwards, Digital_backwards, PWM_speed) style controls.
> ~ @lurch 

Would this be another generic motor class? What would it be called?

Are there other types of motors not covered by this?

lurch, 2016-05-26:
Yup, there's also #276 and #314 and #315 
And I made some "comments about Motor classes" here: https://github.com/RPi-Distro/python-gpiozero/pull/277#issuecomment-211608633

goloplo, 2016-05-26:
Yes, there is L293D DC motor driver.
Keep up the good work.

lurch, 2016-05-27:
I've just had a look through the L293D datasheet, and AFAICT it'll be handled just fine by the current `Motor` class (assuming you hard-wire the EN pin high).

lurch, 2016-06-08:
And #366