gpiozero/gpiozero #337: Slicing LEDBoard

bennuttall, 2016-05-27:
It seems the LEDBoard `.leds` property is now a generator, rather than a list. Great for iterating over but I have to cast it to a list to be able to slice it:

```python
from gpiozero import PiLiter

lite = PiLiter()

for led in list(lite.leds)[::2]:
    led.on()
```

Is there a better way?

lurch, 2016-05-28:
Hmmm, given that neither CompositeDevices nor CompositeOutputDevices (and therefore LEDBoards) can have subordinate devices added or removed (or changed) post-construction, I guess there's no reason that the list-of-leds couldn't be generated just once during construction, stored as a tuple, and then returned from the `leds` property? @waveform80 ?

@bennuttall Another way of doing what you're doing above would be:
```python
from gpiozero import PiLiter

lite = PiLiter()

lite.on(0, 2, 4, 6)
```
Or the last line could be replaced with `lite.on(*range(len(lite))[::2])` if you wanted to make it more generic.

Or given that PiLiter doesn't contain any LEDCollection sub-instances (unlike SnowPi, which does), you could probably do:
```python
from gpiozero import PiLiter

lite = PiLiter()

for led in lite[::2]:
    led.on()
```

waveform80, 2016-05-28:
Yeah, with could just construct it as a tuple after the super call in `LEDCollection.__init__` then serve it as a static property

closed 2016-05-28 by bennuttall

bennuttall, 2016-05-28:
That's really cool, I had no idea these composite devices worked as iterators. That's much neater than accessing via `lite.leds`.

I'm pleased about #338 but the object itself being an iterator is much handier.

bennuttall, 2016-05-28:
```python
In [1]: from gpiozero import PiLiter

In [2]: lite = PiLiter()

In [3]: list(lite)
Out[3]: 
[<gpiozero.LED object on pin GPIO4, active_high=True, is_active=False>,
 <gpiozero.LED object on pin GPIO17, active_high=True, is_active=False>,
 <gpiozero.LED object on pin GPIO27, active_high=True, is_active=False>,
 <gpiozero.LED object on pin GPIO18, active_high=True, is_active=False>,
 <gpiozero.LED object on pin GPIO22, active_high=True, is_active=False>,
 <gpiozero.LED object on pin GPIO23, active_high=True, is_active=False>,
 <gpiozero.LED object on pin GPIO24, active_high=True, is_active=False>,
 <gpiozero.LED object on pin GPIO25, active_high=True, is_active=False>]

In [4]: lite[::]
Out[4]: 
(<gpiozero.LED object on pin GPIO4, active_high=True, is_active=False>,
 <gpiozero.LED object on pin GPIO17, active_high=True, is_active=False>,
 <gpiozero.LED object on pin GPIO27, active_high=True, is_active=False>,
 <gpiozero.LED object on pin GPIO18, active_high=True, is_active=False>,
 <gpiozero.LED object on pin GPIO22, active_high=True, is_active=False>,
 <gpiozero.LED object on pin GPIO23, active_high=True, is_active=False>,
 <gpiozero.LED object on pin GPIO24, active_high=True, is_active=False>,
 <gpiozero.LED object on pin GPIO25, active_high=True, is_active=False>)

In [5]: lite[::2]
Out[5]: 
(<gpiozero.LED object on pin GPIO4, active_high=True, is_active=False>,
 <gpiozero.LED object on pin GPIO27, active_high=True, is_active=False>,
 <gpiozero.LED object on pin GPIO22, active_high=True, is_active=False>,
 <gpiozero.LED object on pin GPIO24, active_high=True, is_active=False>)
```

That's really neat.

bennuttall, 2016-05-28:
So which composite devices work like this, and which don't?

lurch, 2016-05-28:
They all do. But for `LEDCollection` subclasses only, there's a special `leds` property which returns a 'flattened' view of the LEDCollection device hierarchy (since we know it can only contain LEDs or other LEDCollections).

```python
>>> from gpiozero import PiLiter, SnowPi, FishDish
>>> from gpiozero.pins.mock import MockPin
>>> import gpiozero.devices
>>> gpiozero.devices.pin_factory = MockPin
>>> lite = PiLiter()
>>> lite
<gpiozero.PiLiter object containing 8 devices:  and 8 unnamed>
>>> lite[::]
(<gpiozero.LED object on pin MOCK4, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK17, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK27, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK18, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK22, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK23, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK24, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK25, active_high=True, is_active=False>)
>>> lite.leds
(<gpiozero.LED object on pin MOCK4, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK17, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK27, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK18, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK22, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK23, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK24, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK25, active_high=True, is_active=False>)
>>> lite.close()
>>> snow = SnowPi()
>>> snow
<gpiozero.SnowPi object containing 3 devices: eyes,nose,arms and 0 unnamed>
>>> snow[::]
(<gpiozero.LEDBoard object containing 2 devices: left,right and 0 unnamed>,
<gpiozero.LED object on pin MOCK25, active_high=True, is_active=False>,
<gpiozero.LEDBoard object containing 2 devices: left,right and 0 unnamed>)
>>> snow.leds
(<gpiozero.LED object on pin MOCK23, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK24, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK25, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK17, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK18, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK22, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK7, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK8, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK9, active_high=True, is_active=False>)
>>> snow.arms
<gpiozero.LEDBoard object containing 2 devices: left,right and 0 unnamed>
>>> snow.arms.leds
(<gpiozero.LED object on pin MOCK17, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK18, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK22, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK7, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK8, active_high=True, is_active=False>,
<gpiozero.LED object on pin MOCK9, active_high=True, is_active=False>)
>>> snow.close()
>>> fish = FishDish()
>>> fish
<gpiozero.FishDish object containing 3 devices: buzzer,button,lights and 0 unnamed>
>>> fish[::]
(<gpiozero.TrafficLights object containing 3 devices: green,red,amber and 0 unnamed>,
<gpiozero.Buzzer object on pin MOCK8, active_high=True, is_active=False>,
<gpiozero.Button object on pin MOCK7, pull_up=False, is_active=False>)
```

That @waveform80 is a clever chap ;-)

bennuttall, 2016-05-28:
That's cool. Your example isn't working for me on v1.2 (but works on master). I can create an `LED` object on `MockPin(2)` but can't create a `PiLiter()` by specifying `gpiozero.devices.pin_factory = MockPin`. Guess that got fixed since the release.

waveform80, 2016-05-29:
Yeah, before master pin_factory is DefaultPin (the name change is to make clear things like construction of remote pins by default with a function or lambda). The change to make composite devices into sequences was ... erm 1.2 I think? Can't remember off the top of my head but it's in the base class so yeah, it applies to everything.

Finally coming to a close on picamera 1.11 btw, so I'll start looking at the servo stuff next week with a bit of luck!

bennuttall, 2016-05-29:
Great!

PR #338 (2016-05-28): Change LEDBoard.leds to return a static tuple
fixes #337
commit 871c9268 (2016-05-28): Change LEDBoard.leds to return a static tuple

fixes #337