gpiozero/gpiozero #340: Add ButtonBoard

bennuttall, 2016-05-28:
This has been suggested in #131 and #133. It could come in handy.

I had a go at trying to add one last night but didn't get very far.

I imagine it would require implementing the following in `boards.py`:

```python
class CompositeInputDevice(ValuesMixin, CompositeDevice):
    pass

class ButtonBoard(CompositeInputDevice):
    pass
```

I imagine `ButtonBoard` would be the equivalent of `LEDCollection`. Or should it be `ButtonCollection`? I can't see a need for both `ButtonBoard` and `ButtonCollection`.

lurch, 2016-05-28:
I think the reason there's an `LEDCollection` is because there's both `LEDBoard` (which operates on tuples of values) and `LEDBarGraph` (which operates on floats).
Since we can't ensure that the user only presses specific buttons, there's no way to implement a `ButtonBarGraph`, so you're right that `ButtonCollection` isn't needed. All IMHO of course.

bennuttall, 2016-05-29:
I've made a prototype to demonstrate what the API might look like:

```python
from gpiozero import Button

class ButtonBoard(object):
    def __init__(self, *pins, pull_up=True):
        self._buttons = tuple(Button(pin, pull_up=pull_up) for pin in pins)
        self._when_pressed = None

    @property
    def buttons(self):
        return self._buttons

    @property
    def value(self):
        return tuple(button.is_pressed for button in self.buttons)

    @property
    def when_pressed(self):
        return self._when_pressed

    @when_pressed.setter
    def when_pressed(self, fn):
        for button in self.buttons:
            button.when_pressed = fn
        self._when_pressed = fn

    @property
    def pressed(self):
        return tuple(i for i, btn in enumerate(self.buttons) if btn.is_pressed)

    @property
    def pressed_count(self):
        return sum(self.value)

    def close(self):
        for button in self.buttons:
            button.close()
```

I haven't done anything with `CompositieDevice` as mentioned above, as I just want to try out the API. It works quite nicely, and it's made the proposed `DotsBoard` interface (#133) much simpler:

```python
class DotsBoard(ButtonBoard):
    def __init__(self):
        pins = tuple(set(range(27)) - set([2, 3]))
        super(DotsBoard, self).__init__(*pins)
```

`DotsBoard` would need more work but it's a start. I've even played around with making simple games for the board using pygame-zero, which is pretty good fun.

Any suggestions on the `ButtonBoard` API at this stage @waveform80 @lurch?

lurch, 2016-05-31:
I don't think there's a need for a specific `CompositeInputDevice` - the approach taken by most of gpiozero seems to be to assume that every pin / device / etc. is input-only, and then subclasses of those (e.g. CompositeOutputDevice as a subclass of CompositeDevice) add extra 'output' functionality. CompositeDevice already has ValuesMixin in its hierarchy.

It might be nice to add the other "button-ish" methods like `wait_for_press` but I guess that would need wait-on-multiple-events functionality, which is something that I've previously emailed @waveform80 about separately.
Perhaps there could be an `any_pressed` method? Maybe your `pressed` method should be renamed to `which_pressed` or something? Alternatively it might be more gpiozero-ish to not have a `pressed` method, but instead just use `value` as provided by `CompositeDevice` ? `bounce_time` should probably be added to your init-method?
(Hmm, looking again at `CompositeDevice` I guess my `any_pressed` suggestion would just be an alias of `is_active`)

Looks good so far though :+1: 

Maybe in future this could also be extended to handle [keypad matrix](https://duckduckgo.com/?q=keypad+matrix&t=canonical&iax=1&ia=images) devices ;-)

bennuttall, 2016-05-31:
> Perhaps there could be an `any_pressed` method?

I was thinking usage for this could simply be:

```python
any(buttons.pressed)
all(buttons.pressed)
```

> It might be nice to add the other "button-ish" methods like wait_for_press but I guess that would need wait-on-multiple-events functionality

Definitely. I just did a skeleton feature set really.

Multiple `wait_for_press` is interesting. I've done it a dumb way and just provided a setter for all buttons, though individual buttons could be overridden. Not entirely clear, so I'm open to alternative solutions.

> Maybe in future this could also be extended to handle keypad matrix devices ;-)

That would be cool.

dglaude, 2016-06-05:
While I know nothing about Button Board, those words make me think about Keypad Matrix.
Typically a 3*4 or 4*4 array of button that can be used to enter a PIN code or dial a phone number or enter an IP address.

One way of using that is by using 7 or 8 GPIO, half of them as input and the other half as output. And there are quiet a few guide on how to do it on the Pi.

That would be a nice addition and if gpiozero could hide the complexity of scanning the key, it would be great.

bennuttall, 2016-06-05:
Definitely a good idea.

lurch, 2016-06-05:
I opened a new issue #358 for discussing keypad matrix drivers.

lurch, 2016-06-06:
@bennuttall re: the CompositeDevice integration... I initially started down the same route that you did, by setting all the buttons' handlers at once in `@when_pressed.setter`, and extending it to `@when_released.setter` and `@when_held.setter`. Worked okay, but then I realised what you meant about 'individual buttons could be overridden' - it goes down the "two different views of reality" approach that @waveform80 is always cautioning against. In theory you could set a when_pressed handler on the whole button board, and then set each button's when_pressed handler to None, and reading the button board's when_pressed property would still give you the (now unused) handler.

So I had a bit of a think, and I eventually managed to come up with the "proper" solution - ButtonBoard itself now also inherits from HoldMixin (which in turn inherits from EventsMixin) which means each Button still has its own `when_pressed` (etc.) properties, and ButtonBoard itself has its *own* `when_pressed` (etc.) properties! The Button events get fired before the ButtonBoard events, and if both handlers are set then both will get called. Quite neat IMHO :-)

```python
class ButtonBoard(HoldMixin, CompositeDevice):
    """
    Extends :class:`CompositeDevice` and represents a generic button board or
    collection of buttons.

    :param int \*pins:
        Specify the GPIO pins that the buttons of the board are attached to.
        You can designate as many pins as necessary.

    :param bool pull_up:
        If ``True`` (the default), the GPIO pins will be pulled high by
        default. In this case, connect the other side of the buttons to
        ground. If ``False``, the GPIO pins will be pulled low by default. In
        this case, connect the other side of the buttons to 3V3. This
        parameter can only be specified as a keyword parameter.

    :param float bounce_time:
        If ``None`` (the default), no software bounce compensation will be
        performed. Otherwise, this is the length of time (in seconds) that the
        buttons will ignore changes in state after an initial change. This
        parameter can only be specified as a keyword parameter.

    :param float hold_time:
        The length of time (in seconds) to wait after any button is pushed,
        until executing the :attr:`when_held` handler. Defaults to ``1``. This
        parameter can only be specified as a keyword parameter.

    :param bool hold_repeat:
        If ``True``, the :attr:`when_held` handler will be repeatedly executed
        as long as any buttons remain held, every *hold_time* seconds. If
        ``False`` (the default) the :attr:`when_held` handler will be only be
        executed once per hold. This parameter can only be specified as a
        keyword parameter.

    :param \*\*named_pins:
        Specify GPIO pins that buttons of the board are attached to,
        associating each button with a property name. You can designate as
        many pins as necessary and use any names, provided they're not already
        in use by something else.
    """
    def __init__(self, *args, **kwargs):
        pull_up = kwargs.pop('pull_up', True)
        bounce_time = kwargs.pop('bounce_time', None)
        hold_time = kwargs.pop('hold_time', 1)
        hold_repeat = kwargs.pop('hold_repeat', False)
        order = kwargs.pop('_order', None)
        super(ButtonBoard, self).__init__(
            *(
                Button(pin, pull_up, bounce_time, hold_time, hold_repeat)
                for pin in args
                ),
            _order=order,
            **{
                name: Button(pin, pull_up, bounce_time, hold_time, hold_repeat)
                for name, pin in kwargs.items()
                })
        def get_new_handler(device):
            def fire_both_events():
                device._fire_events()
                self._fire_events()
            return fire_both_events
        for button in self:
            button.pin.when_changed = get_new_handler(button)
        # Call _fire_events once to set initial state of events
        self._fire_events()
[... 54 lines cut ...]
>>> bb[2].pin.drive_low()
buttonboard pressed!
>>> bb[2].pin.drive_high()
buttonboard released!
>>> bb.value
ButtonBoardValue(left=False, down=False, right=False, up=False)
>>> any(bb.value)
False
>>> bb[0].pin.drive_low()
button1 pressed!
buttonboard pressed!
>>> bb.down.pin.drive_low()
button2 pressed!
>>> bb.value
ButtonBoardValue(left=True, down=True, right=False, up=False)
>>> sum(bb.value)
2
>>> bb[0].pin.drive_high()
button1 released!
>>> bb.down.pin.drive_high()
button2 released!
buttonboard released!
```

Have a play, let me know if it's the sort of thing you're after, and I can fix up the rest of the docs, write some unit-tests, and PR it. I haven't tested it on real hardware (so please do so!), but I believe you should be able to do `bb.wait_for_press()` too.

lurch, 2016-06-06:
Just occurred to me that we could even do something like:
```python
class Joystick(ButtonBoard):
    def __init__(self, up, right, down, left, pull_up=True, bounce_time=None, hold_time=1, hold_repeat=False):
        super(Joystick, self).__init__(up=up, right=right, down=down, left=left, _order=('up', 'right', 'down', 'left'), pull_up=pull_up, bounce_time=bounce_time, hold_time=hold_time, hold_repeat=hold_repeat)
        self._directions = {
            self.namedtuple(True, False, False, False): 'Up',
            self.namedtuple(True, True, False, False): 'UpRight',
            self.namedtuple(True, False, False, True): 'UpLeft',
            self.namedtuple(False, False, True, False): 'Down',
            self.namedtuple(False, True, True, False): 'DownRight',
            self.namedtuple(False, False, True, True): 'DownLeft',
            self.namedtuple(False, True, False, False): 'Right',
            self.namedtuple(False, False, False, True): 'Left',
        }

    @property
    def direction(self):
        # ignore invalid directions
        return self._directions.get(self.value, None)
```

lurch, 2016-06-06:
Had additional thoughts - given that ButtonBoard has multiple values, and isn't simply on/off the way a single Button is, it might be useful to have a `when_changed` handler, that gets triggered every time a button is pressed or released; rather than only having the `when_pressed` and `when_released` handlers that get triggered when any button is pressed, or all buttons are released?
Here's a modified version of the ButtonBoard class:
```python
class ButtonBoard(HoldMixin, CompositeDevice):
    """
    Extends :class:`CompositeDevice` and represents a generic button board or
    collection of buttons.

    :param int \*pins:
        Specify the GPIO pins that the buttons of the board are attached to.
        You can designate as many pins as necessary.

    :param bool pull_up:
        If ``True`` (the default), the GPIO pins will be pulled high by
        default. In this case, connect the other side of the buttons to
        ground. If ``False``, the GPIO pins will be pulled low by default. In
        this case, connect the other side of the buttons to 3V3. This
        parameter can only be specified as a keyword parameter.

    :param float bounce_time:
        If ``None`` (the default), no software bounce compensation will be
        performed. Otherwise, this is the length of time (in seconds) that the
        buttons will ignore changes in state after an initial change. This
        parameter can only be specified as a keyword parameter.

    :param float hold_time:
        The length of time (in seconds) to wait after any button is pushed,
        until executing the :attr:`when_held` handler. Defaults to ``1``. This
        parameter can only be specified as a keyword parameter.

    :param bool hold_repeat:
        If ``True``, the :attr:`when_held` handler will be repeatedly executed
        as long as any buttons remain held, every *hold_time* seconds. If
        ``False`` (the default) the :attr:`when_held` handler will be only be
        executed once per hold. This parameter can only be specified as a
        keyword parameter.

    :param \*\*named_pins:
        Specify GPIO pins that buttons of the board are attached to,
        associating each button with a property name. You can designate as
        many pins as necessary and use any names, provided they're not already
        in use by something else.
    """
    def __init__(self, *args, **kwargs):
        pull_up = kwargs.pop('pull_up', True)
        bounce_time = kwargs.pop('bounce_time', None)
        hold_time = kwargs.pop('hold_time', 1)
        hold_repeat = kwargs.pop('hold_repeat', False)
        order = kwargs.pop('_order', None)
        super(ButtonBoard, self).__init__(
            *(
                Button(pin, pull_up, bounce_time, hold_time, hold_repeat)
                for pin in args
                ),
            _order=order,
            **{
                name: Button(pin, pull_up, bounce_time, hold_time, hold_repeat)
                for name, pin in kwargs.items()
                })
        def get_new_handler(device):
            def fire_both_events():
                device._fire_events()
                self._fire_events()
            return fire_both_events
        for button in self:
            button.pin.when_changed = get_new_handler(button)
        self._when_changed = None
        self._last_value = None
        # Call _fire_events once to set initial state of events
        self._fire_events()
[... 90 lines cut ...]
button1 pressed!
buttonboard pressed!
ButtonBoard value now: ButtonBoardValue(left=True, down=False, right=False, up=False)
>>> bb.down.pin.drive_low()
button2 pressed!
ButtonBoard value now: ButtonBoardValue(left=True, down=True, right=False, up=False)
>>> bb.value
ButtonBoardValue(left=True, down=True, right=False, up=False)
>>> sum(bb.value)
2
>>> bb[0].pin.drive_high()
button1 released!
ButtonBoard value now: ButtonBoardValue(left=False, down=True, right=False, up=False)
>>> bb.down.pin.drive_high()
button2 released!
buttonboard released!
ButtonBoard value now: ButtonBoardValue(left=False, down=False, right=False, up=False)
>>> bb.down.pin.drive_high()
>>> 
```

Obviously this is now getting quite complex, so I'd rather have some feedback from @waveform80 before PRing it ;-)
Is it worth creating a separate Mixin for this `when_changed` property, or is it unlikely to be useful elsewhere, in which case it should just be left in ButtonBoard?

And following on from the ideas in #261, maybe it'd be useful for the `when_changed` callback to optionally be able to be told *which* button(s) changed their value?

lurch, 2016-06-06:
...and then the `when_changed` handler allows my earlier Joystick subclass to be exercised as follows:
```python
>>> joy = Joystick(10,11,12,13)
>>> joy.when_changed = lambda device: print(device.direction)
>>> joy.up.pin.drive_low()
Up
>>> joy.up.pin.drive_high()
None
>>> joy.right.pin.drive_low()
Right
>>> joy.left.pin.drive_low()
None
>>> joy.down.pin.drive_low()
None
>>> joy.right.pin.drive_high()
DownLeft
```

waveform80, 2016-08-14:
Sorry, just catching on all this, so here's some replies to everything I've missed so far:

> I think the reason there's an LEDCollection is because there's both LEDBoard (which operates on tuples of values) and LEDBarGraph (which operates on floats).
> Since we can't ensure that the user only [...]

Yup, precisely.

> I was thinking usage for this could simply be:
>
> ```python
> any(buttons.pressed)
> all(buttons.pressed)
> ```

Love the use of the universal and existential quantifier functions - but I do think @lurch has a point about simply calling it `value`:

> Alternatively it might be more gpiozero-ish to not have a `pressed` method, but instead just use `value` as provided by `CompositeDevice` ?

It'll feed nicely into `values` and so on, and so forth.

> (Hmm, looking again at CompositeDevice I guess my `any_pressed` suggestion would just be an alias of `is_active`)

That too.

> So I had a bit of a think, and I eventually managed to come up with the "proper" solution - ButtonBoard itself now also inherits from HoldMixin (which in turn inherits from EventsMixin) which means [...]

Yup, that definitely qualifies as neat - honestly could not have done it better.

> \[Joystick class\]

Nice! I'd probably just include the neutral value (False, False, False, False) in the dict and just use a straight lookup rather than the `get` method, but I love the idea.

> Is it worth creating a separate Mixin for this `when_changed` property, or is it unlikely to be useful elsewhere, in which case it should just be left in `ButtonBoard`?

Erm, not sure off the top of my head. For now I'd just leave it in `ButtonBoard` and if it turns out to have use elsewhere we can always pull it out into a mixin later (that's what I did with `Button.when_held` - I initially implemented just on Button and then realized it ought to be factored out - it's just that PR took so long no one saw my messy intermediate steps)!

Anyway, I'm going to have a quick play with this on some actual hardware (just because I want to ;) and then I'll merge it

waveform80, 2016-08-14:
Yup, this is great stuff. I'm happy to merge this now if you want, even if it reduces our coverage a bit - or I can hold off until you've got unit tests done. I can handle docs if you want (you've done the lion's share of the work by adding a doc-string for the class - all that really needs adding for now is an autoclass entry in `api_boards.rst` and a few more nodes in the class graphs - recipes would be nice, but not essential - they can always be added later).

lurch, 2016-08-14:
I'm glad you like what I managed to come up with :-)
Now that I know you're happy with the code side, I'll try to spend some time next week polishing up the other aspects of #370

waveform80, 2016-08-30:
#370 is merged and I've just merged some basic docs stuff, so I'll close this for now (want to concentrate on separating out stuff that's being pushed to 1.4)

closed 2016-08-30 by waveform80