gpiozero/gpiozero #346: Import Error: bad magic number in 'gpiozero' : b'\x03\xf3\r\n'

jayxonlim85, 2016-05-30:
Hi,
I tried run gpio zero code in Pi3 and it return this error:

**File "test.py" , line 1, in <module>
     from gpiozero import LED
Import Error: bad magic number in 'gpiozero' : b'\x03\xf3\r\n'** 

What did i miss?
Thank you

bennuttall, 2016-05-30:
Can you confirm you do not have a file in your working directory named `gpiozero.py` (and if you deleted one, please also delete `gpiozero.pyc`)? This is likely to be the cause of your issue.

If not, can you please provide the following information:

- How did you install gpiozero? apt-get? pip?
- Which Python version are you using? 2 or 3?
- Are you using Raspbian Jessie or Wheezy?

Can you also try using gpiozero in a Python shell (Type `python` at the command line and then `import gpiozero` to see if this causes the same error)

Thanks

lurch, 2016-05-31:
Looking at a few of the results in https://duckduckgo.com/?q=Import+Error%3A+bad+magic+number+in+%27gpiozero%27+%3A+b%27%5Cx03%5Cxf3%5Cr%5Cn%27&t=canonical&ia=web it sounds like you've got some stale .pyc file somewhere that needs to be deleted?

jayxonlim85, 2016-05-31:
Hi, thanks for the solution, i removed the gpiozero.pyc and now is no error prompt.
I use Raspbian Jessie and i use apt-get install python3-gpiozero python-gpiozero.
I run sudo python 3 test.py in terminal shell to run the code.
Thank you

bennuttall, 2016-05-31:
It's working now then?

I wonder if we should have a common errors page in the docs?

lurch, 2016-05-31:
I'm not sure how 'common' this error would actually be?

closed 2016-06-02 by bennuttall

waveform80, 2016-06-10:
Yeah, this generally only happens when you're doing naughty things with venvs (like switching their python revision without rebuilding them), or trying to use the same installation with multiple python vers.