gpiozero/gpiozero #347: Should 'Pin Numbering' be moved from Recipes to Notes?

bennuttall, 2016-05-30:
@waveform80 any reason the note on pin numbering is on the recipes page?

lurch, 2016-05-30:
It used to be on the Notes page, before being moved to the Recipes page (which means there's a bunch of broken references to it scattered throughout the rest of the docs, as I noted in #208 ).

I guess it depends if we're using the Recipes page as a kind of "start here first" ?  \*shrug\*

waveform80, 2016-06-04:
Yeah, I deliberately shoved recipes up front on the basis of experience from picamera (recipes is hit *vastly* more than anything else on there - so it is indeed intended to be a "start here first" section. I really should fix up those notes references...

lurch, 2016-08-31:
I guess 'Pin Numbering' is now going to stay on the Recipes page for good?
@waveform80 has also fixed all the broken references now.

closed 2016-08-31 by lurch

waveform80, 2016-09-01:
Finally remembered to fix up all those references (found one when merging servos and just got on with tidying up the rest). I don't really mind where the pin numbering goes provided it's prominent and prior to recipes. Recipes is where everybody heads by default and given that recipes involve wiring we need to make sure everyone's aware of our pin numbering policy before they start wiring stuff onto the Pi - hence the positioning. Notes is probably going to wind up as a sort of FAQ like thing (might rename it that at some point) as that's the first place people tend to head when they've tried a recipe and "things don't work".