gpiozero/gpiozero #350: v1.3 release

bennuttall, 2016-06-02:
I'm trying to decide what should make the v1.3 release.

Here's what I think should be included:

- #340 ButtonBoard
- #133 DotsBoard
- #248 Servo
- #93 Temperature sensor
- #294 CPU temperature
- #329 ADC.voltage

Documentation and house-keeping:

- #87 Test suite
- #261 activated/deactivated callback modification
- #289 Add note about remote access
- #345 Alias amber to yellow

Drop milestone, leave on wishlist:

- #144 Stepper motor
- #169 Piezo speaker
- #199 GPIO expander chips
- #228 RGBLED.cycle()
- #230 Compositing robots with sensors

(and more in this category that don't have the milestone)

I'm open to suggestions on what should be in the release - especially if any I've missed out are low-hanging fruit (maybe #228?), or if any I included which require too much work (maybe #93). I just want to nail the list down and start working towards a release.

There have been 100 commits since v1.2 was released! It would be good to get that functionality, and those docs fixes, out in the wild. It's hard to determine how big a step is required to call it a new release. We could easily do more regular releases - I don't know if that's something we want to do or not.

dglaude, 2016-07-13:
Maybe you could add to the list #359 pin_factory based on environmental value

I provided the Pull request #356 and showed the result of my tests on a Mac talking to a Pi and on a Pi.

What else:
-* Is there anything expected from me ?
-* Is there someone to review the code ?
-* Are we waiting for another/better implementation by @waveform80 or somebody else (without if / elseif sequence) ?
-* Should I redo the Pull request based on current version so that it has no conflict?

Please advise as it is one month old already and soon I will have forgot everything.

Thanks.

waveform80, 2016-07-14:
The PR looks good from a quick skim - I'm just rather swamped under with another project at the moment. I should be able to get back to GPIO Zero next week, once this current project is cleared (also managed to get picamera 1.11, and then 1.12 out recently so other than GPIO Zero my todo list is actually shrinking for once!)

waveform80, 2016-08-30:
Raspbian freeze coming up tomorrow, so 1.3 will be going out sometime tonight. I'll be bumping everything not yet finished to the 1.4 milestone in a bit.

lurch, 2016-08-31:
Next time, would it be possible to have more advance notice of the upcoming new-release, so we get a bit more time to work on finishing off / polishing incomplete features?

lurch, 2016-08-31:
Just 'ticked off' the items in the first post where the corresponding issue or PR has been closed / merged.

waveform80, 2016-08-31:
Yeah, this was a bit dis-organised - it's partly due to the Raspbian freeze today, and partly wanting to get another release out we can show-case at pycon-uk. I was hoping to have more time over the long weekend to sort things out, but as always other things got in the way. Sorry about that!

Maybe we should set a date on the 1.4 milestone and try sticking to it, i.e. even if everything's not complete just release anyway so at least users know when to expect a release and devs know what the release deadline'll actually be. Unfortunately that may lead to a rather spartan release if we select something too short (1.3 would've been rather bare if we'd released on the monthly cycle Ben originally proposed), but a quarterly cycle might be more realistic?

lurch, 2016-08-31:
Yeah, I guess a quarterly cycle with a known-in-advance release date might work :)

bennuttall, 2016-08-31:
I think 3 months should work. Shall we say 16th November (my birthday, as that's when we released v1.0)?

waveform80, 2016-08-31:
Sounds good

lurch, 2016-08-31:
I guess this issue should be closed now, and @bennuttall should create a new "v1.4 release" issue? ;)

bennuttall, 2016-08-31:
In progress!

closed 2016-08-31 by bennuttall