gpiozero/gpiozero #351: Has Energenie support been tested on RPi3?

DeeJay, 2016-06-02:
This is less an 'issue', more a 'heads-up' and pointer to a Raspi forum thread. I don't have a an RPi3 to check this.

Two users in this [https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=145161&hilit=energenie] thread appear to be having difficulty using Energenie devices with an RPi3.

A direct suggestion that they report the issue here was not followed up, so perhaps it has been resolved. Just thought you might like the opportunity to verify this prior to the release of v1.3.

lurch, 2016-06-02:
I've just read that thread - I also can't see why the Pi3 would be any different.
The only thing that comes to mind is: I wonder if the users have two pimote boards, and they've paired the sockets to the pimote attached to their Pi1, and are then confused when it doesn't work with the pimote attached to their Pi3?

AIUI the cheap 433MHz sockets (e.g. sold in Maplin) use a fixed number of codes (something like three different codes on each of four different channels) - so any transmitter can control any socket (which can obviously lead to undesired interference if both you and your neighbour are trying to control a socket with the same code and channel!). But I believe with Energenie sockets, each transmitter has a unique ID, and you need to pair each socket to a specific transmitter (by getting the socket to 'learn' the ID of the transmitter). This prevents interference, but it also prevents two different pimotes controlling the same socket. AFAIK.

Just a guess... ;)

waveform80, 2016-06-04:
Well, that was fortuitous: just saw this and realized I was on a Pi3 (working on picamera), with an Energenie board sat in a drawer next to me! I'll just test it ...

Yup, works fine. As observed in the thread: the older Energenie boards are trivially controlled by 6 GPIO pins so there's no reason they shouldn't work on a Pi3 (unless the board isn't quite seated properly, etc.?). There is a newer generation of sockets (MiHome or something like that?) which are more complicated (uses SPI or I2C - can't remember which off the top of my head, but I trawled through their C code a while back in anticipation of adding it to GPIO Zero one day) but again that should work fine with a Pi3.

closed 2017-07-27 by bennuttall