gpiozero/gpiozero #353: Better pigpio support

dglaude, 2016-06-03:
I believe the pigpio support should respect pigpio environmental variable.

Current definition of PiGPIOPin function default to host='localhost' and port=8888.

PIGPIO Python library way of working is to check the environment variable PIGPIO_ADDR/PIGPIO_PORT

If any of those is set, it is also a clear indication that PIGPIO is available... and how to access those PINs

I also believe pigpio should automatically be used if local GPIO are not found and any of those environmental value is found.

The goal is to have the same code working whatever if the pigpiod is running locally or remotely.
Actually transparent use of pigpiod is the ultimate goal, this mean without having to specify PiGPIOPin.

See below for an experiment:

Please note that this code is working when changing the led and button definition the following way:
```
amber = LED(PiGPIOPin(22,'slave.local'))
button = Button(PiGPIOPin(7,'slave.local'), pull_up=False)
```

The experiment setup
- All command typed from a Mac
- PiZero connected with USB in Ethernet Gadget mode (hostname is 'slave')
- pigpiod started as root on the PiZero
- FishDish connected to the PiZero

**dglaude$** env | grep PIGPIO
PIGPIO_ADDR=slave.local
**dglaude$** cat gpiozero.pigpio.FishDish.py 
```
from gpiozero.pins.pigpiod import PiGPIOPin
from gpiozero import LED, Button
from signal import pause

amber = LED(PiGPIOPin(22))
button = Button(PiGPIOPin(7), pull_up=False)

button.when_pressed = amber.on
button.when_released = amber.off

pause()
```
**dglaude$** python gpiozero.pigpio.FishDish.py 
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Can't connect to pigpio at localhost(8888)

Did you start the pigpio daemon? E.g. sudo pigpiod

Did you specify the correct Pi host/port in the environment
variables PIGPIO_ADDR/PIGPIO_PORT?
E.g. export PIGPIO_ADDR=soft, export PIGPIO_PORT=8888

Did you specify the correct Pi host/port in the
pigpio.pi() function? E.g. pigpio.pi('soft', 8888))
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Traceback (most recent call last):
  File "gpiozero.pigpio.FishDish.py", line 5, in <module>
    amber = LED(PiGPIOPin(22))
  File "build/bdist.macosx-10.11-intel/egg/gpiozero/pins/pigpiod.py", line 110, in __new__
  File "/Library/Python/2.7/site-packages/pigpio.py", line 1760, in get_hardware_revision
    return _pigpio_command(self.sl, _PI_CMD_HWVER, 0, 0)
  File "/Library/Python/2.7/site-packages/pigpio.py", line 848, in _pigpio_command
    sl.s.send(struct.pack('IIII', cmd, p1, p2, 0))
AttributeError: 'NoneType' object has no attribute 'send'

dglaude, 2016-06-03:
Just to be clear, this Python script is running on my Mac and the FishDish is connected to the PiZero.
gpiozero and pigpio is installed on my Mac.
pigpiod is running on the PiZero.

So I have Raspberry Pi compatible GPIO "on my Mac"... without the "RTK.GPIO, a USB Raspberry Pi Compatible GPIO Header" for the price of a PiZero (with soldered pin), a microSD card and a USB cable.

I plan to document all of this so that it can be reproduced by many.

bennuttall, 2016-06-03:
Thanks for the input. We definitely want to improve remote pin support for exactly this reason.

Are you using the latest release, or the up-to-date version from github?

dglaude, 2016-06-03:
Git was not installed on my mac (yet), so I downloaded https://codeload.github.com/RPi-Distro/python-gpiozero/zip/master from https://github.com/RPi-Distro/python-gpiozero at 22:19 CET.
Then **unzip python-gpiozero-master.zip** and later **sudo python setup.py install**

I assume this is the latest from the master branch.

bennuttall, 2016-06-03:
Yes

dglaude, 2016-06-04:
Code that you could borrow from https://github.com/joan2937/pigpio/blob/master/pigpio.py around line 3800:

```

   def __init__(self,
                host = os.getenv("PIGPIO_ADDR", ''),
                port = os.getenv("PIGPIO_PORT", 8888)):
      """
      Grants access to a Pi's GPIO.
      host:= the host name of the Pi on which the pigpio daemon is
             running.  The default is localhost unless overridden by
             the PIGPIO_ADDR environment variable.
       
      port:= the port number on which the pigpio daemon is listening.
             The default is 8888 unless overridden by the PIGPIO_PORT
             environment variable.  The pigpio daemon must have been
             started with the same port number.
```

What is the best way to contribute?
Clone, make my change, test then Pull request?

lurch, 2016-06-05:
> What is the best way to contribute?
Clone, make my change, test then Pull request?

Yep

dglaude, 2016-06-05:
"pigipod"
Is this a typo and should it be "pigpiod"?

devices.py:46:            from .pins.pigipod import PiGPIOPin

(I have no clue about how import/from works in python and what starting with "." make, but this seems like a typo. There is a "pins" sub-folder and within a file "pigpiod.py". I assume this is the one to be referenced)

bennuttall, 2016-06-05:
Good spot! Fixed in 8b81057f8b9ac020d763300c386819ee398b4a1c

That would be hard to find as the typo would have caused an `ImportError`, but it was in a `try` / `except ImportError` block, so it would always fall back to `NativePin` in that case.

lurch, 2016-06-05:
The "." at the start means it's a relative import https://docs.python.org/2/tutorial/modules.html#intra-package-references

dglaude, 2016-06-05:
After fixing the typo, I reused the pigpio way of reading the environment variable for the ADDR and PORT. (I hope both change are in the Pull request.

```
def __new__(cls, number, host = os.getenv("PIGPIO_ADDR", 'localhost'), port = os.getenv("PIGPIO_PORT", 8888) ):
```

As a consequence, the following 3 tests program give exactly the same result while run on my MAC talking to a PiZero over USB/Ethernet. The last example is not even making a reference to pigpio, it is written the same way as running locally on the Pi.

PS1: I am a Python newbie and have not done serious programming since years... so please double check and feel free to do it another way.

PS2: Maybe the documentation should be updated too.

$ **env | grep GPIO**
PIGPIO_ADDR=slave.local

$ **cat gpiozero.pigpio_explicit.py** 
```
from gpiozero.pins.pigpiod import PiGPIOPin
from gpiozero import LED, Button
from signal import pause

amber = LED(PiGPIOPin(22,'slave.local'))
button = Button(PiGPIOPin(7,'slave.local'), pull_up=False)

button.when_pressed = amber.on
button.when_released = amber.off

pause()
```
$ **cat gpiozero.pigpio_implicit.py** 
```
from gpiozero.pins.pigpiod import PiGPIOPin
from gpiozero import LED, Button
from signal import pause

amber = LED(PiGPIOPin(22))
button = Button(PiGPIOPin(7), pull_up=False)

button.when_pressed = amber.on
button.when_released = amber.off

pause()
```
$ **cat gpiozero.pigpio_nosay.py** 
```
from gpiozero import LED, Button
from signal import pause

amber = LED(22)
button = Button(7, pull_up=False)

button.when_pressed = amber.on
button.when_released = amber.off

pause()
```

lurch, 2016-06-05:
See also #279 with regards to `pin_factory` (github version) or `DefaultPin` (released 'apt-get' v1.2.0 version).

I believe your last example (`gpiozero.pigpio_nosay.py`) only works by virtue of RPi.GPIO not being installed on your Mac - so you couldn't run the same script on a Pi (which has RPi.GPIO installed by default) to control a different Pi. IMHO it'd be better to be explicit by setting `pin_factory` to `PiGPIOPin`.

dglaude, 2016-06-05:
Yes my `gpiozero.pigpio_nosay.py` work by pure magic and it even surprised me.

But it answer my ultimate goal that was to run the same piece of code locally or remotely.
This require some external way to influence the fallback mechanism or access method to GPIO.

That is the elegant solution used by pigpio where you define in the environment how to connect to the pigpiod of your Pi (if nothing is explicitly said in the code).

That same trick could/should be used into gpiozero and provide a way to externally choose the pin implementation. Currently it is hard wired starting from line 32 of `devices.py`. It could be softwired with an single environmental variable such as `GPIOZERO_PIN_FACTORY` that could be not set for current behaviour or set to `RPiGPIOPin` or `RPIOPin` or `PiGPIOin` or `NativePin`.

Should I create a new issue **write once run everywhere** for this?

This also permit better testing of alternative pin implementation, otherwise 99% will only use RPiGPIOPin. Some developper could use another one, and when you encounter an issue, you can test with other pin factory to verify where the problem is.

Please notice that I don't try to access an arbitrary set of GPIO, but a Pi structured set of GPIO (since it is a Pi). `pin_factory` still make a lot of sense. That is also what I expect and would like pi_info to work for a remote Pi. ( see #278 closed but that seems only partially solved to me )

dglaude, 2016-06-05:
Find below a pin_factory version of my test code.
This works on my Mac, and I guess it should work on a Pi connected to a Pi.
But it less sexy than the same code running everywhere.

$ **env | grep PIGPIO**
PIGPIO_ADDR=slave.local
$ **cat gpiozero.pigpio_pin_factory.py** 
```
from gpiozero.pins.pigpiod import PiGPIOPin
import gpiozero.devices

from gpiozero import LED, Button
from signal import pause

gpiozero.pin_factory = PiGPIOPin

amber = LED(22)
button = Button(7, pull_up=False)

button.when_pressed = amber.on
button.when_released = amber.off

pause()
```

lurch, 2016-06-06:
> It could be softwired with an single environmental variable such as GPIOZERO_PIN_FACTORY that could be not set for current behaviour or set to RPiGPIOPin or RPIOPin or PiGPIOin or NativePin.

Interesting suggestion! Yeah, it's probably worth creating a separate issue for discussing it further.

`gpiozero.pin_factory = PiGPIOPin` in your latest example should actually be `gpiozero.devices.pin_factory = PiGPIOPin`
(Assigning to `gpiozero.pin_factory` doesn't error, because it's silently creating a new `pin_factory` attribute on the `gpiozero` module. Whereas what you _want_ to do is set a new value for the existing `gpiozero.devices.pin_factory` attribute.)

waveform80, 2016-06-10:
As mentioned on #278 I'm definitely in favour of this going in - the straight forward `os.getenv` (or `os.environ.get` - whatever people are more comfortable with) is absolutely fine.

waveform80, 2016-08-29:
Okay, as of 97de7e973e3aa0a830f02048ad814795e325ef73 we now support the `GPIOZERO_PIN_FACTORY`, `PIGPIO_ADDR`, and `PIGPIO_PORT` environment vars, `pi_info` works for local and remote pi's, and we've got decent extensibility for any third party pin implementations. I think that pretty much covers everything mentioned here, so I'll close this for now (feel free to re-open if I've missed anything!)

closed 2016-08-29 by waveform80