gpiozero/gpiozero #354: pinout function

bennuttall, 2016-06-04:
Just playing around with `pi_info()` and wondered about adding a pinout function to show the pinout for the current Pi. For example:

```python
> def pinout():
> ... info = pi_info()
> ... pins = info.headers['P1']
> ... for i in range(1, len(pins), 2):
> ...     print(pins[i].function, pins[i+1].function)
> ...
> pinout()
```
```
3V3 5V
GPIO0 5V
GPIO1 GND
GPIO4 GPIO14
GND GPIO15
GPIO17 GPIO18
GPIO21 GND
GPIO22 GPIO23
3V3 GPIO24
GPIO10 GND
GPIO9 GPIO25
GPIO11 GPIO8
GND GPIO7
```

(yes, I'm on a Pi 1 rev1)

It just needs padding out. And I'm sure @waveform80 would have a better way of pairing them.

Any thoughts?

bennuttall, 2016-06-04:
```python
pins = info.headers['P1']
pin_pairs = zip(pins[::2], pins[1::2])
for pair in pin_pairs:
    print(' '.join(pin.function for pin in pair))
```

bennuttall, 2016-06-04:
Got it...

```python
def pinout():
        info = pi_info()
        pins = list(info.headers['P1'].values())
        pin_pairs = zip(pins[::2], pins[1::2])
        for pair in pin_pairs:
            l, r = (pin.function for pin in pair)
            print('{:>7} {:7}'.format(l, r))
```
```
    3V3 5V     
  GPIO2 5V     
  GPIO3 GND    
  GPIO4 GPIO14 
    GND GPIO15 
 GPIO17 GPIO18 
 GPIO27 GND    
 GPIO22 GPIO23 
    3V3 GPIO24 
 GPIO10 GND    
  GPIO9 GPIO25 
 GPIO11 GPIO8  
    GND GPIO7
```

dglaude, 2016-06-05:
Will that work with "Remote Pi"?

bennuttall, 2016-06-05:
I think `pi_info()` currently breaks when you're not running it on a Pi. It would be good to think about what `pinout()` should do in those circumstances. Maybe it should optionally take a Pi model argument, e.g. `pinout("zero")`

bennuttall, 2016-06-05:
Something like `if "GPIO4" in pinout()` might be handy (it would have to return the string rather than (or as well as) print. Or maybe another function returning a list of GPIOs would be better?

dglaude, 2016-06-05:
The remote pi and issue with pi_info is discussed here: #278
Not sure if and how it is fixed.
I can try 

You may want to look at this pigpio code for a way to output the pin status:
https://github.com/joan2937/pigpio/tree/master/EXAMPLES/Python/GPIO_STATUS

I did not find how pigpio can report the hardware it is running on... maybe this is a feature that should be added... Here is what I found:
http://abyz.co.uk/rpi/pigpio/index.html#GPIO

lurch, 2016-06-05:
> Or maybe another function returning a list of GPIOs would be better?

Yeah, probably. Otherwise `"GPIO2" in pinout()` would match both "GPIO21" and "GPIO25" (if `pinout` returned a string).

bennuttall, 2016-06-05:
Good catch

waveform80, 2016-08-14:
Hmm, pi_info's an interesting case. It takes an optional parameter specifying the revision of Pi you want information on (so you can use it retrieve info on any Pi potentially). However, if you don't specify this it attempts to look in /proc/cpuinfo for the revision of the local Pi.

So, what about remote pins? At the moment `_get_pi_revision` has no way of telling that remote pins are being used (other than introspecting pin_factory which would be horrid). However, #353 proposes using the pigpio environment vars. I'll try and merge relevant stuff from #353 today, and once that's done, _get_pi_revision should probably be enhanced to use that information (pigpio has functions for retrieving the revision from remote pi's - this is [already used](https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/pins/pigpiod.py#L111) in our PiGPIOPin implementation to get remote pin info, but it'd be nice to include this support in straight calls to pi_info without making the user jump through the hoops of getting the pi revision from pigpio directly.

waveform80, 2016-08-15:
After playing around with this for a while this evening, this is a little more complex than I thought. On the other hand, it's been useful to figure out the next steps in the pins API. Mostly for my own reference, here's my current thinking:

* There's two use-cases for `pi_info()` currently which are both handy and need to continue being supported:

  1. `pi_info(revision)` which returns information about the specific revision of Pi specified. At the moment `revision` is specified as a string like that found in `/proc/cpuinfo`
  2. `pi_info()` which returns information about "this" Pi

* First question: should `revision` remain as a string or should we simply use an integer (given they're all integers at the moment anyway, with the possible exception of the impossibly rare "Beta board")? I'm tempted to leave it as a string because firstly that preserves backward compatibility (so no pain), and secondly it future proofs us in case revision ever does change to include non-numeric components (like the aforementioned "beta").

* Second question: what does "this" Pi mean? In the case of RPi.GPIO that's easy: the Pi we're running on. In the case of pigpio, it's *much* harder. It could be the local Pi if PIGPIO_ADDR and PIGPIO_HOST are unset. It could still be the local Pi if they're "localhost" and 8888 (the defaults). It could be a remote Pi if localhost:8888 is, say, an SSH tunnel to a remote location (okay, I'm stretching but it's perfectly possible) and PIGPIO_ADDR/PIGPIO_PORT are still unset. And what if someone's changed the default pin_factory to be something else entirely and specified `host` and `port` manually in the PiGPIOPin constructor?

So, here's a solution: extend the pins API so that all `Pin` classes have a `pi_info` class method which returns the result of `pi_info` for whatever revision of Pi the class would construct for, if it were set as the default `pin_factory`. Then, `pi_info()` with no arguments simply returns `pin_factory.pi_info()`.

Advantages:

* Works exactly the same as the current `pi_info()` for all current (local) cases
* Works properly when the default `pin_factory` is PiGPIOPin, whether PIGPIO_ADDR or PIGPIO_PORT are set or not
* Works properly if `pin_factory` is set to some other class entirely

Disadvantages:

* Changes the definition of `pin_factory` from an arbitrary callable which takes a single mandatory parameter (the pin to construct) to a definite derivative of the abstract `Pin` class, and expands the Pins API further.

I'm not so bothered about the downsides here: the Pins API isn't marked stable and I've got a couple of expansions planned anyway (for remote SPI and I2C support), so this is a minor thing. As to changing what `pin_factory` means - I don't think there's many serious users of it yet, and the vast majority of use-cases for it (i.e. remote pins) become vastly easier now we support PIGPIO_ADDR / PIGPIO_PORT anyway (i.e. effectively removes those use-cases).

Note to self: most pin implementations are local; perhaps move `_get_pi_revision` to base `Pin` class; cache result in class attribute if so? Override in `PiGPIOPin` obviously and only cache with connection (not class level). Watch for circular refs (pins->pins.data->devices->pins ?)

Will see what I can do about implementing all this tomorrow.

lurch, 2016-08-30:
Did the `pinout` function that @bennuttall originally asked for actually get added? ;-)

waveform80, 2016-08-30:
Hmm, not exactly :) I'm not sure there's a perfect version that can be made at the moment - for example the headers on Pi's are typically 2-column ... except for the Compute Module. For now I'm happy enough we've got a nice [pin-out diagram](http://gpiozero.readthedocs.io/en/v1.2.0/recipes.html#pin-numbering) but I admit it would be nice to have something like this ...

bennuttall, 2016-09-05:
It would be good to get this function or similar added. I see your point re: compute module but I think it would be useful to beginners to have this.

bennuttall, 2016-09-05:
Would be good to provide `pinout` as a command line tool too.

joan2937, 2016-09-05:
pigpio does have the http://abyz.co.uk/rpi/pigpio/python.html#get_hardware_revision method.  It returns the bottom 7 nibbles of the Revision field of /proc/cpuinfo (converted from hex to decimal).  So you can tell which Pi an instance is running on.

lurch, 2016-09-06:
Technically, the Compute Module (i.e. the raw SODIMM) **is** two-column; it's only the CMIO board which is four-column (which has no representation in `pins/data.py`).

@joan2937 Yeah, @waveform80 already added support for that, see https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/pins/pigpiod.py#L267

waveform80, 2016-09-08:
@lurch true - although I envisaged that the pinout function's purpose is really to output the physical layout of the headers (okay ... again, the CM's "header", i.e. the SODIMM module, is two-column ... but I was more thinking of the pins that the user is likely to connect to).

Anyway, I take the point that the CM is a minority case at best. I'll add some layout info to the headers dict in pins/data and whip up a little print function for it (we should provide both a function for basic use, and the info itself for those that want to make something prettier like a GUI representation).

Actually - there's a quick question: would people be happier with an actual print function (printing to stdout) or would they prefer something that just returns the sequence of strings that *would* be printed so they can redirect them to, e.g. stderr, an HTML \<pre\> block, etc.? The columns info will be in `pi_info().headers` anyway but I imagine that'd be useful for some.

lurch, 2016-09-08:
Nah, for people who want to do anything fancy, there's always the data from https://github.com/gadgetoid/Pinout2 (the 'source' for http://pinout.xyz/ )  ;-)

I'm guessing Ben wanted to add this so that people could just open up a python REPL on their Pi's and do `pinout()`, in which case it makes sense for the function to print to STDOUT directly. (perhaps call the function `print_pinout()`, with it taking an optional revision (defaulting to the current Pi's revision) ?)

P.S. There's no guarantee that the CM *is* being used in a CMIO board, and not in some other random 3rd party carrier board. Perhaps the pinout() function should just print an error message (or a reference to e.g. http://elinux.org/RPi_BCM2835_GPIOs) if run on a Compute Module? Especially as it's likely to be more than a screenful of text.

P.P.S. Although I _have_ been wondering if it'd be worth adding another 'type' column to the "pins" data, that explicitly states whether a pin is "GPIO", "GND" or "POWER", to save having to check by string-matching against e.g. "GPIO17" ?

waveform80, 2016-09-09:
You want fancy? I admit I *may* have gone overboard with this (and yes, it's still string matching - sorry - I haven't come up with a good way of dealing with the CM's myriad pin types yet) ...

waveform80, 2016-09-09:
See PR #431 :)

waveform80, 2016-09-09:
Oh ... in case you want to test it:

```python
>>> from gpiozero import *
>>> pi_info().pprint()
>>> pi_info('000e').pprint()
>>> pi_info('900092').pprint()
>>> pi_info('0014').headers['SODIMM'].pprint()
```

etc.

lurch, 2016-09-10:
> You want fancy?

Oh my!  ;-0

waveform80, 2016-09-10:
Hmm, doesn't work in IDLE - need to make the term-color detection a tad more intelligent.

waveform80, 2016-09-11:
> Oh my! ;-0

I can't help but hear that in George Takei's voice!

closed 2016-09-13 by waveform80

lurch, 2016-09-13:
I've just noticed that while the examples in the [documentation](http://gpiozero.readthedocs.io/en/latest/api_pins.html#gpiozero.PiBoardInfo) work, if I instead do `print('{0}'.format(pi_info()))` I get `KeyError: 'pop from an empty set'`.

I've not really used Python's `format` function before, so I dunno if this is expected behaviour, or whether it'd be worth defaulting it to 'full' output if no other option is specified?

lurch, 2016-09-14:
This afternoon I received two 512MB A+ boards that I ordered from [ModMyPi](https://www.modmypi.com/raspberry-pi/model-a-plus/raspberry-pi-model-a-plus-new) at the weekend.
I was expecting them to come with a new-style revision code, but to my surprise they report a revision of '0015' (which [this thread](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=156834) confirms). At first I wondered if maybe [eLinux](http://elinux.org/RPi_HardwareHistory) was wrong in reporting the '0015' boards as only having 256MB, but [this](http://www.raspberrypi-spy.co.uk/2012/09/checking-your-raspberry-pi-board-version/#comment-148343) confirms that '0015' A+s have been around with 256MB for at least a year before the upgrade to 512MB.

So I was wondering if @bennuttall could ask the Raspberry Pi engineers what the best / most reliable way is to detect whether we're running on a '0015' A+ board with 256MB RAM, or 512MB RAM?

bennuttall, 2016-09-14:
@ghollingworth do you know?

ghollingworth, 2016-09-14:
Think you can use a combination of get_mem calls to find out...

```
root@raspberrypi:~# vcgencmd get_mem arm && vcgencmd get_mem gpu
arm=448M
gpu=64M
```

bennuttall, 2016-09-14:
Thanks.

If there's no way of telling from the board revision we'll have to dynamically populate according to a subprocess?

lurch, 2016-09-14:
> we'll have to dynamically populate according to a subprocess?

I wonder how that'll work with remote pigpio pins? @joan2937 ?

waveform80, 2016-09-14:
intriguing - I use my A+ specifically because it's got 256Mb (it was useful for ensuring the Sense Emulator would work at the bottom end as well as the top end on a 3B), and it's one of the 256Mb models with revision 0015.

Oh well, I'm almost tempted to just alter 0015 to 512Mb in our table and lie about a small portion of the Pi's out there; this strikes me as one of those minor things that's actually quite tricky to fix properly.

bennuttall, 2016-09-14:
I'm not sure why we do it in the first place :P

I'm pretty sure it started with me not allowing pull-down Buttons to be created on pins 2 and 3.

waveform80, 2016-09-14:
Yeah, it did ... and then I went overboard with it. Ooops.

bennuttall, 2016-09-14:
Indeed. This is not a huge deal. I think falling back to sensible values in ambiguous situations is fine, as long as there aren't times where gpiozero falls over because it can't work out pi_info.

ghollingworth, 2016-09-14:
Why do you need to know if the memory is 512, 256 or 128M though.  This really should be required for just running GPIO stuff...

lurch, 2016-09-14:
It's just @waveform80 being too much of a completionist ;-)

waveform80, 2016-09-14:
yup - my fault entirely - it's not required, but GPIO-zero's got a ... rather comprehensive pi_info function. Now including ASCII art. Ahem.

lurch, 2016-09-14:
It all dates back to #246  (with additional comments [here](https://github.com/RPi-Distro/python-gpiozero/issues/294#issuecomment-213144569))

commit 6cc308e4 (2016-08-29): Fix #354, fix #389

Overhaul the pi_info system:

Pin factories are now capable of generating pi_info themselves (although
currently they all just look up the revision and call pi_info with a
specific one).

PiGPIOPin will now return pi_info for the remote pi which can be
specified by parameter or implicitly by the environment vars.

Overvolted Pis should work properly no matter what (some argument over
whether the revision 7 or 8 chars in this case; both should work). Added
some minor tweaks for the new camera-capable Pi Zero

Finally, added a bunch of tests for pins.data
commit 97de7e97 (2016-08-29): Merge pull request #410 from waveform80/remote-pi-info

Fix #354, fix #389
commit 27c07030 (2016-09-13): Fix #354

Actually, "fix" doesn't really do this justice. This is closer to "nukes
from orbit" ...
commit 322ec131 (2016-09-13): Merge pull request #431 from waveform80/pinout

Fix #354