gpiozero/gpiozero #357: Multiplexed seven-segment LED driver

lurch, 2016-06-05:
Something I've been meaning to suggest for ages, but keep forgetting...

It would be great if gpiozero offered a simple API for seven-segment LEDs, like those featured on http://raspi.tv/2015/how-to-drive-a-7-segment-display-directly-on-raspberry-pi-in-python
It would need a similar background-update-thread to that needed to drive a keypad matrix as discussed in #340

bennuttall, 2016-06-05:
Yeah, I've thought about this too.

SrMouraSilva, 2016-10-02:
I implemented once.

My implementation supports more then one seven segments display, may be cathodes and anodes (mixed). My thread implementation maybe isn't good.

I will refactor to support more symbols and custom symbols then I inform in here.

SrMouraSilva, 2016-10-02:
https://github.com/PedalPi/Physical/issues/4

Code:
https://github.com/PedalPi/Physical/commit/eb53f689ee94b3a503a9225aac7783bddd1643fc

SrMouraSilva, 2016-10-03:
I thought that way

## Commentary

``SevenSegmentsBoard`` is a ``SevenSegmentsDisplay`` grouper. With it, you can use ``board.value = "ABCD"``, ``board.value = 1`` or ``board.value = 1.3``. The board split the text (if necessary) and multiplex (usign ``MultiplexThread``).

In the current implementation, the constructor not receive the _common pins controllers_, it occurs in ``add_display`` method. I believe that to stay in gpiozero standard, this should be changed. But I do not know as well as indicate whether the display is anode or cathode (in the constructor), since it is possible to use cathodes and anodes displays together in this implementation.

The common pin is optional, then the ``off()`` methods needs print empty string if the user "plug" the common pin in Gnd or +3 Vcc.

In the current implementation, all the ~~possible~~ default representations are in a [dictionary](https://github.com/PedalPi/Physical/blob/master/component/sevensegments/seven_segments_dictionary.py). If the user requires a unsual pattern, he defines in ``register`` method.

If the user uses a not registred representation, [raises](https://github.com/PedalPi/Physical/blob/master/component/sevensegments/seven_segments.py#L164) ```KeyError('Symbol not registered: ' + value)```

My implementation not considers that the user uses ``SevenSegmentsDisplay``.

``SevenSegmentsDisplay`` not call LED or DigitalOutput in not common pins, it only uses the LEDs defined in ``SevenSegmentsBoard``. I think a refactoring should be carried out in this part, to do [similar work with SPI](https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/spi.py#L381-L383).

## Interface

```python
class SevenSegmentsBoard(object):
    def __init__(self, a, b, c, d, e, f, g):
        ...

    def add_display(self, common=None, anode=False):
        ...

    @property
    def value(self):
        return self._value

    @value.setter
    def value(self, text):
        ...

    def off(self):
        ...

    def on(self):
        ...

    def register(self, char_symbol, byte):
        """
        Link a char_symbol with a byte. The byte represents a led state

        :param char_symbol: if
        :param byte: Each bit represents a led state
                     In order: 1º MSB is ``a led``,
                               2º MSB is ``b led``,
                               consecutively
                               The LCB is dot ``led``;
        """
        dictionary[char_symbol] = byte

class SevenSegmentsDisplay(object):

    def __init__(self, a, b, c, d, e, f, g, common=None, anode=False):
        self.pins = [a, b, c, d, e, f, g]
        ...

    @property
    def status(self):
        ...

    @status.setter
    def status(self, status):
        ...

    def on(self):
        ...

    def off(self):
        ...

    @property
    def value(self):
        ...

    @value.setter
    def value(self, value):
        if value not in dictionary:
            raise KeyError('Symbol not registered: ' + value)
        ...

```

## Example

```python
import time

board = SevenSegmentsBoard(a=13, b=6, c=16, d=20, e=21, f=19, g=26)

board.add_display(common=5, anode=False)
board.add_display(common=1, anode=True)

for i in range(99):
    board.value = i
    print(board.value, i)
    time.sleep(0.2)

board.register('~', 0b10101010)
board.value = '~~'
```

## Additional commentary

The wikipedia cites anothers segments displays types

[image]

lurch, 2016-10-17:
We've recently been discussing 7-segment displays over on #485 too.

IMHO SevenSegmentDisplay should be standalone and not tightly coupled to SevenSegmentsBoard, and therefore not have a 'common' parameter. I think SevenSegmentDisplay should inherit from LEDBoard and perhaps have an init-method like:
```python
    def __init__(self, a, b, c, d, e, f, g, dp=None, pwm=False, active_high=True, initial_value=False):
        segments = OrderedDict((('a': a), ('b': b), ('c': c), ('d': d),  ('e': e),  ('f': f), ('g': g)))
        if dp is not None:
            segments['dp'] = dp
        super(LEDBoard, self).__init__(pwm=pwm, active_high=active_high, initial_value=initial_value, _order=segments.keys(), **segments)
```

I potentially like your idea of being able to 'register' new characters, and IMHO that should be part of SevenSegmentDisplay and would work well with the `display(char)` method from #485 - as long as it wouldn't lead to lots of confusing code with different people defining different characters to illuminate different LEDs...

I definitely don't like the add_display of your SevenSegmentsBoard class - it's very un-gpiozero-like. Instead the init-method should take all args up-front, _perhaps_ something like:
```python
    def __init__(self, a, b, c, d, e, f, g, dp=None, common_pins=None, pwm=False, active_high=True, initial_value=False):
```
(with common_pins being a list). As GpioZero is "device-based" then IMHO the SevenSegmentsBoard would typically be dealing with devices like http://raspi.tv/2015/how-to-drive-a-7-segment-display-directly-on-raspberry-pi-in-python where the segments are either all common-cathode or all common-anode (which is dealt with by the `active_high` parameter in GpioZero).

I **could** go on to write more, but IMHO we ought to thrash out the design of a `SevenSegmentDisplay` class first, before moving on to a `SevenSegmentsBoard` class. Ping @bennuttall and @martinohanlon (and @waveform80 , in case he happens to be walking past)

And regarding the different layouts of segment-displays you mention at the bottom of your last comment - IMHO they'd be handled with e.g. a FourteenSegmentDisplay class (if and when the need arises).

Over in #485 @bennuttall asked:
> Do various 7seg displays tend to work the same way?

I don't have much experience, but I believe they do (i.e. all the 'a' segments paralleled, all the 'b' segments paralleled, etc.) with the only differences being whether they're wired up as common-anode or common-cathode, and how many digits there are. Some other differences are whether each digit has its own decimal point or not, and sometimes there's a colon between the second and third digits (e.g. like a digital clock).

SrMouraSilva, 2016-10-18:
Out of curiosity, my current work can be viewed at:
https://github.com/PedalPi/Physical/tree/master/component/sevensegments

Latest, I will answer!

SrMouraSilva, 2016-10-18:
> I think SevenSegmentDisplay should inherit from LEDBoard and perhaps have an init-method like:

``SevenSegmentDisplay extends LEDBoard`` and ``SevenSegmentsBoard extends CompositeOutputDevice`` ?

> SevenSegmentDisplay should be standalone and not tightly coupled to SevenSegmentsBoard, and therefore not have a 'common' parameter

The common pin is the common ground (or common high voltage) for leds in 7seg. It is necessary for multiplexation in SevenSegmentsBoard.

> I potentially like your idea of being able to 'register' new characters, and IMHO that should be part of SevenSegmentDisplay and would work well with the display(char) method from #485

https://github.com/RPi-Distro/python-gpiozero/issues/485#issuecomment-254236118

> Being a geeky programmer type, I'd love there to be a hexdisplay function too ;-)

I usign [Hex idea](https://github.com/PedalPi/Physical/blob/master/component/sevensegments/seven_segments_dictionary.py)

> and would work well with the display(char) method from #485

I understand you are not yet of how consensus will display the value (``.value`` attribute / ``.display`` attribute / tuple, int, char, hex, byte). My current implementation show the char in ``.value`` attribute. @bennuttall did not find that interesting (https://github.com/RPi-Distro/python-gpiozero/issues/485#issuecomment-254211789).

What I can say is that I believe that:
* Be possible to obtain the current character in string representation;
* Be possible to obtain a hexadecimal representation of the current state;
* Be possible to obtain the tuple of LEDs with their values.

What is easy to think when you consider only `[a-g]`(ignoring dot).

> as long as it wouldn't lead to lots of confusing code with different people defining different characters to illuminate different LEDs...

I think it will not be a problem, since [we provide the most obvious characters](https://en.wikipedia.org/wiki/Seven-segment_display#/media/File:7-segment.svg).

> I definitely don't like the add_display of your SevenSegmentsBoard class - it's very un-gpiozero-like. Instead the init-method should take all args up-front, perhaps something like:

Ok, I agree

> where the segments are either all common-cathode or all common-anode (which is dealt with by the active_high parameter in GpioZero).

> My implementation supports [...] cathodes and anodes (mixed)

I have two 7seg in here, one is cathode and another is anodes. The result multiplexation isn't good, but (maybe) a kid in another place in the world the same sintuation.

By the way, if SevenSegmentsDisplay implementation contains ``common`` controller pin and ``active_high`` constructor parameter, the people can implements as he wants.

My conclusion:
* SevenSegmentsDisplay needs OPTIONAL ``common`` controller pin and ``active_high`` constructor parameter
* SevenSegmentsBoard is not necessary support mixed cathodes and anodes

> Do various 7seg displays tend to work the same way?

I did not understand the question

lurch, 2016-10-18:
> SevenSegmentDisplay extends LEDBoard and SevenSegmentsBoard extends CompositeOutputDevice ?

Yeah, I think that sounds about right. In reality a multiplexed `SevenSegmentsBoard` will be composed of a SevenSegmentDisplay and a list of DigitalOutputDevices (for the common pins), but from a user-API point of view I guess we want it to _appear_ as a collection of SevenSegmentDisplay devices, i.e. the `.value` property should return a tuple of 8-tuples of bools.
And it makes sense for a SevenSegmentsBoard to have a `display(string)` method (perhaps with an optional `fill_char` parameter for 'short' strings) - a bit like you suggested, but without needing a whole set of separate `TextWriteStrategy` classes. And then we could also add:
```python
    def hexdisplay(self, hexnumber, fill_char='0'):
        self.display(hex(hexnumber)[2:], fill_char=fill_char)
```

> What I can say is that I believe that: Be possible to obtain the tuple of LEDs with their values.

IMHO that's the most logical representation, given how the rest of GpioZero works. See e.g. http://gpiozero.readthedocs.io/en/v1.3.1/api_boards.html#gpiozero.LEDBoard.value

> Be possible to obtain the current character in string representation;

I think we should allow the user to **set** the current character (e.g. with @martinohanlon 's `display(char)` method), but I don't think we should allow the user to **get** the 'current character', since it's possible for the 7-segment to be displaying a combination of LEDs that doesn't map back to a valid character, and property-getters probably shouldn't be throwing Exceptions.

> I have two 7seg in here, one is cathode and another is anodes. The result multiplexation isn't good, but (maybe) a kid in another place in the world the same sintuation.

I'm afraid I disagree. IMHO with the "device-based" approach used by GpioZero, a common-cathode 7-segment and a common-anode 7-segment should be two separate objects in GpioZero, and we shouldn't allow them to be combined into a single SevenSegmentsBoard. Look at how http://gpiozero.readthedocs.io/en/latest/api_boards.html#ledboard and http://gpiozero.readthedocs.io/en/latest/api_boards.html#ledbargraph work.

Everything written here is obviously only IMHO, @bennuttall has the final decision.

SrMouraSilva, 2016-10-18:
> Yeah, I think that sounds about right. In reality a multiplexed SevenSegmentsBoard will be composed of a SevenSegmentDisplay and a list of DigitalOutputDevices (for the common pins), but from a [...]

I totally agree. Even, I reflected on what I wrote about the mandatory common_pin in SevenSegmentsDisplay and I saw that it is not necessary.

Your suggestion Implementation with a single `SevenSegmentsDisplay` and a list of commons_pins explains this.

> And it makes sense for a SevenSegmentsBoard to have a display(string) method (perhaps with an optional fill_char parameter for 'short' strings) - a bit like you suggested, but without needing a whole set of separate TextWriteStrategy classes. And then we could also add:

The main objective of Strategies there was for cases like: `sevenSegments.display (1.23)` (which I do not implemented)

Perhaps a method as described below solves
```python
    def floatdisplay(self, number, formatter='{:03.2f}'):
        size = len(self.common_pins)
        text = formatter.format(number)
        if '.' in text:
            size += 1

        if size != len(text):
            raise SegmentsException('Float does not fit on the DisplayBoard')
        self.display(hex(hexnumber)[2:], fill_char=fill_char)
```

(But I think too complicated for the GpioZero.)

Maybe a ltr (left to right) boolean is an interesting ideia too.

> I think we should allow the user to set the current character (e.g. with @martinohanlon 's display(char) method), but I don't think we should allow the user to get the 'current character', since [...]

If the user register a char in Chars Dictionary display, why not?
But, if is possible manually change a pin state, yes, it is a problem.

> I'm afraid I disagree. IMHO with the "device-based" approach used by GpioZero, a common-cathode 7-segment and a common-anode 7-segment should be two separate objects in GpioZero, and we shouldn't allow them to be combined into a single SevenSegmentsBoard. Look at how

I'm sorry, I think I was not clear. I meant that despite my initial implementation allow it, I agree with you.

> Everything written here is obviously only IMHO, @bennuttall has the final decision.

I believe that the discussion is very useful!

SrMouraSilva, 2016-10-18:
This is a sketch of the signature of the methods and features (implementation is only illustrative).

```python
class SevenSegmentsDisplay(object):

    def __init__(self, a, b, c, d, e, f, g, dp=None, active_high=True):
        ...

    @property
    def status(self):
        """ :return boolean It is on?"""
        ...

    @status.setter
    def status(self, status):
        """ :param boolean status: On if true, off else False"""
        ...

    def on(self):
        ...

    def off(self):
        ...

    @property
    def value(self):
        """ :param Tuple(pins status)"""
        ...

    @value.setter
    def value(self, value):
        """ It makes sense that? """

    def hexdisplay(self, hexnumber):
        """ :param int hex: In format 0x_ where _ is [0-9] || [A-F]"""
        self.display(hex(hexnumber)[2:])

    def display(self, char)
        """ char is case insensitive? """
        char = str(char).upper()

        if len(char) != 1:
            raise ValueError('char is a unique char :P')
        if char not in dictionary:
            raise KeyError('Symbol not registered: ' + value)
        ...
        self._char = char  # For on/off purposes

    @property
    def dot_point(self):
        if self.dp is not None:
            return self.dp.value
        return False

    @dot_point.setter
    def dot_point(self, value):
        if self.dp is None:
            raise DisplayError("It display not registred a dot point")
        self.dp.value = value

    @property
    def displayed(self):
        """ MAYBE """
        return self._char
```

lurch, 2016-10-18:
> Perhaps a method as described below solves
\[snip floatdisplay method\]
> (But I think too complicated for the GpioZero.)

On the contrary - some of the code 'behind the scenes' in GpioZero is actually _really_ complicated, in order to provide a simple-to-use API to the end user. Given that #488 adds a `display_hex` method, I guess there's no reason SevenSegmentsBoard couldn't have a `display_float` method.

> Maybe a ltr (left to right) boolean is an interesting ideia too.

Could you explain what you mean?

> I believe that the discussion is very useful!

I agree! :+1: 

> This is a sketch of the signature of the methods and features

Some of the methods you mention (or similar) get provided automatically if you derive from e.g. `LEDBoard` instead of deriving all the way from `object`.

> int hex: In format 0x_ ...

That's not strictly correct - `hex(13)` and `hex(0xd)` both return the str `'0xd'`

SrMouraSilva, 2016-10-18:
>> Maybe a ltr (left to right) boolean is an interesting ideia too.

> Could you explain what you mean?

https://en.wikipedia.org/wiki/Right-to-left

> Some of the methods you mention (or similar) get provided automatically if you derive from e.g. LEDBoard instead of deriving all the way from object.

Good to know

>> int hex: In format 0x_ ...
> That's not strictly correct - hex(13) and hex(0xd) both return the str '0xd'

It's a draft rsrs. I'm sorry

> On the contrary - some of the code 'behind the scenes' in GpioZero is actually really complicated, in order to provide a simple-to-use API to the end user. Given that #488 adds a display_hex method, I guess there's no reason SevenSegmentsBoard couldn't have a display_float method.

:smile:

lurch, 2016-10-18:
>>> Maybe a ltr (left to right) boolean is an interesting ideia too.

>> Could you explain what you mean?

> https://en.wikipedia.org/wiki/Right-to-left

I still don't know what you mean in relation to 7-segment displays...?

SrMouraSilva, 2016-10-18:
I'm sorry! SevenSegmentsBoard.
But, really...

Maybe a text align (left, right, center) have any sense

lurch, 2016-10-18:
Ah! So did you mean an option to reverse the order of the digits within the display? So that "4cat" would display as "tac4"?

> Maybe a text align (left, right, center) have any sense

We _could_, but I think that's probably something best left to user-code.

SrMouraSilva, 2018-02-19:
#488

waveform80, 2021-03-12:
Fixed in #930

closed 2021-03-12 by waveform80