gpiozero/gpiozero #359: default pin_factory based on environmental value

dglaude, 2016-06-06:
Based on #353 discussion, here is the idea.

In pigpio you define in environment variables how to connect to the pigpiod of your Pi (if nothing is explicitly said in the code).

That same trick could/should be used into gpiozero and provide a way to externally choose the pin implementation. Currently it is hard wired starting from line 32 of devices.py. It could be softwired with an single environmental variable such as GPIOZERO_PIN_FACTORY that could be not set for current behaviour or set to RPiGPIOPin or RPIOPin or PiGPIOin or NativePin.

lurch, 2016-06-06:
If this gets implemented, it would also make @ryanteck 's RTk.GPIO even *more* plug'n'play (assuming he adds the promised GpioZero integration) with exisiting GpioZero scripts than he already thought ;-)

dglaude, 2016-06-10:
Here is the proposed "test" code to verify the behaviour based on environmental value.
Latter try to write some code that bypass the fall-back sequence if GPIOZERO_PIN_FACTORY is set.

```
import gpiozero.devices
print ("Selected pin factory is " , gpiozero.devices.pin_factory)

import os
print ("GPIOZERO_PIN_FACTORY is ", os.getenv("GPIOZERO_PIN_FACTORY", 'not set'))
print ("PIGPIO_ADDR is ", os.getenv("PIGPIO_ADDR", 'not set'))
print ("PIGPIO_PORT is ", os.getenv("PIGPIO_PORT", 'not set'))

from gpiozero import LED
red = LED(9)
red.blink()

from signal import pause
pause()
```

This it the output on my Mac:

> $ python factory.py 
> ('Selected pin factory is ', <class 'gpiozero.pins.pigpiod.PiGPIOPin'>)
> ('GPIOZERO_PIN_FACTORY is ', 'not set')
> ('PIGPIO_ADDR is ', '192.168.192.113')
> ('PIGPIO_PORT is ', 'not set')

This is the output on my Pi:

> $ python factory.py 
> ('Selected pin factory is ', <class 'gpiozero.pins.rpigpio.RPiGPIOPin'>)
> ('GPIOZERO_PIN_FACTORY is ', 'not set')
> ('PIGPIO_ADDR is ', 'not set')
> ('PIGPIO_PORT is ', 'not set')

I remove* RPiGPIOin with  `sudo apt-get remove python-rpi.gpio` this is the output on my Pi:

> $ python factory.py 
> ('Selected pin factory is ', <class 'gpiozero.pins.native.NativePin'>)
> ('GPIOZERO_PIN_FACTORY is ', 'not set')
> ('PIGPIO_ADDR is ', 'not set')
> ('PIGPIO_PORT is ', 'not set')

* I noticed that with apt-get, if you remove 'python-rpi.gpio' it remove 'python-gpiozero':

> The following packages will be REMOVED:
>   python-gpiozero python-rpi.gpio
> 0 upgraded, 0 newly installed, 2 to remove and 10 not upgraded.
> After this operation, 428 kB disk space will be freed.
> Do you want to continue? [Y/n] Y
> (Reading database ... 123688 files and directories currently installed.)
> Removing python-gpiozero (1.2.0) ...

I believe this is a mistake in the package definition since gpiozero can work without rpi.gpio.

waveform80, 2016-06-10:
I'm coming around to this idea, if only due to @lurch's comment on #278 regarding easier usage of MockPin (not quite the same as a `set_pin_factory` function, but close enough as one could invoke the interpreter with something like `GPIOZERO_PIN_FACTORY='MockPin' python`).

However, rather than a series of "if" statements hard-coding the test (which we'd be responsible for expanding), I'd rather have something that registers Pin derivatives which can then be searched for matches (which then means if people want to add their own implementations, e.g. the Rtk.GPIO, they can do so without having to wait for us to do a release expanding the "if" statement).

I can probably build that machinery to happen automatically without having to modify any existing Pin classes, but I won't have time this weekend. Anyway, I'll take this one and try and get it in before 1.3.

dglaude, 2016-06-10:
Without having to wait, for next release, @ryanteck could propose this solution:
```
from ryanteck.RtkGPIO import RtkGPIOPin
import gpiozero.devices

gpiozero.devices.pin_factory = RtkGPIOPin
```
to the 1300 potential owner of a Rtk.GPIO, that would use that waiting for next release or get a github version of the library.

So you were the one saying:

> The other reason is that while #353 has us reading data from the environment (hostnames and ports), #359 potentially has us reading executable code from the environment which makes me extremely nervous.

Then @lurch explained that this was likely not the case showing his if elseif template code. And that was exactly my idea, test the environment variable, and if not set, use the existing fallback algorithm.

And now you seems to want to implement it with dynamically loaded python library one based on an environment variable???

With such a feature, if someone can control the environment under witch any python program, using gpiozero library run, will likely be able to run arbitrary python code.

Whatever, when you have your code, I should be able to test it from a Mac to a Pi, from a Pi to a Pi and from a Pi to itself. But this will not stop me from writing a version based on @lurch kind of code.

waveform80, 2016-06-10:
Not dynamically loaded - if a Pin derivative is defined within the current set of available classes then it should be available to be discovered by some associated key (unqualified class name is *probably* good enough for this). So, no direct code execution, and no dynamic loading of libraries; if the class has been registered (created) it's discoverable otherwise it's not.

Obviously all Pin derivatives shipped with GPIO Zero that are capable of being imported at package import time would be implicitly discoverable; third party implementations would require an explicit import in scripts that used them but nothing more. If we wanted to eliminate even the explicit import we could define a setuptools entry-point but that edges us closer to arbitrary code execution under certain circumstances (assumes attacker has write access to a directory within sys.path, or one mentioned in setuptools' `.pth` paths - which isn't that difficult in most setups).

waveform80, 2016-06-10:
Oh, and on the RTk.GPIO stop-gap solution - absolutely right - nothing wrong with having something useful in the meantime.

Sorry I don't have more time at the moment, unfortunately picamera 1.11 has turned into something more complicated than I'd intended (although I'm still convinced it'll be worth it in the long run). I'll try and get back to GPIO Zero as soon as I can!

dglaude, 2016-06-11:
I am testing my piece of code and a few question came to me:
1. What should be done if the environment variable contain an unknown factory?
2. What should be done if we fail to load the requested library?
3. What is the MockPin factory and should it be included?

Here are my test result. Pull request is incomming.

This is an explicit typo to check the error message:
```
$ export GPIOZERO_PIN_FACTORY="typo"; python factory.py
Valid value for GPIOZERO_PIN_FACTORY are: 'RPiGPIOPin' 'RPIOPin' 'PiGPIOPin' 'NativePin' 'MockPin'
('Selected pin factory is ', <class 'gpiozero.pins.rpigpio.RPiGPIOPin'>)
('GPIOZERO_PIN_FACTORY is ', 'typo')
```

```
$ export GPIOZERO_PIN_FACTORY="typo"; python factory.py
Valid value for GPIOZERO_PIN_FACTORY are: 'RPiGPIOPin' 'RPIOPin' 'PiGPIOPin' 'NativePin' 'MockPin'
('Selected pin factory is ', <class 'gpiozero.pins.rpigpio.RPiGPIOPin'>)
('GPIOZERO_PIN_FACTORY is ', 'typo')
```

```
$ export GPIOZERO_PIN_FACTORY="RPiGPIOPin"; python factory.py
('Selected pin factory is ', <class 'gpiozero.pins.rpigpio.RPiGPIOPin'>)
('GPIOZERO_PIN_FACTORY is ', 'RPiGPIOPin')
```

PiGPIOPin tested with and without the library installed.
```
$ export GPIOZERO_PIN_FACTORY="PiGPIOPin"; python factory.py
Failure to import requested pin factory requested by GPIOZERO_PIN_FACTORY. Using default fallback sequence.
('Selected pin factory is ', <class 'gpiozero.pins.rpigpio.RPiGPIOPin'>)
('GPIOZERO_PIN_FACTORY is ', 'PiGPIOPin')

$ sudo apt-get install python-pigpio python3-pigpio
…

$ export GPIOZERO_PIN_FACTORY="PiGPIOPin"; python factory.py
('Selected pin factory is ', <class 'gpiozero.pins.pigpiod.PiGPIOPin'>)
('GPIOZERO_PIN_FACTORY is ', 'PiGPIOPin')
```

```
$ export GPIOZERO_PIN_FACTORY="NativePin"; python factory.py
('Selected pin factory is ', <class 'gpiozero.pins.native.NativePin'>)
('GPIOZERO_PIN_FACTORY is ', 'NativePin')
```

I don't know what the MockPin is and if it should be included:
```
$ export GPIOZERO_PIN_FACTORY="MockPin"; python factory.py
('Selected pin factory is ', <class 'gpiozero.pins.mock.MockPin'>)
('GPIOZERO_PIN_FACTORY is ', 'MockPin')
```

I had to be root do to the RPIOPin because I had 'RuntimeError: No access to /dev/mem. Try running as root!':
```
# export GPIOZERO_PIN_FACTORY="RPIOPin"; python factory.py
('Selected pin factory is ', <class 'gpiozero.pins.rpio.RPIOPin'>)
('GPIOZERO_PIN_FACTORY is ', 'RPIOPin')
```

lurch, 2016-06-12:
Replying to various bits'n'pieces...

> I noticed that with apt-get, if you remove 'python-rpi.gpio' it remove 'python-gpiozero' ... I believe this is a mistake in the package definition since gpiozero can work without rpi.gpio.

Perhaps in https://github.com/RPi-Distro/python-gpiozero/blob/master/debian/control `python-rpi.gpio` and `python3-rpi.gpio` could be moved from 'Depends:' to 'Recommends:' ? https://www.debian.org/doc/debian-policy/ch-relationships.html#s-binarydeps
IIRC on Raspbian it auto-installs the Recommends packages, but I know very little about packaging and dunno if you can remove a 'Recommended' package without it trying also remove the 'Recommender' package?

> I am testing my piece of code and a few question came to me:
> 1\. What should be done if the environment variable contain an unknown factory?

IMHO it should just drop through to the current behaviour of selecting the 'best' already-installed pin library. You can either silently ignore it, or perhaps you could [warn](https://docs.python.org/2/library/warnings.html#warnings.warn) that `GPIOZERO_PIN_FACTORY` contains an unrecognised value.

> 2\. What should be done if we fail to load the requested library?

Based on @waveform80 's comments, I'm guessing he's proposing adding a `register_pin_class` (or similar) function, which would update an internal dictionary-like structure. So e.g. in theory you could add `import rtk_gpiozero` somewhere near the top of your existing python script (that uses gpiozero), and then inside the rtk_gpiozero library it would automatically do something like `register_pin_class('RTkGPIOPin', rtk_gpiozero.pins.RTkGPIOPin)`. And then you could set the environment-variable `GPIOZERO_PIN_FACTORY=RTkGPIOPin` and your script would automatically use `rtk_gpiozero.pins.RTkGPIOPin` to create its pins, without anything having to be modified inside GpioZero itself.
All AIUI of course, which may be wrong!! ;-)

So it's loading the library itself that adds the pin-class to the list of allowable-values for `GPIOZERO_PIN_FACTORY`, meaning we'd never get a "we fail to load the requested library" scenario.

> 3\. What is the MockPin factory and should it be included?

It's used for testing GpioZero (see e.g. https://github.com/RPi-Distro/python-gpiozero/blob/master/tests/test_outputs.py or https://github.com/RPi-Distro/python-gpiozero/pull/324#issue-154256395 ) and allows you to test / develop / debug the GPIO functionality of GpioZero on your PC when a Pi isn't available. As well as `MockPin`, there's also `MockPulledUpPin`, `MockChargingPin`, `MockTriggerPin` and `MockPWMPin`; which all emulate various pin-related functionality.
IMHO `GPIOZERO_PIN_FACTORY` should at least allow `MockPin` (digital-only input/output) and `MockPWMPin` (digital input/output and PWM output).

BTW doing all of `export GPIOZERO_PIN_FACTORY="RPiGPIOPin"; python factory.py` isn't strictly necessary - you can achieve the same result with `GPIOZERO_PIN_FACTORY=RPiGPIOPin python factory.py` (note the lack of semi-colons - this sets the env-var _just_ for the specifically-launched `python` process, rather than setting it shell-wide).  For root-usage with RPIO, you can do `sudo GPIOZERO_PIN_FACTORY=RPIOPin python factory.py` .

dglaude, 2016-06-12:
All my code is [here](https://github.com/dglaude/python-gpiozero-pigpio) and I believe I made a pull request.

If the value in GPIOZERO_PIN_FACTORY is unknow, I use the fallback and give a warning.
If the value is GPIOZERO_PIN_FACTORY is known but I fail to load, fallback with a warning.
I did put MockPin, but not the other Mock*

So far the code is a big if elif else build into a exception catcher. There might be better way to write it and @waveform80 has his idea. Waiting for that to translate into code, it is at least functional.
I believe I tested extensively, installing and removing various library, but there can be corner case.

Week-end is over, so I will not touch it for a while.
Anybody feel free to use/test/modify or redo it another way.

dglaude, 2016-07-26:
This PR has now been replaced by two separate PR:
- #395 for the PIGPIO Address and Port
- #396 for the PIN_FACTORY selection

waveform80, 2016-08-14:
> Based on @waveform80 's comments, I'm guessing he's proposing adding a register_pin_factory (or similar) function

Yup, that's exactly it. Though I'd like to explore setuptools entry points first (I've had a read up of the security risks surrounding them and it basically amounts to: attacker must be able to inject stuff into your sys.path ... but if they can do that you're pretty screwed already). Using setuptools entry points eliminates even the need for something like `import rtk_gpiozero` - the simple fact of the module being installed would be enough for us to locate it via setuptools based on an environment variable setting.

And as @lurch goes on to suggest, this means no need to alter gpiozero itself to register new pin factories.

I'm in two minds about what to do when the GPIOZERO_FACTORY env-var contains an invalid / unrecognized value. Personally, I'd favour just throwing an error and bombing out. I know that seems "nasty" but it's also the safest option: what if someone expects the env-vars to direct all pin manipulations to a remote Pi, but they've spelt something wrong and all of sudden we start messing with pins on their pi because we've fallen through to a local library? Or they're expecting a library with full hardware PWM, but we fall through a soft one and their servos start twitching. Basically, the least I'd expect is a warning to be thrown, but I'm leaning towards a full blown exception.

waveform80, 2016-08-14:
Oh ... and I must update debian/control as suggested ... I'll just do that now while I've got a mo.

lurch, 2016-08-14:
> what if someone expects the env-vars to direct all pin manipulations to a remote Pi, but they've spelt something wrong and all of sudden we start messing with pins on their pi because we've fallen through to a local library?

Yeah, that's a good point.

waveform80, 2016-08-14:
Okay, got the setuptools entry-points working - remarkably easy as it turns out. One thing to bear in mind once I've merged this: because this commit introduces new entry-points you *must* re-run "python setup.py develop" in any virtual-envs containing a gpio-zero checkout otherwise it won't be able to locate the pin factories.

For the time being I've left the default order alone (RPiGPIOPin, RPIOPin, PiGPIOPin, NativePin) as I figure changing that ought to be a separate commit.

lurch, 2016-08-14:
Does this mean that the 3rd-party pin libraries (RPi.GPIO etc.) need to be modified to add entrypoints to their setup.py, if they aren't doing so already?

waveform80, 2016-08-14:
Nope - the entry points are specifically for GPIO Zero pin implementations, so we'll define the ones we include (RPiGPIOPin, etc.). If some other package defines a new GPIO Zero pin implementation, let's take RtkGPIOPin as an example, all it has to do is add something like:

```
    entry_points = {'gpiozero_pin_factories': ['RtkGPIOPin = rtkgpio.pins:RtkGPIOPin']},
```

to their `setup()` call in their `setup.py` and then when users install their package it'll register the new pin factory for us to find. Users can then simply use `GPIOZERO_PIN_FACTORY='RtkGPIOPin' python foo.py` and it'll all "Just Work (tm)"

lurch, 2016-08-14:
Cool!
Pinging @ryanteck ;-)

commit 1ec2b763 (2016-08-14): Fix #359, close #396
commit 64cde7f4 (2016-08-14): Merge pull request #400 from waveform80/pin-entry-points

Fix #359, close #396
closed 2016-08-14 by waveform80