gpiozero/gpiozero #363: Failure to add edge detection on GPIO4 using RPi.GPIO pin

bennuttall, 2016-06-06:
I have reproduced a bug found by @marcscott. It's impossible to create an input device (e.g. `Button` - any with edge detection) on GPIO4 due to a `RunTimeError` produced by RPi.GPIO.

To reproduce:

```python
>>> from gpiozero import Button
>>> b = Button(4)
```

To show it's just pin 4:

```python
>>> from gpiozero import Button
>>> for i in range(27):
...     try:
...         b = Button(i)
...     except:
...         print(i)
```

This is similar to a bug (#128) we've had before which was fixed in a recent RPi.GPIO update.

@bcroston are you aware of this?

bcroston, 2016-06-06:
I've seen it.  I'm so busy working on things that earn badly needed money at the moment that I haven't had chance to look at RPi.GPIO for months.

bennuttall, 2016-06-06:
Ok at least you're aware.

lurch, 2016-06-06:
@bennuttall Do you have more detailed reproduction steps? Works for me with both python and python3 on a RaspberryPi3. I wondered if it might be kernel-related, so I apt-get dist-upgraded my Jessie 2016-03-18 from `Linux raspberrypi3 4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l GNU/Linux` to `Linux raspberrypi3 4.4.11-v7+ #888 SMP Mon May 23 20:10:33 BST 2016 armv7l GNU/Linux` but it still works okay :-S

bennuttall, 2016-06-06:
Actually, we realised this is not the bug we thought it was. It's due to the one-wire protocol being enabled, which uses GPIO4 by default (can be mapped to another in config.txt).

However, we ought to find a way of detecting this and showing an appropriate error.

bcroston, 2016-06-06:
I intend to add one-wire to RPi.GPIO as one of the next features - when implemented, trying to use the onewire pin will give a RuntimeException with a suitable error message (e.g. channel already in use) rather than a warning.

lurch, 2016-06-06:
I'm guessing this probably affects other GPIO pin-implementations than just RPi.GPIO though, and so the error ought to be detected at the GpioZero level?

closed 2016-06-16 by bennuttall