gpiozero/gpiozero #364: Problem with images on RTD documentation

lurch, 2016-06-06:
I'm not sure how it's gone wrong, or what needs to be done to get it back in sync, but for some reason http://gpiozero.readthedocs.io/en/stable/_images/composed_devices.svg (as displayed on http://gpiozero.readthedocs.io/en/stable/api_boards.html#base-classes) is lagging behind https://github.com/RPi-Distro/python-gpiozero/blob/master/docs/images/composed_devices.svg

waveform80, 2016-08-14:
You're looking at gpiozero.readthedocs.io/en/stable - which isn't "latest". I'm not sure exactly how RTD defines "stable" ... I think it's the last named version prior to master, in which case looking at "stable" is equivalent to looking at "1.2.0" which won't include that updated graph. If you look at "latest" it is indeed up to date.

lurch, 2016-08-14:
Doh! My mistake, sorry.

closed 2016-08-14 by lurch