gpiozero/gpiozero #365: Using sourcetools with CompositeDevices

lurch, 2016-06-08:
I've just been playing around with my potential ButtonBoard implementation... ( #340 )

I've got 5 different-colour LEDs (from http://rasp.io/experiment/ ) and 5 different-colour buttons (from http://www.hobbytronics.co.uk/electronic-components/switches-relays/tactile-button-assortment ) all hooked up on a breadboard, and I'm using this code:
```python
>>> from gpiozero import ButtonBoard, LEDBoard
>>> from gpiozero.tools import *
>>> leds = LEDBoard(red=8, yellow=10, green=12, blue=14, white=16, _order=('red', 'yellow', 'green', 'blue', 'white'))
>>> buttons = ButtonBoard(red=9, yellow=11, green=13, blue=15, white=17, _order=('red', 'yellow', 'green', 'blue', 'white'))
>>> leds
<gpiozero.LEDBoard object containing 5 devices: yellow,green,red,blue,white and 0 unnamed>
>>> leds.value
LEDBoardValue(red=False, yellow=False, green=False, blue=False, white=False)
>>> buttons
<gpiozero.ButtonBoard object containing 5 devices: yellow,green,red,blue,white and 0 unnamed>
>>> buttons.value
ButtonBoardValue(red=False, yellow=False, green=False, blue=False, white=False)
```

I can then do:
```python
>>> leds.source = buttons.values
```
and as expected, pressing each button lights up the matching colour LED, which is fun ;-)
However if I try this:
```python
>>> leds.source = negated(buttons.values)
>>> Exception in thread Thread-30:
Traceback (most recent call last):
  File "/usr/lib/python3.4/threading.py", line 920, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.4/threading.py", line 868, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/github/lurch/python-gpiozero/gpiozero/mixins.py", line 80, in _copy_values
    self.value = v
  File "/home/pi/github/lurch/python-gpiozero/gpiozero/devices.py", line 299, in __setattr__
    return super(CompositeDevice, self).__setattr__(name, value)
  File "/home/pi/github/lurch/python-gpiozero/gpiozero/devices.py", line 146, in __setattr__
    return super(GPIOBase, self).__setattr__(name, value)
  File "/home/pi/github/lurch/python-gpiozero/gpiozero/boards.py", line 86, in value
    for device, v in zip(self, value):
TypeError: zip argument #2 must support iteration
```
it doesn't work. I'm only able to get it to work with:
```python
>>> leds.source = None
>>> for i in range(len(leds)):
...     leds[i].source = negated(buttons[i].values)
```
and then the LEDs are always on until the matching button is pressed.

Is there any way to make the 'sourcetools' methods work transparently with CompositeDevices, the same way they do with regular Devices?

Also (perhaps this part should be a separate issue?), if I do:
```python
>>> for i in range(len(leds)):
...     leds[i].source = negated(buttons[i].values)
...
>>> leds.source = buttons.values
```
then the LEDs are always on, and when I press one of the buttons the corresponding LED flickers, as though the two sources are 'fighting for control' of the LED. Possibly this is related to #200 and it reminds me of the situation we had a while ago (which @waveform80 managed to resolve) of having different 'blinks' on an LEDBoard and one of it's sub-device LEDs.

waveform80, 2016-08-14:
> I can then do:

> ```python
> >>> leds.source = buttons.values
> ```

Damn, that's nice! Must go and check out this `ButtonBoard` thing ... then I'll come back to this.

waveform80, 2016-08-14:
> Also (perhaps this part should be a separate issue?), if I do:
>
> ```python
> ...     leds[i].source = negated(buttons[i].values)
> ...
> ```
>
> then the LEDs are always on, and when I press one of the [...]

Yup, they are fighting for control: think about the handlers you've created there. You've got a negated source handler on each individual LED, and a (non-negated) source handler for the composite device. The composite device handler will thus disagree with each individual device handler and depending on which source thread gets scheduled at any given moment, will flip the LED to its value (negated or not).

I'll see if I can refine `negated` a bit to detect when it's dealing with a composite (detecting iterables in Python is actually a subtle and tricky thing to get right in all circumstances, but I think something's possible here especially as we never expect string values, which are usually the complicating factor).