gpiozero/gpiozero #366: Motor with enable control

bennuttall, 2016-06-08:
Motors on the new MotoZero board require 3 pins to control them. Forward pin, backward pin and an enable pin. They can be controlled with GPIO Zero by using `Motor` for the forward and backward pins, and `OutputDevice` for the enable pin like so:

```python
>>> from gpiozero import Motor, OutputDevice
>>> motor = Motor(24, 27)
>>> motor_enable = OutputDevice(5)
>>> motor.forward()  # nothing happens
>>> motor_enable.on()  # the motor goes forward
>>> motor_enable.off()  # the motor stops
>>> motor.backward()  # nothing happens
>>> motor_enable.on()  # the motor goes backward
>>> motor.value = 0.5  # the motor goes forward at half speed
>>> motor.stop()  # the motor stops
```

My instinct is to want to remove the enabling step, and just let `Motor` turn on the enable pin on init, and never change its state, allowing `forward()`, `backward()`, `stop()` and `value` to be used to control the state of the motor.

However, Keith has [vocalised](https://twitter.com/PiTutorials/status/740489152253288448) his views that the enable pin is particularly useful:

> Nice benefit of using enable pin is that it can be PWMed to control the speed, independent of dir.

and

> Enable pin 0 = motor off, Enable pin 1 = Motor full speed. Enable pin PWM 0.5 = Motor 1/2 speed

But I don't think this makes it any easier than gpiozero already does:

```python
>>> motor = Motor(17, 18)
>>> motor.forward()
>>> motor.value = 0.5
>>> motor.backward()
>>> motor.value = -0.5
>>> motor.reverse()
```

I propose we add an optional third pin to `Motor` and set it high on init. This will allow it to easily fit into `Motor` and `Robot`.

Any thoughts?

keithellis74, 2016-06-08:
Yes for compatibility adding a third option pin makes perfect sense.  My experience was a while ago with RPi.GPIO when things were not as easy as they are now with gpiozero.

However it would be good to be able to set the enable pin low to enable debug mode, i.e. set enable low and allow code to run without having to detach power to motors.

bennuttall, 2016-06-08:
It's much easier to do PWM in gpiozero than in RPi.GPIO. So yeah.

lurch, 2016-06-08:
From a bit of internet searching, it seems like you can control the L293D by either treating the IN1 and IN2 pins as digital, and then PWM the speed on ENABLE1; or you can treat the ENABLE1 pin as digital and then PWM on the IN1 and IN2 pins.
For microcontrollers with limited PWM pins it's obviously better to only PWM on ENABLE1, but I couldn't find any definitive references saying whether one approach was better than the other or not. So perhaps either approach is equally valid?
(In contrast, the datasheet for the L298N explicitly recommends only PWMing the ENABLE pin)

bennuttall, 2016-06-08:
Ah I see, the `(Digital_forwards, Digital_backwards, PWM_speed)` makes sense now. In which case, I'd rather stick with one implementation than two. Either we opt for #367, and PWM is done on the forward or backward pin, or we opt for having a different motor class for these types of motors, which are controlled via the enable pin.

I prefer the former solution, but if there's enough support for doing it the latter way, we might have to.

keithellis74, 2016-06-09:
From the perspective of an L293 or equivalent,  I don't think it matters how you PWM it, its just a case implementation.  Since gpiozero does the implementation we do not need to care any more, so the solution in #367 works well.

I was not aware the L298 should be PWMed on the enable pin.  Our PiWars 2014 entry used a L298 PWMed on the drive pins, it seemed to work well.  After reviewing the data sheet today the only reason I can see for having to PWM on the enable pin is because of the 'Fast Stop' feature

For 'Fast Stop' to be engaged, enable pin must be high and and the drive pins must be equal.  Therefore if the drive pins are being PWMed, each cycle of the PWM the 'Fast Stop" feature will kick in.  If PWM is on the enable pin the motors will 'free spin' between cycles.

In critical applications this may be important, but in terms of hobby electronics, I'm not sure how important it is.  It might reduce the life cycle of the motor driver or the motor. L298 data sheet attached, refer to page 6 for 'Fast Stop' feature.
[en.CD00000240.pdf](https://github.com/RPi-Distro/python-gpiozero/files/306496/en.CD00000240.pdf)

bennuttall, 2016-06-09:
@waveform80 what do you think?

We also need to consider the four motor robot. Is that a trivial extension?

lurch, 2016-06-09:
Ahh, I was looking at a different datasheet previously - see https://github.com/RPi-Distro/python-gpiozero/issues/300#issuecomment-215999378
Although now I read it again, I see it refers to both L293 and L298 in the same category, so it seems I was wrong to assume they needed to be treated differently. \*shrug\*

commit f8ef07de (2018-09-20): Add optional enable init param to motor, close #366
commit 0dc72da0 (2018-09-20): Add optional enable init param to motor, close #366
commit 990cfb39 (2018-09-20): Merge pull request #677 from RPi-Distro/motor-enable2

Add optional enable init param to Motor, close #366
closed 2018-09-20 by bennuttall