gpiozero/gpiozero #368: Four wheeled Robot

bennuttall, 2016-06-09:
Currently our `Robot` class comprises two motors. Could we trivially allow users to specify four sets of motor pins within the same class, or would we need a new class?

lurch, 2016-06-09:
IMHO a new class would be better.
I also discussed various Motor and Robot class names here https://github.com/RPi-Distro/python-gpiozero/pull/277#issuecomment-211608633 - in hindsight IMHO the current 'Motor' and 'Robot' class names are too generic, and the 'Robot' classes should be passed 'Motor' instances (just like other composite devices) rather than tuples-of-pin-numbers - which would eliminate the need for a PhaseEnableRobot in #276 

For the current two-wheeled robot implementation (using 'tank steering') it's fairly obvious how the API should work in terms of forward/backward/left/right.
What should a four-wheeled robot API look like? Do we have any example 4-wheeled (4 motor) robots we can use for inspiration?

MrHarcombe, 2016-06-23:
Four wheel drive is tricky without a differential, isn't it? Unless you have snazzy wheels like the ones I've seen on the VEX Robotics Omni wheels (as in [this video](https://www.youtube.com/watch?v=S0twpNrD1pM)), that is...