gpiozero/gpiozero #377: Problem changing LED assignment while led is on.

DarrenTownsend, 2016-06-11:
Example:

led = LED(4)
led.on()

This turns on the led connected to gpio4, as one would expect. However, if you then do;

led = LED(18) _this could be any available gpio_;

the led connected to gpio4 turns off (but, interestingly, an led connected to gpio18 does NOT turn on). If you then issue an 'led.on()', the led connected to gpio18 will turn on.

bennuttall, 2016-06-11:
That is the expected behaviour.

When you create an LED object and assign it to `led`, you hold a reference to it in `led`. You can create another with another object name and both references are then active. However, you can close the object explicitly with `led.close()` which will remove your control of it and return the pin to us original state.

You can also implicitly close the object by deleting the reference with `del led` but that will wait for the garbage collector to actually do the cleanup. Or if you create a new object with the name of an existing one it'll probably just get cleaned up straight away. Cleanup restores the state (which usually turns it off) so that's what you're seeing.

If you want to keep both LEDs on, you'll need to create two separate objects for them e.g. `led` and `led2` or a list `leds = [LED(4), LED(18)]` or even `LEDBoard(4, 17)`.

closed 2016-06-11 by bennuttall

waveform80, 2016-08-14:
Just a quick correction here - Ben's basically covered everything, but I'll walk through the object lifetimes. One thing to bear in mind is that when GPIO Zero shuts down a device it tries to turn it off (it doesn't leave it's state alone). This is largely a safety feature (so if your robot script crashes out the motors don't keep running!).

So, let's walk through line by line:

```python
led = LED(4)
```

At this point an LED object has been created and the name "led" has been associated with it, so the LED object has a reference count of one.

```python
led.on()
```

The "on" method is called and the LED lights up.

```python
led = LED(18)
```

Now the complicated bit. In an assignment we evaluate the right side first (LED(18)) then perform an assignment to the name on the left. So, firstly a new LED object is created against pin 18. The original LED object is still in existence at this point. Next the name "led" is *removed* from the LED(4) object and its reference count drops to zero. In CPython at least, the garbage collector should rapidly destroy that object now, which will result in the LED switching off (due to the aforementioned safety feature in GPIO Zero). Finally, the "led" named is now assigned to the new LED(18) object. The LED(18) object doesn't turn on because you haven't told it to (and it has no association with the LED(4) object that existed previously).

Hopefully that explains what's happening behind the scenes a bit more. Please note that object lifetimes are a complex thing - if you try something similar with Button you'll find they won't disappear as easily (due to internal references keeping them alive). If you definitely want things to shut down at particular times, use `close()` as Ben noted above.