gpiozero/gpiozero #379: value property documentation

lurch, 2016-06-16:
I just noticed that *many* of the classes in the [documentation](http://gpiozero.readthedocs.io/en/stable/index.html) don't document their `value` property. Given that this is fairly essential knowledge in order to be able to utilise the [sourcetools](http://gpiozero.readthedocs.io/en/stable/api_tools.html) functions properly, IMHO it ought to be added.

waveform80, 2016-08-14:
I had left it out figuring it was beyond "novice" level and thus shouldn't be included in the concrete classes. I'd intended to include it only in the base classes on the assumption that if people were playing with sourcetools they were probably advanced enough to go read the base class stuff (you'll note we don't document `source` or `values` in the concrete classes either). That said, I've noticed `value` is missing in several of the base class docs too (GPIODevice for one).

If we add `value` (and potentially `source` and `values`) to all the concrete class docs (which is easy enough to do) - it will increase the length of the API docs a fair bit.

I'm happy to take guidance on what people think is "too little", "just right", or "too much" when it comes to doc lengths - I don't really have a good eye for judging these things (just look at the [PiCamera class](http://picamera.readthedocs.io/en/release-1.12/api_camera.html) docs). Or to put it another way, *my* preference is that the Recipes (which are deliberately at the front) are the "novice" docs as they let you dive in right away and don't go into depth explaining things - they're more "learn as you play", while API docs should be comprehensive and include everything - they're "read all this and you'll know everything, but you'll need some coffee". However, I'm aware many don't share that view hence why the concrete API docs are currently "abridged" (the base classes shouldn't be - that's my fault!).

lurch, 2016-08-14:
I think `source` and `values` are safe to omit, because once you know the data-type of `value`, they're easy to infer.
But with missing concrete-class-documentation for `value`, it's not obvious (for each class) if it's a binary 0/1 value, a continuous 0 -> 1 value, a -1 -> +1 value, a tuple of values, etc. etc.  (at least, I _think_ I remember that's the reason I opened this PR).
And with things like http://gpiozero.readthedocs.io/en/latest/api_input.html#distance-sensor-hc-sr04 or http://gpiozero.readthedocs.io/en/latest/api_input.html#motion-sensor-d-sun-pir it's not clear what `value` actually represents.

All IMHO... ;-) I'm obviously looking at the API-docs from the PoV of an "advanced user" too!

bennuttall, 2018-09-21:
I've created a branch with more verbose docs (i.e. all properties/methods included). I think I prefer it in general, as everything is there (source, value, values, etc). The only issue is duplication of aliased members.

Compare:
- master: https://gpiozero.readthedocs.io/en/latest/api_input.html#button
- docs-maybe: https://gpiozero.readthedocs.io/en/docs-maybe/api_input.html#button

@waveform80 is it possible to state members for exclusion rather than inclusion?

bennuttall, 2019-02-07:
This can be closed when docs-tidy-up is merged

lurch, 2019-02-11:
I'm going to re-open this, see https://github.com/RPi-Distro/python-gpiozero/pull/737#discussion_r255509849 for an example of where this gets confusing. From looking through the docs, it seems to be impossible to tell that LightSensor has a float-value and MotionSensor has a bool-value.

waveform80, 2019-02-11:
Good point - I keep forgetting stuff like this; will prep a branch to add "value" docs back in.

waveform80, 2019-02-11:
Any opinion on what we should do for composite devices? Is the current documentation (linking to base classes until the user reads about CompositeDevice using a tuple of all subordinate device values) sufficient or should there be something explicit for descendents?

lurch, 2019-02-11:
I think as long as the user can work out what the 'value' for each device within a CompositeDevice is, it's then (relatively) easy for them to deduce what the CompositeDevice's tuple-of-values should look like. As @bennuttall noted in #545 it gets a bit complicated otherwise...

waveform80, 2019-02-11:
Right, I'm just prepping a branch which explicitly documents exactly what the "value" means for each concrete device to make things clear, i.e. I'm manually overriding the docs for "value" where the property itself isn't overridden. This has a slightly ugly consequence vis-a-vis sorting attributes/methods - but I think it might be worth it for the extra clarity it brings - will push it in a mo and get RTD to build the branch.

waveform80, 2019-02-11:
Oh, and we've just realized the value for LineSensor is backwards and wrong (my fault). Urgh.

waveform80, 2019-02-11:
re: LineSensor. Seems I wrote the `line_detected` property to invert `is_active` which means `value` is returning ~0 for black and ~1 for white (it's a smoothed value), so `is_active` is False for black (line) and True for white (no line), which `line_detected` then inverts. It should probably be the opposite: `value` returning 0 for white, 1 for black, and `line_detected` being a straight alias of `is_active` just as things are under Button.

lurch, 2019-02-12:
Related to #652 ?

waveform80, 2019-02-12:
> Related to #652 ?

Looking at the git blame I don't think so; that was about adding the `pull_up` parameter on construction. However, the line_detected property implementation pre-dates that (and has my name next to the commit). I'm guessing I just wasn't thinking when implementing that property and flipped the logic at the wrong level (should've been lower down in value).

PR #744 (2019-02-12): Fix #379 (again ;)
Fixes #379
commit 06be85a2 (2019-02-12): Merge pull request #744 from RPi-Distro/value-docs

Fix #379 (again ;)
closed 2019-02-12 by waveform80