gpiozero/gpiozero #380: Motor.active_high

lurch, 2016-06-16:
While re-reading https://github.com/RPi-Distro/python-gpiozero/issues/248#issuecomment-206822622 it occured to me that something similar might be useful for Motor?

If you've wired your motor up to your motor-controller such that calling `my_motor.forward()` actually drives it "backwards" to the way you want it to turn, you can either swap around the motor wires on your motor-controller, or you can swap around the order of the pins in the Motor constructor. Maybe we could also have an `active_high=False` (or `invert_direction=True` or something) init-param that would automatically swap around the pins?
i.e.
```python
    def __init__(self, forward=None, backward=None, active_high=True):
        if not all([forward, backward]):
            raise GPIOPinMissing(
                'forward and backward pins must be provided'
            )
        if not active_high:
            forward, backward = backward, forward
        super(Motor, self).__init__(
                forward_device=PWMOutputDevice(forward),
                backward_device=PWMOutputDevice(backward),
                _order=('forward_device', 'backward_device'))
```

bennuttall, 2016-06-16:
That does make sense...

lurch, 2016-06-16:
What do people think to the reusing (abusing?) of the `active_high` parameter name?

SteveAmor, 2016-06-16:
Is it adding this parameter really necessary?  I would change the forwards and backwards pins around in the code.  Otherwise, it becomes a double negative situation and make it more confusing.

bennuttall, 2016-06-16:
> What do people think to the reusing (abusing?) of the active_high parameter name?

I think it's ok

bennuttall, 2016-06-16:
Actually ... wouldn't you just swap the order of the pins?

```python
motor = Motor(forward=4, backward=14)
```

to

```python
motor = Motor(forward=14, backward=4)
```

lurch, 2016-06-16:
> I would change the forwards and backwards pins around in the code.

> Actually ... wouldn't you just swap the order of the pins?

Yes, you could, that's why I said "or you can swap around the order of the pins in the Motor constructor" in my initial comment.
I was just throwing this out there as a suggestion (and to give Motor a similar API to Servo), that's why I opened it as an issue rather than a pull-request ;-)

lurch, 2016-06-16:
OTOH, if you're using e.g. http://gpiozero.readthedocs.io/en/v1.2.0/api_boards.html#camjam-3-kit-robot and your Motors turn the wrong way, I guess you **have** to rewire your motors (i.e. fix it in hardware) as there's no way to fix it in software.

Which I guess makes this suggestion pretty redundant.

closed 2016-06-16 by lurch