gpiozero/gpiozero #381: Add __version__ attribute to gpiozero

SteveAmor, 2016-06-16:
Is it sensible to add the __version__ attribute to gpiozero so you can use `gpiozero.__version__` to return the current version?

bennuttall, 2016-06-16:
We used to have one ([see?](https://github.com/RPi-Distro/python-gpiozero/commit/a07f89c5619654356d8101be19edbde2e9f3cff9)) but @waveform80 removed it in favour of the method described [here](http://picamera.readthedocs.io/en/release-1.10/faq.html#how-can-i-tell-what-version-of-picamera-i-have-installed).

lurch, 2016-06-16:
Maybe that "How can I tell what version of XXXX I have installed?" could be added to the gpiozero documentation?
Re-opening this issue as a 'reminder'.

bennuttall, 2016-06-16:
I agree

lurch, 2016-06-16:
Hmmm, I've just tried that on my Pi, and both the apt-get installed version and the latest-from-github version report `1.2.0`, which isn't very helpful :-/

( `[gpiozero 1.2.0 (/usr/lib/python2.7/dist-packages)]` and `[gpiozero 1.2.0 (/home/pi/github/lurch/python-gpiozero)]` )

bennuttall, 2016-06-16:
True. We could introduce a policy of changing the version number to `1.4.0-dev` or something as soon as v1.3 is released?

SteveAmor, 2016-06-17:
Yes please. A documentation update would be useful.  I'm happy to create a PR (under Notes?)

SteveAmor, 2016-06-18:
PR #383 added

waveform80, 2016-06-18:
My reason for not bothering (at least in my other projects like picamera, picraft, etc.) goes something like this (my apologies if this comes across as blunt - I don't mean it to be, but it's late and my diplomacy filters aren't working :)

* If you're using a GitHub master clone, I assume you're working in a virtualenv, in which case you know when you're using the master clone because the bash prompt tells you you're in the virtualenv
* If you're using a GitHub master clone and you're *not* in a virtualenv ... you need to learn virtualenv and start using it
* If you're using a GitHub master clone and you *refuse* to use a virtualenv ... you probably deserve all the versioning pain you're encountering ;)

Still, you're quite right that bumping the version after release isn't that big of a deal; it's easy to do and it'll work around this problem, so this gets a thumbs up from me.

lurch, 2016-08-14:
Closing, as the associated PR as now been merged.

closed 2016-08-14 by lurch