gpiozero/gpiozero #382: Deal with one-wire clash properly

bennuttall, 2016-06-16:
As raised in #363 (Failure to add edge detection on GPIO4 using RPi.GPIO pin), it seems we hit a RuntimeError raised by RPi.GPIO when trying to add edge detection on GPIO4 when the one-wire interface is enabled (as presumably the same when it's configured on another pin).

It would be nice if we could catch the error better and display a decent error message.

However, @pelwell mentioned to me that it may be possible to use the GPIO pin as normal in this situation.

waveform80, 2016-08-14:
Any chance someone can post the full traceback of what happens here? If it bombs during `_set_function` we could fairly easily add some checks in there (similar to the ones in `_set_pull`) to the effect that if the pin number is 4 and setting the function failed we should include a note in the exception that gets raised ("unable to set function on pin 4 - do you have 1-Wire enabled?").

It's not a perfect fix, but it'd be a simple first step.

tamboekie, 2017-08-03:
I get the same thing,  and I'm not touching GPIO4 (promise!). I'm using gpiozero v1.4.0.

Something like these lines:
```python
from gpiozero import Button
play_btn = Button(23) 
```

...produces:

```shell
Traceback (most recent call last):
  File "musicbtns.py", line 39, in <module>
    play_btn = Button(23)                   # play/pause button
  File "/usr/lib/python2.7/dist-packages/gpiozero/devices.py", line 95, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/gpiozero/input_devices.py", line 292, in __init__
    pin, pull_up, bounce_time, pin_factory=pin_factory
  File "/usr/lib/python2.7/dist-packages/gpiozero/mixins.py", line 338, in __init__
    super(HoldMixin, self).__init__(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/gpiozero/input_devices.py", line 95, in __init__
    self.pin.when_changed = self._fire_events
  File "/usr/lib/python2.7/dist-packages/gpiozero/pins/__init__.py", line 388, in <lambda>
    lambda self, value: self._set_when_changed(value),
  File "/usr/lib/python2.7/dist-packages/gpiozero/pins/pi.py", line 289, in _set_when_changed
    self._enable_event_detect()
  File "/usr/lib/python2.7/dist-packages/gpiozero/pins/rpigpio.py", line 219, in _enable_event_detect
    bouncetime=self._bounce)
RuntimeError: Failed to add edge detection
```

Configuration:
* Raspberry Pi 1 B+, Revision 0010
* Linux kernel 4.9.35+ #1014 Fri Jun 30 14:34:49 BST 2017 armv6l GNU/Linux
* I am running some neopixels via [rpi_ws281x](https://github.com/jgarff/rpi_ws281x) on GPIO18.
* Same results whether GPIO4 used or not via `config.txt`: `dtoverlay=lirc-rpi,gpio_out_pin=4,gpio_in_pin=3,gpio_in_pull=up`
* Here's my full (messy) script:
[musicbtns.txt](https://github.com/RPi-Distro/python-gpiozero/files/1198249/musicbtns.txt)