gpiozero/gpiozero #384: ComputeModule USB ports

lurch, 2016-06-19:
I've just been looking at https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/pins/data.py#L244

If the ComputeModule is listed as having 2 CSI ports and 2 DSI ports, then surely it should also be listed as having 1 USB port?
(Happy to fix this myself if I get the nod from @bennuttall or @waveform80 )

waveform80, 2016-08-14:
Ooops - good point. Must've missed the (blindingly obvious) USB pins on the SODIMM list :)

closed 2016-08-14 by waveform80

6by9, 2016-08-17:
There is a comment at https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/pins/data.py#L392
>             On the Compute Module this is listed as 0 as the compute module
            itself doesn't have any physical USB headers, despite providing one
            on the I/O development board and having the pins for one on the
            module itself.

That's no longer true.

lurch, 2016-08-17:
Ah, well spotted @6by9 !
#405 ... ;-)

waveform80, 2016-08-18:
Ahh that explains it - I didn't miss the pins, I just made the bizarre choice to count the CSI pins and not the USB ones. Well, let's get this all nice and consistent now...

PR #404 (2016-08-14): Correct number of ComputeModule USB ports
Fixes #384
commit 65285d1b (2016-08-14): Correct number of ComputeModule USB ports

Fixes #384