gpiozero/gpiozero #387: [performance] Button - huge CPU hog

elegantandrogyne, 2016-06-30:
Whenever I instantiate the Button class, I get ca. 35% CPU load per instance:

1. run "sudo python3" in one console and open "top" or "htop" in another one for monitoring
2. find the python3 process in the top list
3. enter in python interactive mode:
`import gpiozero
button1 = gpiozero.Button(4)`
EFFECT: +30...35% CPU load by python3 after instantiation
SHOULD BE: CPU load doesn't rise drastically
4. try creating some more Button instances for different GPIO pins:
`button2 = gpiozero.Button(24)`
EFFECT: the CPU load rises proportionally, causing the Raspberry to run out of resources if too many instances are created
5. try to get rid of the instances:
`button1 = None
button2 = None`
EFFECT: CPU load doesn't drop
SHOULD BE: instances are deleted and the load drops
6. exit the interpreter:
`exit()`
EFFECT: the CPU load finally comes back to normal.

Tested on the old RPi B rev 2 and B+, Raspbian stretch, Python 3.5.2

bennuttall, 2016-06-30:
> run "sudo python3"

Any reason you're using sudo? You shouldn't need it

>  try to get rid of the instances

This method will not necessarily remove your object, as the garbage collector will clean up at an indeterminate time. To ensure the pin is cleared, use `button.close()`.

> Raspbian stretch, Python 3.5.2

Really? You upgraded from Jessie?

Last thing, can you confirm you have `RPi.GPIO` installed, and that's what gpiozero is using? If it's not installed, or not working, or you're in a virtualenv and haven't installed it, then gpiozero will fall back to using an alternative (RPIO, pigpio) or even gpiozero native pins.

bennuttall, 2016-06-30:
Also see #227

elegantandrogyne, 2016-06-30:
As for running with sudo:
`Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/gpiozero/pins/native.py", line 63, in __init__
    self.fd = os.open('/dev/mem', os.O_RDWR | os.O_SYNC)
PermissionError: [Errno 13] Permission denied: '/dev/mem'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.5/dist-packages/gpiozero/devices.py", line 122, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/gpiozero/input_devices.py", line 268, in __init__
    super(Button, self).__init__(pin, pull_up, bounce_time)
  File "/usr/local/lib/python3.5/dist-packages/gpiozero/mixins.py", line 326, in __init__
    super(HoldMixin, self).__init__(*args, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/gpiozero/input_devices.py", line 80, in __init__
    super(DigitalInputDevice, self).__init__(pin, pull_up)
  File "/usr/local/lib/python3.5/dist-packages/gpiozero/mixins.py", line 161, in __init__
    super(EventsMixin, self).__init__(*args, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/gpiozero/input_devices.py", line 37, in __init__
    super(InputDevice, self).__init__(pin)
  File "/usr/local/lib/python3.5/dist-packages/gpiozero/devices.py", line 368, in __init__
    pin = DefaultPin(pin)
  File "/usr/local/lib/python3.5/dist-packages/gpiozero/pins/native.py", line 209, in __new__
    cls._MEM = GPIOMemory()
  File "/usr/local/lib/python3.5/dist-packages/gpiozero/pins/native.py", line 66, in __init__
    'unable to open /dev/gpiomem or /dev/mem; '
OSError: unable to open /dev/gpiomem or /dev/mem; upgrade your kernel or run as root`

(using kernel 4.4.14; /dev/gpiomem is 600 owned by root:root - I may redefine that in udev)

I have a program which doesn't work too well with py3.4 (some problems with argparse), works nicely on py3.5 (which is not in jessie's repo though, so I moved on to stretch...).

Nailed it! RPi.GPIO was not installed and I didn't even know about it... after installing the load is nice.
Adding a dependency should fix it.

closed 2016-06-30 by elegantandrogyne

bennuttall, 2016-06-30:
> OSError: unable to open /dev/gpiomem or /dev/mem; upgrade your kernel or run as root`

@waveform80 does gpiozero native pins use /dev/mem not /dev/gpiomem? gpiomem doesn't require root. Or is this related to it being Stretch not Jessie?

> Nailed it! RPi.GPIO was not installed and I didn't even know about it... after installing the load is nice.
Adding a dependency should fix it.

gpiozero will work "out of the box" for regular users, because RPi.GPIO is installed by default. Developers using virtualenv, or starting with Jessie Lite, will have to be aware of the "bring your own GPIO back-end" situation. We ought to do a better job of pointing that out in the docs.

So I won't merge #388 because it's not actually a dependency (see comments in the PR).

Thanks

lurch, 2016-06-30:
> does gpiozero native pins use /dev/mem not /dev/gpiomem?

Looks like it tries both https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/pins/native.py#L58

> gpiomem doesn't require root.

In the comment previous to yours, @elegantandrogyne said "/dev/gpiomem is 600 owned by root:root" so I guess the setup of gpiomem in Stretch is different to in the officially-supported Jessie.

> We ought to do a better job of pointing that out in the docs.

I wonder if that'd be another place to use https://docs.python.org/2/library/warnings.html ? Perhaps something like "No GPIO backends found, defaulting to a (slow) pure Python backend"

waveform80, 2016-08-14:
Yeah, `NativePin` is horribly slow when it comes to polling at the moment. I should find some time to speed it up by using epoll instead of Python threads - however, this isn't a major priority given it's literally just a fallback implementation at the moment (it becomes a little more interesting with the pypy use-case, but I suspect pigpio ought to work happily with pypy given it's just sending stuff over a socket - or if it doesn't, it should be trivial to make it work).

Anyway, that's the reason this was horrifically slow. As for the warning, if anyone wants to implement it the best place is probably in NativePin's constructor (`__new__` if I recall) - just throw a warning there ("Using slow NativePin") and Python's default warning filter should ensure it doesn't spam the console.