gpiozero/gpiozero #391: Pidee board

bennuttall, 2016-07-08:
The [pidee](http://theworkers.net/pidee) board looks interesting. I have a couple (sent one to you, @waveform80). It features eight dip switches, a push button and three LEDs. Would be good to provide a class for it.

| Component | pin |
| --- | --- |
| button | 18 |
| dip 1 | 6 |
| dip 2 | 5 |
| dip 3 | 25 |
| dip 4 | 24 |
| dip 5 | 23 |
| dip 6 | 22 |
| dip 7 | 27 |
| dip 8 | 17 |
| red led | 13 |
| green led | 12 |
| yellow led | 19 |

Thoughts:

- Dip switches are a perfect example of `ButtonBoard`
- Dip switches are numbered 1-8 on the board, so we might want to offset the index in the API somehow
- Having the push button as `[0]` and the dip switches as `[1:8]` would be a cunning trick but that would break things like `all()` and `any`
- LEDs are traffic light colours but not in traffic light order (red, green, yellow) so might not make sense to use `TrafficLights` object for these, just a generic composite

rc1, 2016-07-08:
_Hi Ross from The Workers here. This is great_

We are changing the LEDs from RGY to RGB in the next batch. RGB makes it easier to distinguish the LEDs as all on when looking a them from a distance.

lurch, 2016-07-08:
The (not yet merged) ButtonBoard is based on CompositeDevice, which means as well as doing:
```python
dipswitches = ButtonBoard(6,5,25,24,23,22,27,17)
print(dipswitches[0].pressed)
print(dipswitches[7].pressed)
```
you can also do:
```python
dipswitches = ButtonBoard(dip1=6, dip2=5, dip3=25, dip4=24, dip5=23, dip6=22, dip7=27, dip8=17)
print(dipswitches.dip1.pressed)
print(dipswitches[7].pressed) # this style still works
```
(note that I've missed out `_order` here, because alphabetical sorting is the order we want anyway)

I've typed this in from memory, so apologies if I've got some of the syntax wrong.

So the implementation of a `Pidee` class might be _something_ like:
```python
class Pidee(CompositeDevice):
    def __init__(pwm=False):
        LEDClass = PWMLED if pwm else LED
        super(CompositeDevice, self).__init__(
            button=Button(18),
            dipswitches=ButtonBoard(dip1=6, dip2=5, dip3=25, dip4=24, dip5=23, dip6=22, dip7=27, dip8=17),
            red=LEDClass(13),
            green=LEDClass(12),
            yellow=LEDClass(19),
            _order=('button', 'dipswitches', 'red', 'green', 'yellow')
        )
```
Or given that it contains LEDs, perhaps it should inherit from CompositeOutputDevice rather than CompositeDevice?

bennuttall, 2016-08-31:
Ok so now `ButtonBoard` is in, we can probably throw something together based on @lurch's ideas above.

Note that `ButtonBoard.pressed` is not implemented. Instead, I think `tuple(ButtonBoard.value)` or similar must be used.

lurch, 2016-09-01:
There *is* `is_pressed`, if that helps? https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/boards.py#L194

bennuttall, 2016-12-02:
Hi @rc1 did the new revision of the Pidee board get made?

closed 2018-09-20 by bennuttall