gpiozero/gpiozero #393: TimeOfDay

thegnnu, 2016-07-23:
QUESTION Can the "timeOfDay" be used with LED or Buzzer as I have tried using the LED but on hour change nothing changes and the  LED carries on flashing due to the activated and deactivated giving it a 5v or 0 never just one of them being actioned. TerryR

lurch, 2016-07-23:
It'll be easier to debug what's going wrong if you actually paste the complete script you're using... ;-)

thegnnu, 2016-07-23:
I have used several scripts this is the latest. I was trying to turn the backlight LED from a 20x4 LCd off at night time and on during the day but that script just gives me a flashing LED so cut down to this.
Regardless of the time led light up and never goes out and it prints both messages.
Yes I do run the program just before the hour and just after. I have used while True: as well.

from datetime import time
from gpiozero import TimeOfDay, LED
from signal import pause
red = LED(18)
morning = TimeOfDay(time(15), time(16))
morning.when_activated = red.on()
print("Lamp on")
morning.when_deactivated = red.off()
print ("Lamp off")
pause()

bennuttall, 2016-07-23:
The problem is unrelated to `TimeOfDay`.

The way `when_activated` works is you pass it a reference to a function it's supposed to run later. You don't call the function, you just pass its name, i.e. `red.on` not `red.on()`. What you're doing is running the function immediately, then setting `when_activated` to the return value of the function (i.e. `None`), disabling the callback.

Working code would be:

```python
from datetime import time
from gpiozero import TimeOfDay, LED
from signal import pause

red = LED(18)
morning = TimeOfDay(time(15), time(16))

morning.when_activated = red.on
morning.when_deactivated = red.off

pause()
```

Also the `print` lines in your code will not work as intended. There's nothing tying the print calls to the activation. Your code is set up to do the following:

1. Turn LED on and set "when activated" to `None`
2. Print "Lamp on"
3. Turn LED off and set "when deactivated" to `None`
4. Print "Lamp off"
5. Pause, waiting for activation that has no effect.

If you really want print calls like this using callbacks, you would have to create custom functions bundling the two actions together, e.g:

```python
def light_on():
    red.on()
    print("red on")

morning.when_activated = light_on
```

However an alternative approach may suit your case better:

```python
while True:
    morning.wait_for_active()
    red.on()
    print("Lamp on")
    morning.wait_for_inactive()
    red.off()
    print ("Lamp off")
```

bennuttall, 2016-07-23:
I've not used `TimeOfDay` before. Just having a play. @waveform80 any reason these callbacks aren't working for me? I'm using local time and checking against `datetime.now()`.

```python
t = TimeOfDay(time(23, 44), time(23, 45), False)
t.when_activated = lambda: print("Hello")
t.when_deactivated = lambda: print("Bye")
```

Can confirm `t.value` goes between `False`, `True` and `False` again as expected. Just no callback fired. Also tried `def foo(): global a; a = 100` and setting `foo` as callback. Same issue.

lurch, 2016-07-23:
I can't remember the details right now (or on which SD card I was playing with this on - I'll try to find up the right one tomorrow) but ISTR finding a problem with `TimeOfDay` a couple of months ago. (I emailed @waveform80 about it at the time, but he was too busy with picamera stuff)

I can't remember if it was the same problem that @bennuttall has found, but IIRC I came up with a partial fix, but a proper solution depended on being able to wait on a 'union of events' (which there doesn't seem to be any built-in Python support for), rather than just waiting on a single event as all the gpiozero code does currently...

thegnnu, 2016-07-24:
Thank you Ben & Andrew I believe the def function will work for me as I wanted to turn a LCD LED off at 11pm and on at 8am. will try all the code later today after F1. Terry

thegnnu, 2016-07-26:
Sorry for delay my Zero passed away during the night running a LED test if you put the GPIO pin on the led not the resistor that is what happens.
I have done this script
from datetime import time
from gpiozero import TimeOfDay, LED
red = LED(18)

while True:
    morning = TimeOfDay(time(15), time(16))
    morning.wait_for_active()
    red.on()
    print("Lamp On")
    morning.wait_for_inactive()
    print ("Lamp Off ")
    red.off()
I got the following results.
1. run scripts before 15H utc and nothing on screen and LED does not light  so do not know if it running correctly.
2. run script after 15H utc LED light and sceen printout Lamp On
3. Leave scripts running pass 16H nothing changes on screen and LEd still stay's lit.
4. run script after 16H utc nothing printed on screen and LED does not light  so do not knowif  it running correctly.
I have grabbed the screen text after pressing CTRL C if you require it.

TerryR

lurch, 2016-08-04:
Now that I think about it some more, I think I remember that the problem was that when you construct the TimeOfDay object, the initial value of `active` is True or False as appropriate, but there's then nothing to "tick it over" from active->inactive (or vice-versa) when the next start-time or end-time is reached (and hence no callbacks get fired).
IIRC I **did** fix it to work properly on one of my private branches (and I did lots of testing to verify that it worked correctly in all scenarios). I'm busy this weekend and all next week, so I'll try to dig it out week-after-next.

thegnnu, 2016-08-31:
Has this bug been fixed in the new 1.3 release

lurch, 2016-08-31:
Nope, I haven't had any further time to look at it yet.

commit 3fe00d40 (2021-03-14): Implement polling for internal devices

And clean up event handling generally. All the logic to do with wrapping
callbacks, and ensuring watch threads is set up should be handled by the
EventsMixin or other helper classes (in this case a property analogue
called "event").

Fixes #393
commit b8b781a0 (2021-03-14): Implement polling for internal devices

And clean up event handling generally. All the logic to do with wrapping
callbacks, and ensuring watch threads is set up should be handled by the
EventsMixin or other helper classes (in this case a property analogue
called "event").

Fixes #393
PR #941 (2021-03-14): Implement polling for internal devices
And clean up event handling generally. All the logic to do with wrapping callbacks, and ensuring watch threads is set up should be handled by the EventsMixin or other helper classes (in this case a property analogue called "event").

Fixes #393
closed 2021-03-14 by waveform80