gpiozero/gpiozero #394: State of pin changed by input device

MaxValue, 2016-07-26:
I wanted to make a script to switch a pin on or off with the ability that it holds the state after the script finishes/exits. I solved this by changing the DefaultPin:

```python
class SleepNoMore(gpiozero.devices.DefaultPin):
	def close(self):
		pass

gpiozero.devices.DefaultPin = SleepNoMore
```

It works very well if I alter the state using DigitalOutputDevice but the pin switches back to ON if i use it as DigitalInputDevice. I found out this happens somewhere in the `__init__` but i couldn't trace it down yet. Here is the complete script (so you can test this):

```python
import argparse, sys, gpiozero

parser = argparse.ArgumentParser()
group = parser.add_mutually_exclusive_group()
group.add_argument('-O', '--off', action='store_true', default=False, help="turns the 5 Volt switch off")
group.add_argument('-I', '--on', action='store_true', default=False, help="turns the 5 Volt switch on")
args = parser.parse_args()

class SleepNoMore(gpiozero.devices.DefaultPin):
	def close(self):
		pass

gpiozero.devices.DefaultPin = SleepNoMore

if args.off:
	gpiozero.DigitalOutputDevice(13, initial_value=False)
elif args.on:
	gpiozero.DigitalOutputDevice(13, initial_value=True)
else:
	current_state = gpiozero.DigitalInputDevice(13).is_active
	sys.stdout.write(str(current_state))
```

Please pardon my ignorance if I missed some software detail I should be knowing.

bennuttall, 2016-07-26:
Interesting!

lurch, 2016-07-26:
This is *kind of* related to discussions in #118 and #323 and #39 

I'm fairly sure there was also another similar discussion, where @waveform80 recommended using RPi.GPIO instead of gpiozero for scripts where you want the GPIO state to persist after the script finishes running, but I can't seem to find that right now.

Have you got any external hardware attached to GPIO 13? IIRC a pin in input mode (with nothing connected) will always be inactive by default. If your external hardware is active-low, then you'll need to set `pull_up=True` in the `DigitalInputDevice` constructor.

MaxValue, 2016-07-27:
Hmm.
#39: The way the cleanup operates is ok in my case. I manage this by changing the DefaultPin as mentioned above.
#118: I read this often and I still can't put the information to use. Should I directly alter firmware device-tree configurations using python? I could, but this would be a very dirty solution IMO.
As for #323: In my case it would be really useful to have a `final_value` parameter.

I already tried the same feat using RPi.GPIO:
```python
import argparse, sys
import RPi.GPIO as GPIO

parser = argparse.ArgumentParser()
group = parser.add_mutually_exclusive_group()
group.add_argument('-O', '--off', action='store_true', default=False, help="turns the 5 Volt switch off")
group.add_argument('-I', '--on', action='store_true', default=False, help="turns the 5 Volt switch on")
args = parser.parse_args()

GPIO.setwarnings(False)
GPIO.setmode(GPIO.BCM)

if args.off:
	GPIO.setup(13, GPIO.OUT, initial=GPIO.LOW)
elif args.on:
	GPIO.setup(13, GPIO.OUT, initial=GPIO.HIGH)
else:
	GPIO.setup(13, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)
	current_state = GPIO.input(13)
	sys.stdout.write(str(current_state))
```
But it behaves exactly the same: Switching works good, but reading changes the hardware state.

There is circuitry attached to port 13, but it has no pull-up resistor applied to it.

To sum it up I think the problem is somewhat like: the RPi.GPIO module reads the default value on each initialization of a class instance and if the hardware state does not equal the firmware default state, it changes the hardware state.

lurch, 2016-07-27:
> I already tried the same feat using RPi.GPIO: ... But it behaves exactly the same:

I guess at least that shows that it isn't a problem specific to GpioZero ;-)

> reading changes the hardware state. ... There is circuitry attached to port 13, but it has no pull-up resistor applied to it.

Then I'm not sure what it is that you're trying to do, or what you expect to happen?
When a GPIO pin is in input mode, it'll report the state of whatever the external circuitry is driving  it to.

Could you clarify the problem? And/or what it is that you're trying to do?

MaxValue, 2016-07-27:
> When a GPIO pin is in input mode, it'll report the state of whatever the external circuitry is driving it to.

Yes, that I am aware of.

> Could you clarify the problem? And/or what it is that you're trying to do?

Imagine a LED connected to port 13. If the LED should be switched on, the script has the job to start, turn the LED on and then quit. The LED should stay on now, regardless of the script running or not.
If the user wants to read the status of the LED the script has the job to start, read the status, return it and then quit. The LED should stay the way it was before the script read its status.

lurch, 2016-07-27:
> If the user wants to read the status of the LED the script has the job to start, read the status, return it and then quit. The LED should stay the way it was before the script read its status.

Ahhh. AFAIK you can't read the output-value of a GPIO pin by switching it to input-mode. I'm not familiar with the low-level hardware (or the low-level software that controls it), but perhaps @bcroston (the author of RPi.GPIO) knows a way of doing what you're trying to accomplish.

I guess the other option would simply be to have a long-running script, rather than a script which repeatedly launches and exits \*shrug\* 
Or perhaps you could connect gpio13 to another gpio-pin, and use the second gpio-pin to read gpio13's current output-value?

lurch, 2016-07-27:
> I think in a software-hardware environment a software/program/module should never change the state of hardware devices only because the software quits or restarts. The software should act as if it wasn't there.

I think part of the 'philosophy' of gpiozero is to be predictable (remember that its primary target is beginners), by resetting the hardware to a known state at program-start (via the `initial_value` parameters) and at program-exit (via the `close` functions). If that doesn't fit with your way of working, then perhaps it's not the right library for you.

MaxValue, 2016-07-27:
> but perhaps @bcroston (the author of RPi.GPIO) knows a way of doing what you're trying to accomplish.

Thanks I will contact him, didn't know he uses github too.

> the other option would simply be to have a long-running script

I thought of this too, but the script would then have to communicate with other processes (that's like - super advanced stuff). It should just be an easy to handle interface for other programming languages (TCL and in the future Java).

> you could connect gpio13 to another gpio-pin

Unfortunately I can't: we have a standardized shield attached to the pi and I don't have the authority to alter the blueprint of it.

> perhaps it's not the right library for you.

Yeah I thought so too, but there aren't any as well documented solutions out there as gpiozero. At least none that I know of.

I understand that you want to help and I want to express my thankfulness for your quick responses and wide-ranging solutions on my problem. I mean it.

lurch, 2016-07-27:
I haven't looked at their documentation, but I believe [pigpio](http://abyz.co.uk/rpi/pigpio/index.html) and [wiringpi](http://wiringpi.com/) are also pretty full-featured GPIO libraries.

And of course the ultimate cross-language way to control the gpio pins (without relying on other scripts or libraries) is to just use [sysfs](https://github.com/torvalds/linux/blob/master/Documentation/gpio/sysfs.txt).

MaxValue, 2016-07-29:
Thank you for suggesting alternatives.

I will mark this as closed.

closed 2016-07-29 by MaxValue

waveform80, 2016-08-14:
> Ahhh. AFAIK you can't read the output-value of a GPIO pin by switching it to input-mode. I'm not familiar with the low-level hardware (or the low-level software that controls it), but perhaps @bcroston (the author of RPi.GPIO) knows a way of doing what you're trying to accomplish.

That is correct - however, you *can* read the output value of a GPIO pin while it's in *output* mode (this is why OutputDevice has a readable `value` property - and no, it's not just querying the last thing the script set it to, it's actually reading the hardware register driving the GPIO).

Or to put it another way, try changing the pen-ultimate line of your script to:

```python
    current_state = gpiozero.DigitalOutputDevice(13).is_active
```

> I think in a software-hardware environment a software/program/module should never change the state of hardware devices only because the software quits or restarts. The software should act as if it wasn't there.

I'm afraid I have to disagree there; the philosophy is sound, but the practical reality is different. Consider a class full of beginners building robots. Their scripts frequently go wrong (because they're beginners), and frequently when the motors are already running at which point an anticipated short burst of movement becomes a continual running of the motor and chaos ensues (during which Pi's run off tables, robots crash to the floor, components go flying left and right, etc. etc.)

Having experienced this, I'm quite keen to keep the current behaviour, at least as a default :) That said, I'm definitely more convinced of the need for a `final_value` parameter in the library; there's obvious use-cases for it.

MaxValue, 2016-08-16:
Thank you for your insights!

I tried this new solution but it has the exact opposite effect: If i read out the state after turning the "LED" on, it turns off. This is because of the default value of the `initial_value` parameter, i think.

I am currently playing with the thought of directly reading out the hardware device via the /sys/class/gpio devices/files(?), but this is kinda hairy in python.

Yay for (maybe) convincing you of the optional  `final_value` parameter.

waveform80, 2016-08-16:
> I tried this new solution but it has the exact opposite effect: If i read out the state after turning the "LED" on, it turns off. This is because of the default value of the initial_value parameter, i think.

Ahh, you're absolutely right - my mistake. Set `initial_value` to `None` (which means "leave the device as you find it" as opposed to `True` or `False`, the default, which force the device to the corresponding state).

MaxValue, 2016-08-16:
Oh my god, i didn't know one can do that. I already thought of this (i would program my device classes like this) but i didn't check the source back then and now i didn't want to summon any strange errors or device states by supplying `None`.

If this works then this is the cleanest - currently possible - solution IMO.

I will try this out when i am back at my desk.

lurch, 2016-08-16:
You don't need to go digging through the source, just digging through the docs ;-)
http://gpiozero.readthedocs.io/en/latest/api_output.html#gpiozero.OutputDevice

MaxValue, 2016-08-17:
Oh. Another proof of A: I can't read and B: I do not have the required engineering mindset to come up with such a solution. I just straight thought "If I want to read something out I need an InputDevice". Well.

As for the solution: It friggin works! Thank you very much to everyone helping me!

waveform80, 2016-08-17:
No prob, glad you got it working! I agree that it is rather strange to use an OutputDevice to read the state of the pin. It might help to think of the class as representing that which is attached to the pin. The LED is an output device, ergo you use the OutputDevice class (well, ideally, the LED class which derives from OutputDevice, but you get the point :-).

It's an interesting case - would I be right in thinking you were experienced with other GPIO libraries already? The reason I ask is I think this might be a symptom of thinking about things at the pin level (which is what all the existing libraries do) rather than thinking about things at the device-attached-to-the-pin level (which is what GPIO Zero tries to do).

It is a different mind-set but I think I can speak for most of us when I say we're reasonably confident it's a more intuitive approach for people completely new to the subject. Unfortunately that does it'll be a little unintuitive for people used to the existing libraries (and there's still plenty more work to do on our docs - I need to add the `value` property to most of the concrete classes like LED).

MaxValue, 2016-08-17:
I agree, this view of things helps with hardware applications. 

No, this is my "first GPIO library". I started with rPI.GPIO, but quickly saw that i needed multiple threads for event based functionality and a more object-oriented (pythonic) approach. A journey through google showed me that this is the library to use (active development, broad userbase and usecases, good documentation).

I was a victim of tunnel vision and only focused my mind on what i want to achieve and not the grand idea behind it.

What i learned through my adventures with Raspberry Pi is: always see the objective goal/output (what you want to achieve) of the program, not just the technical intricacies alone.

Regarding the docs: Pat yourself a little on the back, because these are already one of the more human-readable and detailed docs i saw. Also i really appreciate the cross-linking inside the docs.