gpiozero/gpiozero #406: Legal headers

bennuttall, 2016-08-18:
```
(13:28:17) waveform: incidentally ,there's something we ought to do - I know you're going to hate it, but we ought to do it nonetheless: legal headers at the top of each source file in gpiozero
(13:28:37) ben_nuttall: oh god legal headers
(13:29:54) waveform: I don't like 'em either but I do them anyway, because they do have legal weight
(13:29:57) ben_nuttall: Meh
(13:30:12) waveform: yeah - figured you'd say something like that ;)
(13:30:48) ben_nuttall: It's not the end of the world
(13:31:05) waveform: yeah - just didn't want to horrify you when I merge a patch which just consists of legal comments :)
(13:31:24) ben_nuttall: The project has expanded in ways I didn't even dream of, this isn't a big deal for me.
(13:31:27) waveform: (well, it'll horrify you I'm sure - just don't revert it ;)
(13:31:36) ben_nuttall: :P
```

lurch, 2016-08-31:
Add to the 1.4 milestone? ;)

waveform80, 2016-08-31:
Yeah, may as well - it's a five minute job

closed 2019-02-10 by bennuttall