gpiozero/gpiozero #408: Getting AttributeError: 'module' object has no attribute 'I2'

Robert-Chu, 2016-08-23:
Hi! All,

     I did the "install" and update. When executing the python through "python xxx.py", I got the AttributeError of no attribute 'I2'. This is from the following statement: 
     from gpiozero import LED.
How can I fix it? Thanks for the answering in advance.

Regards,
Robert.

lurch, 2016-08-23:
What exactly do you mean by "install and update"? What did you name your `.py` file? What does `python --version` report? What does http://gpiozero.readthedocs.io/en/latest/notes.html#how-can-i-tell-what-version-of-gpiozero-i-have-installed (typed into the REPL) report? Which OS are you running on your Pi? Which model of Pi?

Robert-Chu, 2016-08-23:
Hi! Andrew, 
     It turns out if I use "python3" to compile, I don't have any error. Thanks for your prompted response.
Regards,Robert.

lurch, 2016-08-24:
Would you like help to fix your script not working with python2, or can this issue be closed?

(and BTW `python` / `python3` are interpreters, not compilers ;-) )

bennuttall, 2016-08-24:
You *could* argue that they're compilers, seeing as Python compiles at runtime :P

closed 2016-09-06 by bennuttall