gpiozero/gpiozero #409: code warning for ADC's (MCP 3208, MCP 3304)

pcopa, 2016-08-24:
FYI, I've flashed a new SD card with the latest versions and even though everything is working fine, I'm getting the following warning when starting the code:

[image]

pcopa, 2016-08-24:
Sarting over new. Cut and past not working across platforms for me.

FYI, I've flashed a new SD card with the latest versions and even though everything is working fine, I'm getting the following warning when starting the code:

"Warning (from warnings module):
  File "/usr/lib/python2.7/dist-packages/gpiozero/spi.py", line 416
    'software (error was: %s)' % str(e)))
SPISoftwareFallback: failed to initialize hardware SPI, falling back to software (error was: [Errno 2] No such file or directory)"

bennuttall, 2016-08-24:
Hi @pcopa 

> (Why is my picture not showing?)

I don't know why, but looking at the URL of the image, it's a 1px x 1px image. I'm guessing that's not what you uploaded so it must have got lost somewhere in the tubes.

> SPISoftwareFallback: failed to initialize hardware SPI, falling back to software (error was: [Errno 2] No such file or directory)"

This is the expected behaviour in your case.

Since the last release, we added a software fallback for SPI. If you have SPI enabled in raspi-config AND you have the spidev Python module installed, then gpiozero uses hardware SPI. If not, it uses a pure Python implementation (you can even change the pins used to any set of GPIOs). See http://gpiozero.readthedocs.io/en/v1.2.0/api_spi.html

The warning you're seeing is a [warning](https://docs.python.org/3/tutorial/errors.html), not an [exception](https://docs.python.org/3/library/warnings.html). You can [disable warnings](http://stackoverflow.com/questions/14463277/how-to-disable-python-warnings).

pcopa, 2016-08-29:
Thanks for the explanation. will close the comment.

closed 2016-08-29 by pcopa