gpiozero/gpiozero #419: Servo enhancements

lurch, 2016-08-31:
After *much* discussion (in #248 ), a minimal-version of a Servo class was added in release v1.3.0 of GpioZero.

However it doesn't include **everything** that people had been asking for, so this is a new 'clean' issue for people to discuss adding things they might want... (see #248 for possible inspiration).

lurch, 2016-09-19:
I bought a (cheap) continuous-rotation servo recently, and as I suspected, a centre-value keeps it still, negative values determine the speed in one direction, and positive values determine the speed in the opposite direction. Each output-value results in a fixed speed, with larger values giving bigger speeds. So perhaps a `ContinuousRotationServo` class should have an API similar to `Motor`? Unlike with a motor you can't reverse the rotation-direction of a servo by swapping wires over (#380), so it'll also need an `invert_direction` or `active_high` init-parameter.

waveform80, 2016-09-26:
> ... Unlike with a motor you can't reverse the rotation-direction of a servo by swapping wires over (#380), so it'll also need an invert_direction or active_high init-parameter.

Interesting. In `AngularServo` I noted some servos go clockwise with greater pulse widths while some go counter-clockwise. To enable both to be represented by the same angle measurements, I permitted min_angle to be *greater* than max_angle in the constructor. Example:

```python
>>> clockwise_servo = Servo(21, min_angle=-45, max_angle=45)
>>> counter_clockwise_servo = Servo(20, min_angle=45, max_angle=-45)
>>> clockwise_servo.angle = 30
>>> counter_clockwise_servo.angle = 30
```

At this point both servos should be pointing to the same angle. If (and this is another big question) we have some sub-class of `Servo` for continuous-rotation servos that permits a certain rpm to be specified, we could pull the same trick of allowing min and max to be inverted. Not sure how intuitive this stuff is though (or if my current design for `AngularServo` is intuitive either ... but then "min" and "max" are pretty vague concepts in the context of angular position, unlike rotational speed which is more clear cut).

lurch, 2016-09-26:
Sorry, the `invert_direction` comment was only targeted at a possible `ContinuousRotationServo` class, as I was imagining it'd have `forward(speed)` and `backward(speed)` (and `stop()`) methods, like `Motor` does. (Which, if my MotorMixin idea ever comes to fruition, would allow a `Robot` to be constructed from two `ContinuousRotationServo`s)

P.S. I guess you meant `AngularServo` instead of `Servo` in your example code above?

ivansager, 2017-03-08:
A ContinuousRotationServo class would help a great deal!