gpiozero/gpiozero #420: v1.4 release

bennuttall, 2016-08-31:
[v1.4 milestone](https://github.com/RPi-Distro/python-gpiozero/milestone/8)

Release date: 16 November 2016

(more detail to follow)

bennuttall, 2016-11-07:
@waveform80 @lurch @martinohanlon

lurch, 2016-11-08:
I'm at a work conference in Athens all this week (such a hard life :wink: ), but will try to squeeze in a bit of GpioZero-time when I'm back after the weekend.

bennuttall, 2016-11-21:
Waiting on a couple of things, this is delayed for another week or so.

@lurch if you're able to review the SPI refactoring that would be handy.

bennuttall, 2016-12-01:
I've dropped almost all items from the 1.4 milestone and all that remains is:

- #459 Pins re-factor for remote SPI support
- #497 Add pinout.py command-line tool
- #501 Add StatusBoard
- #502 Add XmasTree Board
- #465 Slight refactoring of the SPI pin numbers
- #468 Properly support remote SPI with pigpio
- #469 Update all hierarchy charts in the docs

So, in essence: SPI refactoring, new PiHut boards and pinout cli tool.

The PiHut boards just need a look over from @waveform80, and I'll add some tests and try to add to the graphs. I think @waveform80 is just waiting on a code review of SPI stuff from @lurch. And if we can all check out the cli tool and see if it's good to go, that would be handy.

Any thoughts?

lurch, 2016-12-01:
I'll try to have a look at things over the weekend... (assuming nothing else crops up!)

bennuttall, 2016-12-01:
Thanks @lurch are you free on Monday? (see email)

bennuttall, 2016-12-01:
It;d be good if we fixed #503 too :)

closed 2018-02-22 by bennuttall