gpiozero/gpiozero #422: WeatherHAT support

waveform80, 2016-08-31:
Cyntech very kindly sent a [WeatherHAT](https://github.com/CyntechUK/WeatherHAT) over ages ago; would be nice to add support for it.

Note to self: push on proper I2C support in 1.4 and this (and the Touch4Pi they also very kindly sent) should become possible.

lurch, 2016-09-01:
And I guess with proper I2C support, it might be possible to support a bunch of different Pimoroni HATs too.

waveform80, 2016-09-01:
Yeah - it opens up a lot of possibilities as there's plenty of HATs out there with SPI and I2C

bennuttall, 2016-12-02:
Is there any general I2C support in progress anywhere? Does it rely on an i2c library or can we do it natively?

lurch, 2016-12-02:
There are I2C libraries out there https://duckduckgo.com/?q=python+i2c+raspberry+pi&t=canonical&ia=web but I've never tried using them so I don't know how good they are.
Maybe @Gadgetoid can give us a few pointers on what he uses in the Pimoroni software he writes? :-)