gpiozero/gpiozero #424: Default pin factory

bennuttall, 2016-08-31:
Should we change the default pin factory from RPi.GPIO to pigpio in the upcoming v1.4 release?

lurch, 2016-09-01:
https://github.com/RPi-Distro/python-gpiozero/issues/248#issuecomment-205959582

https://github.com/RPi-Distro/python-gpiozero/issues/248#issuecomment-219285355

And IMHO we'd need much more comprehensive unit-tests before switching the default Pin backend.

waveform80, 2016-09-01:
Indeed - also, now that backend can be switched with nothing more than an environment variable (which people could set in .bashrc if they want to use a particular backend all the time), there's less reason to move the default until we've got comprehensive tests.

lurch, 2016-09-01:
> now that backend can be switched with nothing more than an environment variable

Unfortunately that's not been [documented](http://gpiozero.readthedocs.io/en/v1.3.1/api_pins.html) yet :-/

waveform80, 2016-09-01:
Doh - definitely ought to have that in there somewhere ... maybe the pins chapter header. Maybe link to that from the Servo docs.

bennuttall, 2018-02-24:
Probably not the best idea as it requires pigpiod to be running.

closed 2019-01-15 by bennuttall