gpiozero/gpiozero #427: Different behaviour of background=True parameter of led.blink()

mramanathan, 2016-09-04:
Running this line from Python interpreter and within a script has different behaviours.

From Python interpreter, LED blinks 5 times, no issues

`from gpiozero import LED`
`led = LED(17)`
`led.blink(on_time=1, off_time=2, n=5, background=True)`

But the _same code when executed within a script, LED blinks just once_.

How is background=True parameter influencing this ?

FWIW, I have Pi model B+ Rev1.2 and using Python v2.7.x

/Ram

lurch, 2016-09-04:
You need to use http://gpiozero.readthedocs.io/en/v1.3.1/notes.html#keep-your-script-running :-)
(or set `background=False`)

`background=True` means that it starts up a background thread to do the blinking, and then immediately returns execution-flow to the main program. And as there's no commands after the `led.blink`, the interpreter simply 'falls off the bottom' of the program and terminates.

closed 2016-09-04 by lurch

mramanathan, 2016-09-04:
Before posting, I did know about setting background parameter to False and that satisfied my want. Forgot to mention that.

Was curious to know about the behavior with "True", hence posted the query.
Thanks for the reference link. It's pretty clear now.