gpiozero/gpiozero #428: Pi Zero USB OTG Remote pins

bennuttall, 2016-09-06:
Is it possible to use USB OTG with the Pi Zero to access pins remotely using pigpio?

/cc @gbaman see http://gpiozero.readthedocs.io/en/v1.3.1/api_pins.html#pigpiopin

lurch, 2016-09-06:
I believe @dglaude might have already done this in #359 ?

waveform80, 2016-09-06:
Absolutely possible. The Pi Zero can be configured to emulate an Ethernet interface over the USB OTG socket (I've got mine configured for this) at which point you've got a TCP capable connection so pigpio will certainly work over it. The tricky bit is getting both ends of the connection set up. Not so hard on the Pi Zero (give it a static local-only address like 169.254.64.64) but I've not managed to persuade Windows to talk to it yet. Ubuntu's generally fine (once it detects the link, use NM to configure for link-local only) but Xubuntu's defeated me so far (should be the same ... it's just NM after all, but for some reason it won't detect the link ...).

pieman64, 2016-09-06:
@waveform80 what is NM (linux)?

bennuttall, 2016-09-06:
Awesome. As I think someone described earlier, this is a killer feature. Let's get a guide written up in the docs.

waveform80, 2016-09-06:
@pieman64 Network Manager (the GUI thing that manages network connections under modern distros like Ubuntu and Xubuntu)

waveform80, 2016-09-06:
(well, it's got CLI bits but I suspect most people just know it as the network icon in the system tray)

francesco-vannini, 2016-09-06:
@waveform80 but why do you need to setup a static IP for the Zero? Most OSs support Link-Local address and a non-configured interface will automatically get an IP in 169.254.0.0/16. So does Rasbian on the Zero. I covered most of this in my article here http://garagetech.tips/otg-pi-zero-side-car/ and a few other related articles off my site.

waveform80, 2016-09-06:
You don't *have* to but a lot of the time I want to access some service on the Pi Zero (SSH, or in future given this ticket pigpiod). In that case you need to know the address of the Pi Zero. I could *try* and rely on mDNS (raspberrypi.local) but I've had it get confused because there's usually also a raspberrypi.my.local.domain hanging around on local LAN. I could also look up what IP it's been assigned in the log, but frankly it's easier just to give it a static link-local and remember it :)

gbaman, 2016-09-06:
@waveform80 @francesco-vannini Or what might just be easier is using Bonjour? It is included on Ubuntu by default (I think?) under avahi-daemon, included built in with Mac OS and easy to install for Windows from Apple in their [printer drivers](https://support.apple.com/kb/dl999).
All this works nicely as Pi ships with avahi-daemon as well, so just have to connect to raspberrypi.local. Do have to be careful aren't connecting to any other Pis on main network though.

francesco-vannini, 2016-09-06:
@gbaman yes I go through that in my articles too. But I get the point @waveform80 is making

waveform80, 2016-09-06:
> Awesome. As I think someone described earlier, this is a killer feature. Let's get a guide written up in the docs.

It'll be awesome if the setup can be made *damned simple*. To that end, it'd be nice to have a straight Raspbian distro image configured for OTG out of the box.

> @waveform80 @francesco-vannini Or what might just be easier is using Bonjour? It is included on Ubuntu by default (I think?) under avahi-daemon, included built in with Mac OS and easy to install for [...]

That's the mDNS I was talking about (Bonjour is Linux's implementation of mDNS). If you've got nothing else on the network called raspberrypi it works, but I've had it get confused before (I always have *something* called raspberrypi on the LAN - usually whichever test Pi I last plugged in :) - can't remember which order mDNS goes in NSS - I should probably check that!

dglaude, 2016-09-06:
If you are on a Mac, I would recommend Cluster HAT images:
http://clusterhat.com/setup-software
It has composit Ethernet + Console USB Gadget. And you can pick the name
you want to use.
Because it is based on Jessie-Lite, it require extra work to setup, but you
have the minimum and that is good.

If you are on a PC, then composite USB gadget will fail, and you need to
install Bonjour (maybe from Quicktime or iTune).
So everything seems more difficult suddently.

On the Mac, plug it once and configure Internet Sharing on your Mac from
the new USB Virtual Ethernet interface and to your interface that provide
internet.
That will enable you to update your image and make sure pigpiod is running
at boot time too.
Maybe stuff like expanding the file system would be required too.

Any trick mentionned above that only involve modifying the FAT partition
are easy on PC/Mac too.
Not sure if everything is possible, hostname? static IP? ... but any
progress on that side would be great too.

What would be super great for a distribution could be some read-only file
system once everything is setup.
That would make the PiZero less prone to fail if not shutdown properly...
actually providing an easy way to shutdown would be great too.

I am really happy it got perfected and went in 1.3.

David Glaude

lurch, 2016-09-06:
As well as allowing direct-GPIO control from a PC or Mac, I guess this also makes a cheap & easy way to add extra GPIO pins to an existing Raspberry Pi ;-)

Got a Raspberry Pi project (using GpioZero) and you've run out of GPIO pins? Simply buy a PiZero for $5, attach it in USB OTG mode, configure the appropriate Pin constructors in GPIO Zero, and you've suddenly got an extra 28 pins to play with :-D
Works out slightly more expensive per pin than using traditional MCP23S17 IO expanders, but it's probably easier to set up and more flexible.

francesco-vannini, 2016-09-06:
@lurch along those lines and around about the time piggpio came out, Ryanteck also had a similar idea https://www.kickstarter.com/projects/ryanteckltd/rtkgpio-a-usb-raspberry-pi-compatible-gpio-header

gbaman, 2016-09-06:
And just to throw in one more for good measure, is also @tomhartley's [PiShift](https://github.com/tomhartley/piShift) put together back in March with a very similar end goal.

As for Ryan's board, it has the disadvantage that not all same Pi software/libraries work with it.

waveform80, 2016-09-06:
Sorry - just getting back to this; I meant Avahi is Linux's Bonjour above ... anyway ...

> Because it is based on Jessie-Lite, it require extra work to setup, but you have the minimum and that is good. 

Ahh, it was sounding good up to there; actually it still sounds great for an experienced dev. But ideally we want something for a *total* beginner (really needs to be "works on any platform without exceptions and works out of the box"). Actually that's one of the reasons I rather like the static link-local addressing thing. It *shouldn't* be necessary, it *should* work over mDNS, but in case things don't work you've got a nice easy fallback (a known IP address) rather than leaving the poor teacher scrabbling around with Google.

> Works out slightly more expensive per pin than using traditional MCP23S17 IO expanders, but it's probably easier to set up and more flexible.

Interesting idea! I'll get to the IO expanders eventually though :)

> @lurch along those lines and around about the time piggpio came out, Ryanteck also had a similar idea https://www.kickstarter.com/projects/ryanteckltd/rtkgpio-a-usb-raspberry-pi-compatible-gpio-header

We're all in favour of the Rtk.GPIO - in fact we've recently added some stuff that should make it easier to support (GPIOZERO_PIN_FACTORY env var - sorry, haven't got around to documenting it yet).

> And just to throw in one more for good measure, is also @tomhartley's PiShift put together back in March with a very similar end goal.

Intriguing - not seen that yet - when I've got a spare second I'll take a look!

francesco-vannini, 2016-09-06:
@gbaman yep although Tom's has a bit of a different twist; you wouldn't use it to "extend" you GPIOs

lurch, 2016-09-06:
Yup, and Ryan's sent me an Rtk.GPIO beta board as a thank you for the support I've provided him.

bennuttall, 2016-09-07:
@ukBaz You may be interested in this!

ukBaz, 2016-09-07:
@waveform80 did you get it set-up for Windows? You might find this useful:
https://learn.adafruit.com/turning-your-raspberry-pi-zero-into-a-usb-gadget/ethernet-gadget#if-you-are-using-windows-as-the-host-machine

The key for me getting it working on my Chromebook was specifying a different gateway and static address on the Pi Zero. Then using the Pi Zero's gateway address for both the IP address and gateway address on the Chromebook. Then ssh'ing on to the static IP of the Pi Zero worked a treat

waveform80, 2016-09-08:
@ukBaz not had a chance to look at it again yet (currently I'm working on SPI stuff, partly as penance for breaking something in SPI in the 1.3.0 release ;). I'll be back in Windows before long though for some sense_emu stuff so I'll give it a try then - thanks for the link, that could be the key!

dglaude, 2016-09-08:
> into-a-usb-gadget/ethernet-gadget#if-you-are-using-
> windows-as-the-host-machine

There is other tricks that might be usefull if DHCP is in use: forcing the
MAC address at first startup.
Also, I found that connecting frequently to various Pi with the same
raspberry.local name or the same IP make my SSH security go creazy.
It complain because each time I use it, it save in the known host the
keys... but then if I have same IP (or name) with another machine (or same
machine reinstalled from scratch) it make a mess.
So finding and documenting the right ssh settings to disable that security
might help.

commit ac8b5dbc (2017-06-22): Merge pull request #523 from RPi-Distro/docs-updates

Docs updates: add installing, advanced recipes, remote gpio and remote recipes, close #428, close #434, close #520
closed 2017-06-22 by bennuttall