gpiozero/gpiozero #432: Chevrons?

bennuttall, 2016-09-13:
I notice some examples in docstrings use chevrons, like they're on the shell, and others don't.

For example see the examples in the docs for [Servo](http://gpiozero.readthedocs.io/en/v1.3.1/api_output.html#servo) and [AngularServo](http://gpiozero.readthedocs.io/en/v1.3.1/api_output.html#angularservo).

Any reason?

lurch, 2016-09-13:
I did a recursive-grep, and found that http://gpiozero.readthedocs.io/en/v1.3.1/api_generic.html#gpiozero.Device.close also uses chevrons.

From what I can tell, examples without chevrons are designed to be used in scripts (i.e. standalone .py files), and examples *with* chevrons are designed to illustrate how you might use certain features of gpiozero from within an interactive REPL.

bennuttall, 2016-09-13:
I guess it makes sense if, like with AngularServo, you're printing stuff out to inspect in-between lines of code. Sometimes examples just have `sleep(1)` to demonstrate things being spaced apart in time, not immediately following each other.

bennuttall, 2016-09-19:
I don't think there's any fault in use of chevrons. Closing.

closed 2016-09-19 by bennuttall