gpiozero/gpiozero #435: DistanceSensor has bad distance value when running on Raspberry Pi 3

hiro345g, 2016-09-15:
example (a.py)

```python
#!/usr/bin/env python3
# -*- coding: utf-8 -*-
from gpiozero import DistanceSensor
from time import sleep

sensor = DistanceSensor(echo=26, trigger=19)
while True:
    print(sensor.distance)
    sleep(1)
```

result ('... NG' is comment)

```bash
$ python3 a.py 
0.03545699000358581 .... NG
0.03197880864143371 .... NG
0.026004520654678343 .... NG
0.029134883880615234 .... NG
0.1740727472305298
0.18665557980537414
0.1700421488285065
0.1781851851940155
0.034045257568359376 .... NG
0.04990167260169982 .... NG
0.04959477424621582 .... NG
```
I changed "/usr/lib/python3/dist-packages/gpiozero/input_devices.py ". sample_wait from 0.0 to 0.06.
ref: http://akizukidenshi.com/download/ds/sainsmar/hc-sr04_ultrasonic_module_user_guidejohn.pdf : Max Rep. Rate: 50us

before
```python
        super(DistanceSensor, self).__init__(
            echo, pull_up=False, threshold=threshold_distance / max_distance,
            queue_len=queue_len, sample_wait=0.0, partial=partial
        )
```

after
```python3
        super(DistanceSensor, self).__init__(
            echo, pull_up=False, threshold=threshold_distance / max_distance,
            queue_len=queue_len, sample_wait=0.06, partial=partial
        )
```

result (It looks good)

```bash
$ python3 a.py 
0.1930390655994415
0.19252756834030152
0.18688063859939574
0.18403671383857728
0.18315693855285645
0.18332061767578123
0.18393441438674926
0.18393441438674926
0.1851210880279541
0.18338199734687805
0.18342291712760928
0.1842003929615021
0.18342291712760925
0.18424131274223327
```

bennuttall, 2016-09-15:
Thanks for the report. @waveform80 is currently looking into issues with DistanceSensor so I'm sure this will help. See #385.

bennuttall, 2016-09-17:
It looks like you're in a virtualenv. Is that right? If so, you'll need to install RPi.GPIO in the virtualenv (`pip install rpi.gpio`) otherwise it'll fall back to NativePin which is less reliable.

If you run the following it'll tell you which underlying pin implementation you're using:

```python
from gpiozero.devices import pin_factory

print(pin_factory)
```

That will print the repr of the pin class you're using (RPiGPIOPin or NativePin, for example)

Thanks

dglaude, 2016-09-19:
I believe this should be in the recipes (
https://gpiozero.readthedocs.io/en/latest/recipes.html) and/or pins
documentation (https://gpiozero.readthedocs.io/en/latest/api_pins.html):

If you run the following it'll tell you which underlying pin implementation
you're using:

from gpiozero.devices import pin_factory

print(pin_factory)

That will print the repr of the pin class you're using (RPiGPIOPin or
NativePin, for example)

waveform80, 2016-09-19:
> I believe this should be in the recipes ( https://gpiozero.readthedocs.io/en/latest/recipes.html) and/or pins documentation (https://gpiozero.readthedocs.io/en/latest/api_pins.html):... 

It should be, but the pins API is not stable yet. There's still some modifications I need to make to support "proper" remote SPI and I2C. Those modifications are probably going to entail some changes in this area so I don't particularly want to advertise things like this until they're properly locked down (otherwise we'll be breaking stuff for more people).

With any luck, the next round of modifications *should* be the last; once I'm sure they are we will add stuff like this to the docs.

hiro345g, 2016-09-20:
> It looks like you're in a virtualenv. Is that right? 

I'm not in a virtualenv.

```bash
$ python3 -c 'from gpiozero.devices import pin_factory; print(pin_factory)'
<class 'gpiozero.pins.rpigpio.RPiGPIOPin'>
```

hiro345g, 2016-09-20:
I checked my circuit with a wiringpi program. It looks good.

```python
import wiringpi
import math, time

class HcSr04:
    PULSE_TRIGGER = 0.00001  # 10 μsec
    SPEED_OF_SOUND = 340  # 340 m/sec
    PULSE_TRIGGER_INTERVAL = 0.06  # 60 msec
    TIMEOUT = 2 * 180 * 0.01 / SPEED_OF_SOUND  # 約 0.0106 sec

    def __init__(self, trig, echo):
        self.trig = trig
        self.echo = echo
        wiringpi.wiringPiSetupGpio()
        wiringpi.pinMode(self.trig, wiringpi.OUTPUT)
        wiringpi.pinMode(self.echo, wiringpi.INPUT)
        wiringpi.digitalWrite(self.trig, wiringpi.LOW)
        time.sleep(1)

    def wait_for_echo(self, wait_value, wait_time):
        timeout = False
        start_time = time.time()
        current_time = start_time
        while wiringpi.digitalRead(self.echo) != wait_value:
            current_time = time.time()
            timeout = (current_time - start_time) > wait_time
            if timeout is True:
                break
        return current_time, not timeout

    def read(self):
        time.sleep(self.PULSE_TRIGGER_INTERVAL)
        wiringpi.digitalWrite(self.trig, wiringpi.HIGH)
        time.sleep(self.PULSE_TRIGGER)
        wiringpi.digitalWrite(self.trig, wiringpi.LOW)
        start_time, echo_result = self.wait_for_echo(wiringpi.HIGH, self.TIMEOUT)
        if echo_result:
            end_time, echo_result = self.wait_for_echo(wiringpi.LOW, self.TIMEOUT)
            if echo_result:
                return True, (end_time - start_time) * (self.SPEED_OF_SOUND * 1000 / 2)
            else:
                print("wait_for_echo(wiringpi.HIGH) error")
        else:
            print("wait_for_echo(wiringpi.LOW) error")
        return False, 0

hcsr = HcSr04(19, 26)
for i in range(5):
    result, v = hcsr.read()
    if result:
        print("distance: {0} [mm]".format(v))
    time.sleep(1)
```

SteveAmor, 2017-07-10:
Just reviewing this one again.

http://www.micropik.com/PDF/HCSR04.pdf states:

  we suggest to use over 60ms measurement cycle, in order to prevent trigger signal to the echo signal.

  Given the maths, this makes sense.  I add this comment for the avoidance of doubt but see that the OP did indeed set the time to 60msec while quoting the 50usec max [I would have called it min] rep rate.

  Also to be aware that the measurement cycle should really be at DistanceSensor Class level as reflections can cause issue between multiple distance sensors (objects) firing off without concern for other sensors currently measuring.  Something to think about for multiple instances of distance sensors.

typxxi, 2017-09-23:
The problem still exists and I would like to know when a bugfix or 1.5 will be released cause currently is it useless for us ...
Thanks

commit 53fbaccd (2018-02-19): Re #385, fix #435

This fixes the issue with the DistanceSensor class constantly returning
the maximum distance value. The issue was caused by cab6cc80 which
changed the Pin.when_changed attribute to use weak references to all
handlers. DistanceSensor is the only thing that uses a local function
(not a bound method) as its pin handler. As the pin no longer kept a
reference to it, it disappeared after the constructor was complete and
the echo event never got set.

In addition, it introduces a class-level lock to prevent multiple
distance sensors from interfering with each other (it's the only c [... cut]
closed 2018-02-19 by waveform80