gpiozero/gpiozero #436: Event handler rejection too broad

Marrin, 2016-09-15:
Example:

```python
#!/usr/bin/env python
# coding: utf8
from __future__ import absolute_import, division, print_function
from functools import partial
from gpiozero.mixins import EventsMixin

callback = partial(print, 'Hello')
callback()  # -> prints Hello.

events_mixin = EventsMixin()
events_mixin.when_activated = callback  # Exception!
```

Traceback:
```
Traceback (most recent call last):
  File "test.py", line 12, in <module>
    events_mixin.when_activated = callback  # -> BadEventHandler: value must be a
  File "/usr/local/lib/python2.7/dist-packages/gpiozero/mixins.py", line 209, in when_activated
    self._when_activated = self._wrap_callback(value)
  File "/usr/local/lib/python2.7/dist-packages/gpiozero/mixins.py", line 283, in _wrap_callback
    'value must be a callable which accepts up to one '
gpiozero.exc.BadEventHandler: value must be a callable which accepts up to one mandatory parameter
```

`callback` is a perfectly valid callback that accepts no arguments. Looking at the code you are not testing what the exception message suggests, but simply reject all objects where you can't get the functions signature. Which rejects many valid ones, including `partial` objects from the standard library.  What about trusting the programmer and duck typing here, instead of excessive checking up front and rejecting false positives?

This also means working code can break merely by re-implementing a callback in C.

Why checking anyway?  Just wrap the actual call into a try and print a stack trace if it fails, like most GUI libraries do for callbacks.

lurch, 2016-09-15:
I *guess* the reason for checking up-front, rather than deferring until a `try` block, is that otherwise you don't get the Exception until much later, at which point it might be harder to work out what's going wrong? \*shrug\*  This is all @waveform80 code.

Is it possible to work around your problem using `lambda` functions?

Marrin, 2016-09-15:
@lurch In the use case I've encountered the problem I could use `lambda` but just because the function is fixed.

If I want to use a function or method with no arguments written in C the workaround would be wrapping it in a `lambda` where the readers will scratch their heads why there is a seemingly useless lambda.  It needs to be accompanied with a comment explaining why this isn't useless code that can't be optimized away.

Also if both, function and argument(s) where variable, I would have to re-implement `functools.partial()` in Python. Re-Implementing a very generic, often used function from the standard library feels wrong. Reinventing the wheel seems to be missing the point of having generic functions in a standard library. :-)

lurch, 2016-09-15:
GpioZero is largely targeted at beginners, and designed to be easy to use (and I guess therefore catching errors earlier is useful?). How many beginners are likely to be writing callbacks in C? ;)

I expect the only reason to write callbacks in C would be for faster-performance, in which case you're likely to write the whole application in C?  \*shrug\* I've personally never tried mixing C and Python.

waveform80, 2016-09-15:
Actually, this is going to be one of the things mentioned in the PyconUK talk on Sunday :)

As a quick preview though: yes, I'm aware this causes issues for C callbacks (I did consider this when writing _wrap_callback) but right now I don't particularly care because there's a simple workaround for anyone that advanced: use a lambda (as you've discovered; incidentally I wouldn't describe reimplementing a callback in C "mere" - that's a drastic step, and one very rarely taken in my experience). As @lurch notes we have a firm commitment in the library to keep things as easy as possible for newbies. This includes teachers having to debug in a classroom setting...

Consider the case where we *don't* test the callbacks and assume that all callbacks accept a mandatory parameter, i.e. the originating component. In the case the student forgets the mandatory parameter (bearing in mind for most basic use-cases they don't even use it), where will they get an error? When the callback is assigned? No. When the script is run? No. When the callback is fired by a device, yes ... but the error won't point into their code. The stack trace will be from some point in our library where we attempt to call the callback and fail due to the mismatching prototype (given that the callback will be from a pin-spawned background thread, it's unlikely their script will feature at all in the traceback, although the name of the errant function will). Anyway, the upshot is: this is *hard* for a teacher to debug with 20 screaming kids behind them. There's also the issue that the student needs to remember boiler-plate: the unnecessary parameter (which is an explicit goal of the library to avoid). Hence, the introspection and callback wrapping.

Incidentally, C callbacks *can* work, but they're limited to the "no mandatory parameters" version. This is explained in the implementation comment:

```python
            # We can't introspect the prototype of builtins. In this case we
            # assume that the builtin has no (mandatory) parameters; this is
            # the most reasonable assumption on the basis that pre-existing
            # builtins have no knowledge of gpiozero, and the sole parameter
            # we would pass is a gpiozero object
            return fn
```

So: yes, we're aware of the issue, we've considered the consequences, but I'm afraid our commitment to the most basic use-cases means that more advanced users will have to bear a little extra pain in this particular case if they wish to stay with GPIO Zero.

Incidentally, if you want to make the case that this should be explicitly documented, then I'd certainly agree (although we'd probably avoid mentioning it in the recipes as those really are targeted at beginners).

lurch, 2016-09-15:
Hmmm, I've just had a closer look at what the code is doing in https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/mixins.py#L253

When you do `events_mixin.when_activated = callback` there's no way to 'tell' the callback whether you want to be called with no args, or just a single arg of `self`. So the code inside `_wrap_callback` tries to do `inspect.getcallargs(fn)`, and if that works the callback gets invoked as `callback()`, whereas if `inspect.getcallargs(fn, self)` works, then the callback gets invoked as `callback(self)`.
With a `partial`function, *both* `callback()` and `callback(self)` are valid, so `_wrap_callback` has no idea which should be used.

As a 'fix' to your problem, I guess `_wrap_callback` could be modified so that if the type of `callback` is `<type 'functools.partial'>` then it could just assume that it doesn't take any additional parameters, and always call it as just `callback()` ? i.e. change https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/mixins.py#L258 to
```python
    elif inspect.isbuiltin(fn) or type(fn) == functools.partial:
```

waveform80, 2016-09-15:
@lurch - what the `_wrap_callback` test is looking for is whether the function accepts a *mandatory* parameter. If it's happy to be called with no parameters, we assume that's fine and use that. Otherwise, we attempt to bind a single mandatory parameter (the originating object). If that works, we wrap the callback in a closure to send ourselves along with any callbacks. If that fails it means the callback must require two or more mandatory parameters in which case the user's done something wrong so we throw an error.

This will cause an issue for things expecting a GPIO Zero object as an initial *optional* parameter, but firstly there's little I can do about that, and secondly that's a pretty odd use case (hence why I'm none too bothered leaving it as unsupported, though we could probably make it more clear in the docs that things will break if you try it!).

lurch, 2016-09-15:
So would my `elif inspect.isbuiltin(fn) or type(fn) == functools.partial:` break more than it fixes? Apologies for wading out of my depth ;-)

Marrin, 2016-09-15:
@lurch and @waveform80: Both of you make it sound that (re-)implementing a callable in C is exotic and something no beginner will do.  Maybe, but that is not the problem.  I didn't implement `partial`, I just imported it from some module as any beginner could.  Any `from some_module import a_callable` bears the risk that it won't be usable as callback just because `gpiozero` rejects it, not because it wouldn't work.

Reasons for implementing something in C include needing more direct hardware access (e.g. `RPi.GPIO`) or writing a wrapper around some other C or C++ library (e.g. `pygame` wraps `libsdl`). It's not exotic in general.

I understand the reasoning about testing upfront if the callback won't work.  I can't really follow the decision to reject callbacks where it can't be determined.  You already exclude the builtins from the rule, why not every C function or method?  At least the exception message should be corrected, because it was very confusing to get told the callback has to accept up to one mandatory parameter when I was pretty sure the callback satisfied that restriction. :-)

@lurch: Your fix would work for `functools.partial()` but leaves the problem with other non-Python callables.  Starting to explicitly exclude callables one by one isn't a good idea and it will become tricky to do this test for things outside the standard library.

lurch, 2016-09-15:
>  At least the exception message should be corrected

I guess the problem is that currently `_wrap_callback` makes no distinction between callbacks where the callargs can't be determined (like partial functions), and callbacks where it has more than one mandatory argument (like `def foo(a, b)`). Both 'errors' fall down the same code-path.

I'm not familiar with the [inspect](https://docs.python.org/3/library/inspect.html) module, but perhaps an additional check could be placed before https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/mixins.py#L270 and a different (more appropriate) exception-message generated?

waveform80, 2016-09-17:
I'm going to start in an odd order here, to keep my reply vaguely coherent, sorry about the jumping around!

From @Marrin 
> Reasons for implementing something in C include needing more direct hardware access (e.g. RPi.GPIO) or writing a wrapper around some other C or C++ library (e.g. pygame wraps libsdl). It's not exotic in general.

Sorry, that's still pretty weird. Most of RPi.GPIO could be written happily in pure Python (see native pin implementation which does everything RPi.GPIO does *except* PWM). PWM has to be written in C simply because Python's not fast enough. So, performance is certainly an issue where one might resort to breaking out another language, but being closer to hardware? Not required (Python's got file access, mmap and ioctls built-in - not much else is required for direct hardware access on UNIX platforms).

> I understand the reasoning about testing upfront if the callback won't work. I can't really follow the decision to reject callbacks where it can't be determined. You already exclude the builtins [...]

I'm afraid `inspect.isbuiltin` is poorly titled (and documented). Built-in simply means "not a pure Python function" or "function in shared object". For example:

```python
>>> import inspect
>>> from RPi import GPIO
>>> inspect.isbuiltin(GPIO.setmode)
True
```

From @lurch:
> I guess the problem is that currently _wrap_callback makes no distinction between callbacks where the callargs can't be determined (like partial functions), and callbacks where it has more than one mandatory argument (like def foo(a, b)). Both 'errors' fall down the same code-path.

_wrap_callback does distinguish between these cases. If the callback has more than one mandatory argument it will be rejected with an error. If we can't determine the signature (because it's built-in) we accept the callback, and assume it has no mandatory parameters (because the only thing we'd pass anyway is a GPIO Zero object). However, there's an intriguing coda to this....

Now the interesting bit. From @Marrin 
> @lurch and @waveform80: Both of you make it sound that (re-)implementing a callable in C is exotic and something no beginner will do. Maybe, but that is not the problem. I didn't implement partial, [...]

On implementing things in C being exotic: yes, it's an exotic thing that no beginner will do, because no beginner knows C or for that matter Python. If your definition of beginners includes people from the set likely to write C extensions ... well, my set of beginners looks very different!

However, there's been a doubt nagging at me from the start on this ticket ... that partial callback *shouldn't* be rejected. It's not a built-in (partial is implemented in Python, not C) and it accepts no mandatory parameters so we *should* accept it as a valid callback. I finally got some time to look into this last night, and the cause is that partial's doing something magic with introspection. Try the reproduction example at the top of the ticket in Python 3 (many thanks for posting that - makes this much easier!) .... the extended traceback produced is illuminating:

```python
Traceback (most recent call last):
  File "/usr/lib/python3.4/inspect.py", line 971, in getfullargspec
    skip_bound_arg=False)
  File "/usr/lib/python3.4/inspect.py", line 1956, in _signature_internal
    skip_bound_arg)
  File "/usr/lib/python3.4/inspect.py", line 1951, in _signature_internal
    skip_bound_arg=skip_bound_arg)
  File "/usr/lib/python3.4/inspect.py", line 1884, in _signature_from_builtin
    raise ValueError("no signature found for builtin {!r}".format(func))
ValueError: no signature found for builtin <built-in function print>

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 270, in _wrap_callback
    inspect.getcallargs(fn)
  File "/usr/lib/python3.4/inspect.py", line 1165, in getcallargs
    spec = getfullargspec(func)
  File "/usr/lib/python3.4/inspect.py", line 977, in getfullargspec
    raise TypeError('unsupported callable') from ex
TypeError: unsupported callable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.4/inspect.py", line 971, in getfullargspec
    skip_bound_arg=False)
  File "/usr/lib/python3.4/inspect.py", line 1956, in _signature_internal
    skip_bound_arg)
  File "/usr/lib/python3.4/inspect.py", line 1951, in _signature_internal
    skip_bound_arg=skip_bound_arg)
  File "/usr/lib/python3.4/inspect.py", line 1884, in _signature_from_builtin
    raise ValueError("no signature found for builtin {!r}".format(func))
ValueError: no signature found for builtin <built-in function print>

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 276, in _wrap_callback
[... 6 lines cut ...]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 146, in __setattr__
    return super(GPIOBase, self).__setattr__(name, value)
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 209, in when_activated
    self._when_activated = self._wrap_callback(value)
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 283, in _wrap_callback
[... cut ...]
So, this is a bug in GPIO Zero. Our use of introspection needs to be refined to cope with the case we wind up introspecting a built-in even when our initial test says it's not. I need to dig a bit further into the introspection code to see how this trickle down occurs. Is this a special case of partial or does it apply to anything wrapped (note to self: check the `__wrapped__` mechanics).

In conclusion: I'm afraid the introspection of callbacks is not going away - I don't buy the argument that this hurts any beginners (in fact I'm convinced of the opposite). However, you have definitely uncovered an intriguing bug I need to fix. Thank you!

Marrin, 2016-09-18:
@waveform80: Of course I don't think beginners implement functions or types in C, but again, that doesn't stop them from using them by simply importing them from modules or packages.  It's not exotic to implement modules in C for CPython.  At least not to _use_ such modules because there are plenty of them.

`functools.partial()` comes from the builtin `_functools` module, i.e. written in C _and_ compiled into the CPython interpreter binary.  It is not written in Python.  At least not on the CPython versions I've checked (2.7, 3.3, and 3.4).

lurch, 2016-09-18:
I had a quick look, and AFAICT Py2.7 and Py3.3 do:
```python
from _functools import partial, reduce
```

Py3.4 meanwhile does:
```python
def partial(func, *args, **keywords):
    # python code

try:
    from _functools import partial
except ImportError:
    pass
```
which (IMHO) seems unusual. Isn't the more common way of doing fallback-code like:
```python
try:
    from _functools import partial
except ImportError:
    def partial(func, *args, **keywords):
        # python code
```
?

lurch, 2016-09-18:
...so I've done a bit more poking, and if we want `_wrap_callback` to support partials, I wonder if we need to do something like this just before https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/mixins.py#L265:
```python
    elif type(fn) == functools.partial:
        if inspect.isbuiltin(fn.func):
            return fn
        else:
            try:
                inspect.getcallargs(fn.func, *fn.args)
                return fn
            except TypeError:
                try:
                    inspect.getcallargs(fn.func, *(fn.args + (self,)))
                    @wraps(fn)
                    def wrapper():
                        return fn(self)
                    return wrapper
                except TypeError:
                    raise BadEventHandler(
                        'value must be a callable which accepts up to one '
                        'mandatory parameter')
```
i.e. similar to what the existing code does for regular functions, but 'unwrapping' `.func` and `.args` if we have a partial function.

@Marrin Just out of curiosity, is there any reason you want / need to use a `partial` to provide a gpiozero callback function, rather than using a `lambda` ?

waveform80, 2016-09-19:
> functools.partial() comes from the builtin _functools module, i.e. written in C and compiled into the CPython interpreter binary. It is not written in Python. At least not on the CPython versions I've checked (2.7, 3.3, and 3.4).

Ah, must admit I hadn't checked the source. I'd simply done:

```python
>>> import inspect
>>> from functools import partial
>>> inspect.isbuiltin(partial)
False
```

Which is most curious given the above. Now PyConUK's over, I should have some time to look into this today.

Marrin, 2016-09-19:
@lurch I find `partial()` cleaner and simpler than `lambda` expressions, especially when there is a need to use bound variables, which is a little bit of an ugly hack.  And regarding beginners I think `partial()` can be explained easier than the need for the `text=text` part in `lambda text=text: print(text)` and that there are two different variables called `text` involved and bound and free variables and late binding.

Silly example for comparison:

```python
for i, button in enumerate(buttons, 1):
    button.when_pressed = lambda i=i: print('Button', i, 'pressed.')
    # vs.
    button.when_pressed = partial(print, 'Button', i, 'pressed.')
```

Maybe its just my weird preference to use `partial()` unless `lambda` is really needed to express the function.  I.e. when it is not just partial function application.  `partial()` exists because the late binding in `lambda` expressions is a pitfall in Python.

lurch, 2016-09-19:
Couldn't you make your lambda example less confusing though by doing:
```python
    button.when_pressed = lambda x=i: print('Button', x, 'pressed.')
```
?
Perhaps I find the late-binding of `lambda` less confusing because I simply imagine it as a function without a name? (although maybe I have the wrong mental model)

Marrin, 2016-09-19:
@lurch Yes you can give it another name which involves finding another name for a value which has the same meaning which often makes it hard to find another name, unless the original name did not have a real meaning.

You have the right mental model of `lambda`. An expression evaluated to an anonymous function is exactly what it is.  Many beginners seem to just think of it as some voodoo they don't fully understand and then apply wishful thinking as to when and how the values of free variables are determined.  Without knowing bound and free variables and their distinction.  They are beginners after all. :-)  “Why do all the callback functions created with `lambda` in a loop use the same value”  was a question that appeared regularly in the Python newsgroup when I was still reading newsgroups.

waveform80, 2016-09-20:
> EDIT: Although the above doesn't work if you have a partial wrapping a partial (you get TypeError: <functools.partial object at 0x76a43cf0> is not a Python function), so perhaps partial-function support would need some kind of recursive solution? Possibly messy.

Had a bit of time to dig into this on the train home yesterday. Some observations:

* partial returns an object, not a function. The object is an instance of partial. `isinstance(func, partial)` seems to be a reliable method of detecting such things and works widely; in Python 3.4+ partialmethod is implemented via partial, and wraps is implemented with partial too.
* partial objects are considered "callable" by Python but the reason isbuiltin returns False for them (despite their implementation being in C) is that isbuiltin rather simply tests for instances of BuiltinFunctionType (as partial objects aren't functions they return False, despite being callable).
* in Python 3.5, partials don't appear to be recursively implemented, i.e. if you wrap foo as bar, and bar as baz, baz.func will be foo and baz.args will contain all wrapped args (probably a performance optimization). However, in Python 2.7, partials are nested so yes, we'll need some recursion / iteration here.

bennuttall, 2016-09-20:
Would it be possible to just extend an input device class to overwrite the `when_activated` method to skip the `_wrap_callback` stuff and just send it straight to the pin class to add an event?

waveform80, 2016-09-20:
Better to fix wrap_callback in a manner similar to @lurch's suggestion to properly handle wrapped stuffed. Shouldn't be too hard - I've just pushed something to my clone which ought to handle it (no recursion required, just a little iteration).

waveform80, 2016-09-20:
Just going to add some docs and tests, then I'll PR it

lurch, 2016-09-20:
I'm afraid #455 doesn't _quite_ fix the problem properly. I modified `test_input_partial_callback()` to the following (to check that the args were getting passed in the right order) :
```python
def test_input_partial_callback():
    event = Event()
    pin = MockPin(2)
    callback_args = []
    def foo(a, b):
        callback_args.extend([a, b])
        event.set()
    bar = partial(foo, 1)
    baz = partial(bar, 2)
    with DigitalInputDevice(pin) as device:
        device.when_activated = baz
        assert not event.is_set()
        pin.drive_high()
        assert event.is_set()
        assert callback_args == [1, 2]
        pin.drive_low()
        event.clear()
        del callback_args[:]
        device.when_activated = bar
        assert not event.is_set()
        pin.drive_high()
        assert event.is_set()
        assert callback_args == [1, device]
```
and it works on all versions of Python3, but fails on Python2 with:
```
=================================== FAILURES ===================================
_________________________ test_input_partial_callback __________________________

    def test_input_partial_callback():
        event = Event()
        pin = MockPin(2)
        callback_args = []
        def foo(a, b):
            callback_args.extend([a, b])
            event.set()
        bar = partial(foo, 1)
        baz = partial(bar, 2)
        with DigitalInputDevice(pin) as device:
            device.when_activated = baz
            assert not event.is_set()
            pin.drive_high()
            assert event.is_set()
            assert callback_args == [1, 2]
            pin.drive_low()
            event.clear()
            del callback_args[:]
>           device.when_activated = bar

tests/test_inputs.py:101: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
gpiozero/devices.py:159: in __setattr__
    return super(GPIOBase, self).__setattr__(name, value)
gpiozero/mixins.py:209: in when_activated
    self._when_activated = self._wrap_callback(value)
gpiozero/mixins.py:286: in _wrap_callback
    @wraps(fn)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

wrapper = <function wrapper at 0x2b100b3e8e60>
wrapped = <functools.partial object at 0x2b100b3e4a48>
assigned = ('__module__', '__name__', '__doc__'), updated = ('__dict__',)

    def update_wrapper(wrapper,
                       wrapped,
                       assigned = WRAPPER_ASSIGNMENTS,
                       updated = WRAPPER_UPDATES):
        """Update a wrapper function to look like the wrapped function
    
           wrapper is the function to be updated
           wrapped is the original function
           assigned is a tuple naming the attributes assigned directly
           from the wrapped function to the wrapper function (defaults to
           functools.WRAPPER_ASSIGNMENTS)
           updated is a tuple naming the attributes of the wrapper that
           are updated with the corresponding attribute from the wrapped
           function (defaults to functools.WRAPPER_UPDATES)
        """
        for attr in assigned:
>           setattr(wrapper, attr, getattr(wrapped, attr))
E           AttributeError: 'functools.partial' object has no attribute '__module__'

/usr/lib/python2.7/functools.py:33: AttributeError
============== 1 failed, 236 passed, 10 skipped in 18.43 seconds ===============
```

waveform80, 2016-09-20:
Damn, so close! Unfortunately I'm tied up in my monstrous refactor of pins right now so I'll have to get back to this later. Still, it's a step forward.

waveform80, 2016-09-20:
(in other words, feel free to dig into this - it won't be duplicated effort as I'm not going to get back to this today!)

lurch, 2016-09-20:
Okay, I did a bit of digging.... ;-)

The reason that Py3 works, but Py2 doesn't, is because functools.update_wrapper in [Py3](https://hg.python.org/cpython/file/3.4/Lib/functools.py#l43) includes a try / except that [Py2](https://hg.python.org/cpython/file/2.7/Lib/functools.py#l17) doesn't.

So, here's _one_ way of fixing the problem so that my enhanced test passes under **all** Python versions. But I dunno if it's the best way, because I've never really messed about with wrapped functions, partial functions or monkey-patching before...
```patch
diff --git a/gpiozero/compat.py b/gpiozero/compat.py
index 31e5f35..4178aa7 100644
--- a/gpiozero/compat.py
+++ b/gpiozero/compat.py
@@ -81,3 +81,35 @@ class frozendict(collections.Mapping):
             hashes = map(hash, self.items())
             self.__hash = functools.reduce(operator.xor, hashes, 0)
         return self.__hash
+
+
+# Backported from py3.4, to allow partials to be wrapped under py2.7
+def update_wrapper(wrapper,
+                   wrapped,
+                   assigned = functools.WRAPPER_ASSIGNMENTS,
+                   updated = functools.WRAPPER_UPDATES):
+    """Update a wrapper function to look like the wrapped function
+
+       wrapper is the function to be updated
+       wrapped is the original function
+       assigned is a tuple naming the attributes assigned directly
+       from the wrapped function to the wrapper function (defaults to
+       functools.WRAPPER_ASSIGNMENTS)
+       updated is a tuple naming the attributes of the wrapper that
+       are updated with the corresponding attribute from the wrapped
+       function (defaults to functools.WRAPPER_UPDATES)
+    """
+    for attr in assigned:
+        try:
+            value = getattr(wrapped, attr)
+        except AttributeError:
+            pass
+        else:
+            setattr(wrapper, attr, value)
+    for attr in updated:
+        getattr(wrapper, attr).update(getattr(wrapped, attr, {}))
+    # Issue #17482: set __wrapped__ last so we don't inadvertently copy it
+    # from the wrapped function when updating __dict__
+    wrapper.__wrapped__ = wrapped
+    # Return the wrapper so this can be used as a decorator via partial()
+    return wrapper
diff --git a/gpiozero/mixins.py b/gpiozero/mixins.py
index 069b6da..736f550 100644
--- a/gpiozero/mixins.py
+++ b/gpiozero/mixins.py
@@ -13,11 +13,13 @@ from functools import wraps, partial
 from threading import Event
 from collections import deque
 from time import time
+import mock
 try:
     from statistics import median
 except ImportError:
     from .compat import median
 
+from .compat import update_wrapper
 from .threads import GPIOThread
 from .exc import (
     BadEventHandler,
@@ -283,10 +285,11 @@ class EventsMixin(object):
                     # If the above fails, try binding with a single parameter
                     # (ourselves). If this works, wrap the specified callback
                     inspect.getcallargs(wrapped_fn, *(args + (self,)))
-                    @wraps(fn)
-                    def wrapper():
-                        return fn(self)
-                    return wrapper
+                    with mock.patch('functools.update_wrapper', update_wrapper)
+                        @wraps(fn)
+                        def wrapper():
+                            return fn(self)
+                        return wrapper
                 except TypeError:
                     raise BadEventHandler(
                         'value must be a callable which accepts up to one '
diff --git a/tests/test_inputs.py b/tests/test_inputs.py
index 85e26a5..49e8974 100644
--- a/tests/test_inputs.py
+++ b/tests/test_inputs.py
@@ -83,9 +83,10 @@ def test_input_event_deactivated():
 def test_input_partial_callback():
     event = Event()
     pin = MockPin(2)
+    callback_args = []
     def foo(a, b):
+        callback_args.extend([a, b])
         event.set()
-        return a + b
     bar = partial(foo, 1)
     baz = partial(bar, 2)
     with DigitalInputDevice(pin) as device:
@@ -93,6 +94,16 @@ def test_input_partial_callback():
         assert not event.is_set()
         pin.drive_high()
         assert event.is_set()
+        assert callback_args == [1, 2]
+        pin.drive_low()
+        event.clear()
+        del callback_args[:]
+        device.when_activated = bar
+        assert not event.is_set()
+        pin.drive_high()
+        assert event.is_set()
+        assert callback_args == [1, device]
+
 
 def test_input_wait_active():
     pin = MockPin(2)
```

waveform80, 2016-09-21:
Ah, I was just about to note the requirement for `mock` (under Py2.x anyway - mock got included in the stdlib sometime in 3.x but I forget where exactly). Personally, I'd simply back-port the 3.4 into compat and have it monkey patch functools once during import if the Python version was 2.x (or <3.4? Not sure when the exception check was introduced).

This would be dirtier: it doesn't clean up the monkey patch or re-apply it between invocations. However, firstly it removes the necessity for the mock dependency on 2.x, and secondly if you're relying on the monkey patch being applied every time you're doing something seriously weird, and I don't think we care about use-cases *that* weird. As for un-doing the monkey patch between invocations, I don't see that it harms any potential callers of `wraps` so I'm not terribly bothered about that either.

Our general code-writing practice is to write idiomatic Python 3. Occasionally that's going to mean doing horrid things on Python 2.x, but I'm okay with that on the basis it's the end-of-life platform so that's where the nasty stuff belongs :)

commit 9046ea30 (2016-09-20): Fix #436

Permit partial, partialmethod, and wraps to be used with event handlers
commit f607a27c (2016-09-20): Merge pull request #455 from waveform80/partial-handlers

Fix #436