gpiozero/gpiozero #437: BUG: MCP3202

joyopd, 2016-09-15:
Hi,

I found bug at reading using this ADC, probably caused by it's 12bit resolution.
Measured values are from 0 to 1 (or in raw from 0 to 4095) for input voltage from 0 to 1/4 of V_ref, then again from 0 to 1 for voltages between 1/4 to 1/2 of V_ref. This repeats also for voltages 1/2 to 3/4 V_ref and 3/4 to V_ref.
Tested with gpiozero 1.3.1

Thank you

bennuttall, 2016-09-17:
I believe @waveform80 fixed this in #430. Can you try the version currently in master and see if it works?

joyopd, 2016-09-19:
Tested, AD conversion works fine for MCP3202, both channel, both methods (.value and .raw_value).

Bug is still present in rest of the MCP320* family (MCP3204, MCP3208). I don't own these ADs, I just tested this classes with my MCP3202. They should have the same communication syntax just with more channels. 

Thank you for fast fix.

lurch, 2016-09-19:
A comment on the #430 that @bennuttall linked to says "(the protocol on the x1 and x2 series was wrong)", so I suspect that using one chip to test the driver-code for other chips in the family is probably an invalid thing to try?
In which case perhaps this should be closed as fixed?

joyopd, 2016-09-20:
Bug for MCP3202 is fixed. I will try to buy some other AD from the family for testing.
Thank you.

closed 2016-09-20 by joyopd

waveform80, 2016-09-20:
Yeah, I'm afraid the 3204 and 3208 *don't* have the same communication protocol as the 3202 (the mistake we made originally was assuming the xx02's had the same protocol as the xx04's and xx08's but they don't - I originally spotted the issue when playing with an Erlang Embedded board which has a 3002 on it, then went back and read the data sheets more thoroughly).