gpiozero/gpiozero #438: V1.3.1 - DistanceSensor - Failed to add edge detection

SteveAmor, 2016-09-16:
Executing

```
#!/usr/bin/env python3
from gpiozero import DistanceSensor
from time import sleep

sensor = DistanceSensor(echo=18, trigger=17)
while True:
    print('Distance: ', sensor.distance * 100)
    sleep(1)
```
I get:

```
Traceback (most recent call last):
  File "./distance.py", line 5, in <module>
    sensor = DistanceSensor(echo=18, trigger=17)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 135, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 600, in __init__
    self.pin.when_changed = callback
  File "/usr/lib/python3/dist-packages/gpiozero/pins/__init__.py", line 240, in <lambda>
    lambda self, value: self._set_when_changed(value),
  File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 233, in _set_when_changed
    bouncetime=self._bounce)
RuntimeError: Failed to add edge detection
```
If I execute it again, it runs ok.

I ran ```sudo apt-get update``` before installing V1.3.1.

bennuttall, 2016-12-01:
Does anyone have any background on what's causing this?

lurch, 2016-12-02:
Sounds like an RPi.GPIO problem. I wonder if it's something that might have been fixed in the recent 0.6.3 release?

tvanesse, 2017-01-16:
Still having this in 2017.

bennuttall, 2017-01-16:
There hasn't been a new release since this was posted. The issue has not been dealt with yet.

SteveAmor, 2017-03-18:
**Resolved**

Checked the version of RPi.GPIO with the following command:  

```find /usr | grep -i RPi.gpio```

And found it was version `0.6.0a3` so I updated to `0.6.3` using the following command:

```sudo apt-get install python-rpi.gpio python3-rpi.gpio```

closed 2017-03-18 by SteveAmor