gpiozero/gpiozero #439: Consider __add__ method for gpiozero classes

bennuttall, 2016-09-18:
As discussed at PyConUK, we should consider implementing `__add__` methods to gpiozero classes.

The suggestion was to make `__add__` return a tuple combining `self.value` and `other.value`.

lurch, 2016-09-18:
Could you give an example of how that would be used in practice?

bennuttall, 2016-12-01:
I suggested it mostly as a "what would `__add__` do if it existed on, say, `LED`? And would it be of any use" and @waveform80 said he thought it would make compositing devices easier internally.

lurch, 2016-12-02:
I've never needed to use `__add__` before - can you give an example of what it might do in GpioZero if it were added?

bennuttall, 2016-12-02:
`__add__` is very useful. Not what I'm suggesting, but imagine:

```python
class LED:
    def __init__(self, pin):
        self.pin = pin

    def __add__(self, other):
        return LEDBoard(self.pin, other.pin)
```

(ignoring the issue of creating a LEDBoard on pins already in use)

But what we talked about doing was more like:

```python
def __add__(self, other):
    return (self.value, other.value)
```

lurch, 2016-12-02:
...so:
```python
red = LED(4)
green = LED(5)
red.on()
res = red + green
```
would result in `res` being `(True, False)` ?
Not sure if I can see the usefulness in that TBH (unless I'm missing something obvious).

Does it also mean that:
```python
red = LED(4)
green = LED(5)
blue = LED(6)
red.on()
blue.on()
res = red + green + blue
```
would result in `res` being `(True, (False, True))` ?

closed 2019-01-15 by bennuttall