gpiozero/gpiozero #442: Consider adding boolean logic functions in source tools

bennuttall, 2016-09-19:
Ok so we do already have what we need for what I'm talking about, but the names are not obvious for this use case. For example:

```python
from gpiozero import LED, Button
from gpiozero.tools import all_values

button_a = Button(17)
button_b = Button(18)
led = LED(21)

led.source = all_values(button_a.values, button_b.values)
```

When used with two inputs, `any_values` is essentially an OR function. Similarly, `all_values` is an AND function in this case.

We have `all_values`, `any_values` and `negated`, which is all you need to do boolean logic. However, the functions (usefully) work for more than two inputs. It might be more obvious if there were functions for `and_values`, `or_values` and `xor_values`.

Not 100% certain this is necessary but if agreed, we should do it.

lurch, 2016-09-19:
Hah, see my point 4. in this discussion ;-)  #268

lurch, 2016-09-23:
(and of course `negated

bennuttall, 2016-09-26:
Yep!

bennuttall, 2016-12-01:
@waveform80 do you have an opinion on this? Should we add `and_values`, `or_values` and `xor_values`, or not bother?

closed 2019-02-06 by bennuttall