gpiozero/gpiozero #443: Add bee-bot / turtle robot example

bennuttall, 2016-09-19:
As discussed last night in the pub, let's add a bee-bot / turtle robot example for the docs.

Similarly to the existing button-controlled robot recipe, this would comprise two motors and some buttons attached to a Pi. But rather than the events being fired immediately as a (not) remote controller, the directions would be appended to a list and run in sequence when the fifth button (go button) is pressed.

bennuttall, 2016-09-19:
```python
from gpiozero import Button, Robot
from time import sleep
from signal import pause

robot = Robot(????)

left = Button(?)
right = Button(?)
forward = Button(?)
backward = Button(?)
go = Button(?)

instructions = []

def add_instruction(btn):
    instructions.append({
        left:     (-1, 1),
        right:    (1, -1),
        forward:  (1, 1),
        backward: (-1, -1),
    }[btn])

def do_instructions():
    instructions.append((0, 0))
    robot.source_delay = 0.5
    robot.source = instructions
    sleep(robot.source_delay * len(instructions))
    del instructions[:]

go.when_pressed = do_instructions
for button in (left, right, forward, backward):
    button.when_pressed = add_instruction

pause()
```

bennuttall, 2016-09-19:
https://www.bee-bot.us/

lurch, 2016-09-19:
Also similar to https://www.kickstarter.com/projects/ryanteckltd/turtlteck/description

Perhaps you'd want a longer delay-between-instructions for going forward, compared to turning on the spot? I guess each should probably be configurable in the constructor, given that different robots' motors will run at different speeds.

commit ed12ac19 (2017-01-08): Add bee-bot / turtle robot example, close #443
closed 2017-01-08 by bennuttall