gpiozero/gpiozero #444: pinout command line tool

bennuttall, 2016-09-19:
We currently have a `pi_info` function which allows you to find out information on the Pi you're on, or another specified Pi. We want to add a command line tool `pinout` which would print the pinout for your Pi by default, or a specified Pi.

See #354 for info on how to get pinout from pi_info.

An example of argparse: https://github.com/pyjokes/pyjokes/blob/master/scripts/pyjoke

closed 2017-06-13 by waveform80

bennuttall, 2017-06-22:
@waveform80 has something broken with `pinout`? I get the following trying to run anything (on my laptop) - including looking up help `pinout -h` and specific Pi revisions e.g. `pinout -r 0002`.

```
$ pinout -h
/home/ben/Projects/github/rpi-distro/python-gpiozero/gpiozero/devices.py:499: PinFactoryFallback: Failed to load factory rpigpio: No module named 'RPi'
  'Failed to load factory %s: %s' % (name, str(e))))
/home/ben/Projects/github/rpi-distro/python-gpiozero/gpiozero/devices.py:499: PinFactoryFallback: Failed to load factory rpio: No module named 'RPIO'
  'Failed to load factory %s: %s' % (name, str(e))))
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Can't connect to pigpio at localhost(8888)

Did you start the pigpio daemon? E.g. sudo pigpiod

Did you specify the correct Pi host/port in the environment
variables PIGPIO_ADDR/PIGPIO_PORT?
E.g. export PIGPIO_ADDR=soft, export PIGPIO_PORT=8888

Did you specify the correct Pi host/port in the
pigpio.pi() function? E.g. pigpio.pi('soft', 8888)
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
/home/ben/Projects/github/rpi-distro/python-gpiozero/gpiozero/devices.py:499: PinFactoryFallback: Failed to load factory pigpio: failed to connect to localhost:8888
  'Failed to load factory %s: %s' % (name, str(e))))
/home/ben/Projects/github/rpi-distro/python-gpiozero/gpiozero/devices.py:499: PinFactoryFallback: Failed to load factory native: unable to open /dev/gpiomem or /dev/mem; upgrade your kernel or run as root
  'Failed to load factory %s: %s' % (name, str(e))))
Traceback (most recent call last):
  File "/home/ben/Projects/github/rpi-distro/gpiozero-env/bin/pinout", line 11, in <module>
    load_entry_point('gpiozero', 'console_scripts', 'pinout')()
  File "/home/ben/Projects/github/rpi-distro/gpiozero-env/lib/python3.5/site-packages/pkg_resources/__init__.py", line 560, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/home/ben/Projects/github/rpi-distro/gpiozero-env/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2648, in load_entry_point
    return ep.load()
  File "/home/ben/Projects/github/rpi-distro/gpiozero-env/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2302, in load
    return self.resolve()
  File "/home/ben/Projects/github/rpi-distro/gpiozero-env/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2308, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/ben/Projects/github/rpi-distro/python-gpiozero/gpiozero/__init__.py", line 22, in <module>
    from .devices import (
  File "/home/ben/Projects/github/rpi-distro/python-gpiozero/gpiozero/devices.py", line 528, in <module>
    Device.pin_factory = _default_pin_factory()
  File "/home/ben/Projects/github/rpi-distro/python-gpiozero/gpiozero/devices.py", line 500, in _default_pin_factory
    raise BadPinFactory('Unable to load any default pin factory!')
gpiozero.exc.BadPinFactory: Unable to load any default pin factory!
```

waveform80, 2017-06-23:
Fixed in eb175cc129c1b27e6dfe775b994ebf16b4d06fb4

commit 32d6db4d (2017-06-13): Merge pull request #504 from RPi-Distro/pinout-cli-tool

Add pinout cli tool, close #444, close #497