gpiozero/gpiozero #445: Graphical Pin Factory

bennuttall, 2016-09-19:
Could we add a new pin factory which created a GUI representing the pin header?

As you define input devices, those pins become clickable. As you define output pins, they can flash. Essentially just LEDs and Buttons would be a useful first version. I wouldn't particularly request advanced features beyond this but would not object. It certainly wouldn't be a priority.

We could also create a pin factory which paired with a real pin implementation, so it both showed the graphical representation and worked for real on the pins.

lurch, 2016-09-19:
Something like a GpioZero version of the SenseHAT emulator? ;-)

thijstriemstra, 2016-09-19:
I discovered [fritizing](http://fritzing.org/home/) today; would be really cool to import/export there. A simple .csv import/export of the pin/btn/led etc config would be useful as well (that anyone with libreoffice or excel can update and load etc).

lurch, 2016-09-20:
Some of the pictures on http://gpiozero.readthedocs.io/en/latest/recipes.html were made with Fritzing IIRC :)

What kind of import/export data did you have in mind? Things like "Button on GPIO25, LED on GPIO12"? That's probably more a question for the Fritzing folks than the GpioZero folks. Or are you expecting GpioZero to create 'template' objects for you?

bennuttall, 2017-01-08:
Not a solution to this issue, but related. I threw together a GUI using guizero to allow interaction with mockpins and to visualise the state of real pins:

https://twitter.com/ben_nuttall/status/818053505411907586

https://gist.github.com/bennuttall/e8075517e548e213a967bccc47dfd4e5

bennuttall, 2018-09-20:
I did some more work on this yesterday at the PyCon UK sprints: https://gist.github.com/bennuttall/e8075517e548e213a967bccc47dfd4e5

[image]

It needs implementing as a pin factory based on mockpin, which may be easy enough. Use of this pin factory would depend on guizero and tk. tk is present in Raspbian and guizero landing in Raspbian isn't out of the question. I don't think it needs to be a hard dependency though.