gpiozero/gpiozero #451: Developer docs

bennuttall, 2016-09-19:
We could do with a page explaining to developers wanting to contribute where things are, how the class hierarchy works and why we do the things we do. It should include conventions, contributing guidelines and information on how to run the tests and how to use `MockPin` and things.

@doismellburning anything else you think developers would need?

lurch, 2016-09-19:
I find writing and testing code on my laptop using the MockPins (before then transferring it to a real Pi) to be quite a handy development method.

It also ought to detail the big list of prerequisites needed before you can successfully do 'make doc'.

Isn't the class hierarchy already explained by the diagrams [like this](http://gpiozero.readthedocs.io/en/latest/api_input.html#base-classes)?

lurch, 2016-09-19:
And this should also mention the requirement to pip-install a Pin implementation (typically rpi.gpio) into each virtualenv.

thijstriemstra, 2016-09-19:
I have a few of these ssd1306 oled screens that I could add. How do I handle new dependencies? Where does my class go? Test coverage required?

lurch, 2016-09-20:
@thijstriemstra At the 'level' that GpioZero operates at currently (buttons, LEDs, motors, etc.), I'm not sure if an OLED screen would be a good fit? Feel free to create a separate issue though, where it can be discussed further.

thijstriemstra, 2016-09-20:
@lurch I see what you mean, but it's actually as simple as https://github.com/rm-hull/ssd1306/blob/master/examples/pi_logo.py but thanks for the feedback.

doismellburning, 2016-09-25:
So yeah, from my perspective, "how do the classes fit together" would be a nice and convenient intro, and definitely a good thing for onboarding new contributors, but you _can_ go look at the code and work it out yourself.

"How to run the tests" - useful, but something ideally mostly automated - there's `make test` which is great, but right now the UX is "get a bunch of errors" because it assumes deb/lsb tooling and manual installation of dependencies; hopefully at some point I'll find time to make #450 good which should help a little here. Fundamentally, again, you _can_ still just push a PR and let Travis find any problems.

The things @lurch mention sound absolutely ideal for inclusion.

For me the usual biggest thing is intent and policy What are the guiding principles? https://gpiozero.readthedocs.io/en/v1.3.1/contributing.html is awesome, <3 for that - are there any more? Usually there's more and they're unstated and they're a pain to discover. Are there devices you're explicitly uninterested in supporting? Are there features you specifically don't want to implement?

waveform80, 2016-09-25:
> It also ought to detail the big list of prerequisites needed before you can successfully do 'make doc'.

Does the "make develop" target not take care of installing all the necessary stuff via `pip install -e .[doc,test]`? I know there's some stuff like mscgen, dot, and inkscape that won't get pulled in (for image generation) ... but there's not a great deal I can do about those things (system package manager stuff) but "make doc" with an unaltered repo *should* work after "make develop".

> And this should also mention the requirement to pip-install a Pin implementation (typically rpi.gpio) into each virtualenv.

Indeed - I should add that #459 should help a bit with this in as much as default pin factories that fail to load (e.g. as happens when you've forgotten to install RPi.GPIO in a venv) will now fire a warning when importing gpiozero.

> So yeah, from my perspective, "how do the classes fit together" would be a nice and convenient intro, and definitely a good thing for onboarding new contributors, but you can go look at the code and work it out yourself.

Yeah ... I'd strongly recommend new devs start at the concrete devices and work their way backward to the point they're comfortable with. The stuff at the base of the device hierarchy is likely to scare the pants off people if they see it up-front, and the less said about the mess I'm making in the pins implementations the better (but the base abstract classes are probably worth reading through so you can see what you can do with pins ... I've added a load of ASCII art diagrams into the pins docs in #459 to try and make this a little easier for new devs).

> "How to run the tests" - useful, but something ideally mostly automated - there's make test which is great, but right now the UX is "get a bunch of errors" because it assumes deb/lsb tooling and [...]

Definitely - I would *love* to be able to test all the variants on my local machine. I just never seem to find the time to get to grips with tox (or I'm too lazy and just think "yeah, I could dig into that, but I could just push to Travis").

On deps - same question as above: does "make develop" not install the necessary stuff? If not, that's my fault (as is the fact that "make develop" should be better documented!)

lurch, 2016-09-25:
> but "make doc" with an unaltered repo should work after "make develop".

On a 'fresh box' I always need to install 'ctags' before 'make develop' will work, and IIRC you need to install a whole bunch of Latex stuff before 'make doc' will work (I'd previously suggested breaking 'make doc' up into more granular sub-targets). And as you say, 'dot' (graphviz) and 'inkscape' also need to be documented as pre-requisites.

> I'd strongly recommend new devs start at the concrete devices and work their way backward to the point they're comfortable with. The stuff at the base of the device hierarchy is likely to scare the [...]

But again, it's not obvious from the current docs which are "concrete devices" and which are "base abstract classes". I don't even think the colouring in the class-hierarchy diagrams is explained anywhere?

waveform80, 2016-09-26:
> On a 'fresh box' I always need to install 'ctags' before 'make develop' will work...

Ah, I keep forgetting we don't have an installation chapter in the docs. Would it help if I did a copy-paste-munge of the [picamera install](http://picamera.readthedocs.io/en/release-1.12/install.html) chapter? I probably wouldn't stick it up front in the docs given we're aiming for GPIO Zero to be "installed by default" in Raspbian. Maybe somewhere nearer the back? For that matter, I should probably think about doing the same with picamera ... it's been default in Raspbian for a good while.

> But again, it's not obvious from the current docs which are "concrete devices" and which are "base abstract classes". I don't even think the colouring in the class-hierarchy diagrams is explained anywhere?

Doh - that's my fault. You're absolutely right: the key can be inferred from reading the corresponding definitions but that's rubbish compared to having an up-front explanation. I'll bung one in the relevant places.

lurch, 2016-09-26:
> Would it help if I did a copy-paste-munge of the picamera install chapter?

Ah, that looks quite good :)  Might be worth splitting the "development" and "test" sections off into a different page though? (as they won't be needed by the majority of users)

bennuttall, 2016-09-26:
I think that's a good idea. I'd point out more clearly that it's installed by default (in both picamera and gpiozero), and demonstrate how to upgrade the single package (i.e. `apt-get install python3-picamera` will upgrade picamera to the latest version). Upgrading a very old image could involve downloading all of mathematica again...

waveform80, 2016-09-26:
Ah yes, upgrades. Good point. That's a potentially common use-case. So - install (and upgrade) chapter near the front or back of the manual?

Either way I'll include some nice bold text at the start of the Raspbian section explaining it's installed by default so you probably don't need to do anything except maintain it (I'll do picamera too while I'm at it).

waveform80, 2016-09-26:
Okay, just pushed an update to the picamera docs - have a look at [the latest build](http://picamera.readthedocs.io/en/latest/) and let me know if that looks reasonable for GPIO Zero as well.

Basically the changes involve:

* Installation as the first chapter, but keep it real short - just Raspbian and non-Raspbian (don't need the firmware bit as that's camera specific)
* Development as a much later chapter (before API docs? After? I opted for before so it'd go next to camera hardware which is relevant for picamera)
* Wherever it gets placed, Contributing should either go directly before it or even get merged with it
* Development chapter includes full dependency list and I tweaked it to use virtualenvwrapper because it's much friendlier.
* Tempted to reduce the main ToC to one level (as in picamera) as GPIO Zero's just looks scarily huge at the mo.

lurch, 2016-09-26:
> update to the picamera docs - have a look at the latest build

In http://picamera.readthedocs.io/en/latest/install.html I'm not sure if it's appropriate to link 'Raspbian' to http://www.raspbian.org/ - should it instead link to https://www.raspberrypi.org/downloads/raspbian/ ?
I think the RPF version of Raspbian is what we expect most users to be using, and this is actually slightly different to the 'pure' Raspbian found at raspbian.org

Is it worth mentioning if the version of picamera pre-included in Raspbian includes the 'array' stuff or not?

http://picamera.readthedocs.io/en/latest/development.html
Given how huge the texlive packages are (and does installing inkscape only work if you have X11 installed?), I wonder if it's worth having a separate list of dependencies needed for building the documentation? (As opposed to just building the code & running the tests)

waveform80, 2016-09-27:
> In http://picamera.readthedocs.io/en/latest/install.html I'm not sure if it's appropriate to link 'Raspbian' to http://www.raspbian.org/ - should it instead link to [...]

Ah, good point - I hadn't realized they differed (just bunged in whatever Google came up with). I'll get that fixed up in a mo.

> Is it worth mentioning if the version of picamera pre-included in Raspbian includes the 'array' stuff or not?

Erm ... yeah, probably. The Raspbian package always includes it because pulling in numpy isn't that big of a deal (given it's not building) and I kinda figured anyone that wanted that would be advanced enough to figure it out for themselves, but it's probably worth a note in there.

> http://picamera.readthedocs.io/en/latest/development.html
Given how huge the texlive packages are (and does installing inkscape only work if you have X11 installed?), I wonder if it's worth having a separate list of dependencies needed for building the documentation? (As opposed to just building the code & running the tests)

Inkscape will pull in a whole pile of X11 deps (if you're on a system entirely without X) but then so will a lot of other stuff (even graphviz has X deps because it can display preview windows). Texlive is indeed huge, although I suspect with its X11 and GTK deps included, Inkscape is probably even larger. 

It is kinda ridiculous that we use Inkscape for SVG conversion; unfortunately nothing else does a *really* reliably good job of SVG conversion yet (imagemagick breaks on simple stuff; librsvg is getting *close* to being good enough for most purposes but most distros' versions of it are still too outdated to be really useful). Unfortunately this is something I've played around with a fair bit as my wife's comic-site uses SVG for most internal representation (it uses librsvg for default PNG rendering because obviously the server it runs on doesn't have X, but there's an option to override the PNG output for the occasions when librsvg gets stuff wrong).

I could change the "develop" target so it installs `gpiozero[test]` instead of `gpiozero[doc,test]` and add something like and "develop-docs" target to do the other. Then I could split the Development section into "Developing Code" and "Developing Docs" or something like that?

lurch, 2016-09-27:
> I could change the "develop" target so it installs gpiozero[test] instead of gpiozero[doc,test] and add something like a "develop-docs" target to do the other. Then I could split the Development section into "Developing Code" and "Developing Docs" or something like that?

Sounds good :+1: But perhaps "Developing Code" and "Building Docs"? (IMHO "Developing Docs" sounds a bit strange). And it'd probably be worth mentioning _why_ the building docs section is different from the developing code section i.e. because of the large dependencies for building docs :)

closed 2019-01-15 by bennuttall

bennuttall, 2019-01-23:
I closed this last week thinking it wasn't going to happen and then @waveform80 comes along and BAM! f05efa993e1b8a2c130e785e352e913f6fdada86