gpiozero/gpiozero #456: Include LICENCE.txt in packages?

lurch, 2016-09-20:
I'm not familiar at all with Python packaging, but should `MANIFEST.in` include:
```
include LICENCE.txt
```
?

waveform80, 2016-09-20:
Erm ... good question. "Probably" is my answer, but I'd have to check what that does with the deb packaging (which partially depends on the python packaging)

thijstriemstra, 2016-09-27:
why not?

mirabilos, 2017-02-07:
@waveform80 the Debian packaging ships its own `debian/copyright` file in the Debian source package, which ends up in `/usr/share/doc/*/copyright` in the Debian binary package, and will omit the `LICENCE.txt` file completely, but its presence in the `.orig.tar.gz` file is still required by the licence itself.

lurch, 2017-02-08:
Fixed as part of #535

closed 2017-02-08 by lurch