gpiozero/gpiozero #457: DistanceSensor

mcghemic, 2016-09-20:
I have tried from version 1.2 to 1.31 to use the DistanceSensor with HC-SR04.  Using pizero (have tried Pi3 as well)  I am connected to Echo=18 Trigger=17.  The sensor respond and I can use the example to get the distance, but when I get close to about 10 to 11 cm the reading returns with 0.0.  

using default setting (assume using RPi.GPIO) with gpiozero

I can use RPi.GPIO  with same pins and can get to almost 1 cm before readings are unusable.

Have tried changing max distance,  threshold, with no luck or change.

Would like to use gpiozero to control motor, sensor and servo but this is a road block.  Saw some others who seem to have same issue, but could not find a solution.  

Not sure if there is an issue with gpiozero or something else

lurch, 2016-09-20:
> Not sure if there is an issue with gpiozero

Yep ;-) See #435 and #385

lurch, 2016-09-23:
Closing for now as a duplicate of the already-reported issues.

closed 2016-09-23 by lurch