gpiozero/gpiozero #461: Re-title header on B+ onwards

waveform80, 2016-09-22:
Just realized (whilst in the middle of the pins re-factor, mulling over exactly how to handle inter-device pin locking) that the header on the B+ onwards *isn't* P1 ... it's J8. I'll get this fixed up while I'm at it to avoid any confusion of users looking at their Pi's and going "but where the hell is P1?!"

lurch, 2016-09-22:
...but it's labelled as "GPIO".

waveform80, 2016-09-22:
look at the bit just by pin 1

waveform80, 2016-09-22:
had a look on a Zero, a 2B, and a 3B (haven't got a B+ but I've found some photos online) - I'm pretty sure that header is labelled J8 but you're right there's a "GPIO" label silk-screened on too.

lurch, 2016-09-22:
Ah yes. They even kept the "J8" labelling on the PiZero, which is surprising.

lurch, 2016-09-22:
BTW I had already noticed this (P1 discrepancy), but I didn't think it was worth fixing ;-)

waveform80, 2016-09-22:
Somehow the fact you'd already noticed it doesn't surprise me ;)

bennuttall, 2016-09-22:
What do P1 and J8 stand for?

lurch, 2018-12-24:
@bennuttall I think this has been done, as the pinout diagrams now include 'J8' ?

closed 2019-01-14 by bennuttall